esphome/issues #3971: ESP8266 devices won't connect with WPA3, while ESP32 does

acfnews, 2022-12-27:
### The problem

I have several ESP32 and ESP8266 devices.
After migrating home devices (laptops/phones/iPads) onto WPA3 wifi network, it is now time for my ESP devices.

I successfully migrated a ESP32 device, but none of my ESP8266 boards will connect to a WPA3 wifi network.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Board

demos-d1-mini

### Component causing the issue

wifi

### Example YAML snippet

```yaml
I am using a secret.yaml file with both the old and new network names:
`# Your Wi-Fi SSID and password
wifi_ssid: "XXX"
wifi_password: "YYYY"

wifi_ssid_wpa3: "333"
wifi_password_wpa3: "ZZZZ"`

In the devices config I have either:
`  platform: ESP8266
  board: d1_mini`
or
`  platform: ESP32
  board: esp32dev`

The wifi config looks like on all:
`wifi:
  networks:
  - ssid: !secret 'wifi_ssid'
    password: !secret 'wifi_password'
    priority: 1.0
  - ssid: !secret 'wifi_ssid_wpa3'
    password: !secret 'wifi_password_wpa3'
    priority: 1.5`
```

### Anything in the logs that might be useful for us?

```txt
the logs only show connecting to the old network, no errors
```

acfnews, 2022-12-27:
I have tested with another ESP32, which connects successfully using WPA3.

However, also tested with another spare ESP8266 (Wemos D1 mini), and this does not connect using WPA3.
See attached the log, I have scrubbed my SSID's and MAC addresses.
[logs_mhi-test_upload.txt](https://github.com/esphome/issues/files/10308086/logs_mhi-test_upload.txt)

Jason2866, 2022-12-28:
esp8266 wifi Stack (closed source) does not support WPA3.
esp8266 is EoL.

acfnews, 2023-01-03:
OK, thanks!

acfnews, 2023-01-03:
hhhhm, still wondering what is stated here:
https://www.espressif.com/en/products/modules/esp8266

It states wpa3 is supported.

Jason2866, 2023-01-03:
Different SDK. The underlying SDK used for Arduino does not support.
There is not just one SDK for the esp8266.

acfnews, 2023-01-15:
ok, thanks! will try to shift my hardware to esp32 boards.

closed 2023-01-15 by acfnews

bkbartk, 2023-04-20:
> esp8266 wifi Stack (closed source) does not support WPA3.
> esp8266 is EoL.

Not according to this article. 
https://www.espressif.com/en/products/longevity-commitment
that there is a successor doesn’t mean it’s eol. It’s still supported until 2026.  

It neither means wpa3 or any feature will be added. But definitely not eol.