esphome/issues #3972: sun: sunrise and sunset => Wrong time

Loic691, 2022-12-27:
### The problem

My sun object gps position is bases on a France location.
My time zone is 
`time:
  - platform: sntp
    timezone: Europe/Paris
    id: sntp_time`

I can display local time and GMT time wirth lambda
`- platform: template
    id: local_time
    name: Locale Time
    lambda: |-
      char str[25];
      time_t currTime = id(sntp_time).now().timestamp;
      strftime(str, sizeof(str), "%H:%M:%S", localtime(&currTime));
      //strftime(str, sizeof(str), "%Y-%m-%dT%H:%M:%S.000Z", gmtime(&currTime));
      return (std::string) str;`

But the sunset time and sunrise time are only refered to GMT time. So these 2 times are 1 hour gap (for my position)
`- platform: sun
    name: "${friendly_name} next sunrise"
    type: sunrise
    id: next_sunrise
    update_interval: 4h
  - platform: sun
    name: "${friendly_name} next sunset"
    type: sunset
    id: next_sunset
    update_interval: 4h`

Is this normal ? Bug ? Or how can I solve this issue ?

### Which version of ESPHome has the issue?

 v2022.12.3 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

AZ Delivery ESP32 NodeMCU Develomeent Module

### Component causing the issue

sun

ssieb, 2022-12-27:
Your local time text sensor is overly complicated.  All you need is:
`return id(sntp_time).now().strftime("%H:%M:%S");`

Where are you seeing the sunrise and sunset times?

Loic691, 2022-12-27:
Thanks. Yes I can change to this
I see 
Next sunset : 15:58:55 instead of 16:58:55
Next sunrise : 07:16:45 instead of 08:16:45

ssieb, 2022-12-27:
Where do you see that?  Also, show logs.

Loic691, 2022-12-28:
This is the state of the sensors....

ssieb, 2022-12-28:
Please show logs.

Loic691, 2022-12-28:
```
[11:44:16][D][text_sensor:067]: 'Prochain lever du soleil': Sending state '08:16:57'
[11:44:19][D][text_sensor:067]: 'Prochain coucher du soleil': Sending state '16:58:55'
```

ssieb, 2022-12-28:
Aren't those the times you're wanting?  Or did you write the earlier comment backwards?

Loic691, 2022-12-28:
> Aren't those the times you're wanting? Or did you write the earlier comment backwards?

These sunrise/sunset time are GMT time and not local time. I have a delta of 1 hour and 2 hours when daylight.

ssieb, 2022-12-28:
I just tested with your config and I get:
```
[14:13:16][D][text_sensor:067]: 'Time': Sending state '23:13:16'
[14:13:21][D][text_sensor:067]: 'Sunset': Sending state '17:02:07'
[14:13:32][D][text_sensor:067]: 'Sunrise': Sending state '08:46:00'
```
And that's correct for all 3 values.

Loic691, 2022-12-28:
So I think I found my problem
A startup, I have bad time with the delta of one hour, because de SNTP request isn't yet pushed...
My sensor is refreshed each 4 hours. So since 4 hours, the time is bad.
At next refresh, the time become correct.
Do you know how I can call the update after the smtp refresh ?

ssieb, 2022-12-28:
https://esphome.io/components/time/index.html#on-time-sync-trigger

closed 2022-12-28 by ssieb

Loic691, 2022-12-28:
Yes I justed test and it was OK
my code

```
time:
  - platform: sntp
    timezone: Europe/Paris
    id: weatherstation_sntp_time
    on_time_sync: # after sntp update
      then:
        - logger.log: "Synchronized system clock via SNTP" 
        - component.update: weatherstation_next_sunrise
        - component.update: weatherstation_next_sunset
```