esphome/issues #3975: Unable to get weather state and attributes

tricoveri, 2022-12-27:
### The problem

Hello,
I am trying to access the weather entity state and attributes in a lambda call in order to publish them to a nextion display with poor results. They seem to be returned as floats but even trying  with .c_str() I can't convert the to char

Other sensors like temperature sensors and time are correctly displayed

Thanks in advance for any help

### Which version of ESPHome has the issue?

2022.8.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.12.7

### What platform are you using?

ESP32

### Board

Esp32 Devkit 1

### Component causing the issue

weather and weather hourly

### Example YAML snippet

```yaml
esphome:
  name: esp32svegliacamera
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xx"
  password: "x"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32svegliacamera Hotspot"
    password: "o4QwBuzpvYVFFF"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.135
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

captive_portal:

# Enable logging
logger:
  #baud_rate: 0

# Enable Home Assistant API
api:
  password: "xx"

ota:
  password: "xx"
  
    
uart:
  id: uart2
  rx_pin: 16
  tx_pin: 17
  baud_rate: 9600
################################################################################
sensor:
#--------------------------------------------------------------------    
  - platform: homeassistant
    id: bme280Tempcucina
    entity_id: sensor.bme280tempcucina
    on_value:
      then:
        lambda:  |-
            id(nextion1).set_component_text_printf("stanza.tempcuc","%.1f",id(bme280Tempcucina).state);
             
  - platform: homeassistant
    id: bme280Tempgarage
    entity_id: sensor.bme280tempgarage
    on_value:
      then:
        lambda:  |-
            id(nextion1).set_component_text_printf("stanza.tempext","%.1f",id(bme280Tempgarage).state);

  - platform: homeassistant
    id: tempo
    entity_id: weather.casa
    internal: true
                

    
################################################################################  
time:
  - platform: homeassistant
    id: esptime
    on_time:
    
      - seconds: 0
        minutes: '*'
        then:
            lambda: |-
                auto time_text = id(esptime).now().strftime("%H:%M");
                id(nextion1).set_component_text("page0.mainwatch",time_text.c_str());
                    
    
      - seconds: 30
        minutes: '*'
        then:
            lambda:  |-
                
                id(nextion1).set_component_text("stanza.t3",id(tempo).state.c_str());
                
          
          
           
######################################  
display:
  - platform: nextion
    uart_id: uart2
    update_interval: 5s
    id: nextion1
```

### Anything in the logs that might be useful for us?

```txt
Sveglia.yaml:155:61: error: request for member 'c_str' in 'tempo->esphome::homeassistant::HomeassistantSensor::<anonymous>.esphome::sensor::Sensor::state', which is of non-class type 'float'
*** [.pioenvs/esp32svegliacamera/src/main.cpp.o] Error 1
```

ssieb, 2022-12-27:
A `sensor:` is always a float.  If you want a string, then you need to use a `text_sensor:` instead.
Github issues are not for support requests.  Use the discord server or one of the forums.

tricoveri, 2022-12-28:
hey thanks for the message, I didn't get the difference, will use discord right away