esphome/issues #3976: Compiling fails on Raspberry Pi 3  (with compile_process_limit: 1)

reinisb, 2022-12-27:
### The problem

I'm running Home Assistant OS (64-bit) with ESPHome (non-dev) on Raspberry Pi 3 (1GB RAM).

When I compile a project that includes a single component then compiling succeeds. However, when I try compiling a project with more than one component then compiling fails:
```
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
```

I did add `compile_process_limit: 1` to project's YAML as introduced [here](https://esphome.io/changelog/2022.11.0.html#running-esphome-on-lower-powered-machines) but it does not seem to make a difference.

Can anyone please help? Thank you!

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

compiler?

### Example YAML snippet

```yaml
esphome:
  name: esp32-devkitc-v4
  compile_process_limit: 1
  libraries:
    - xxtea-iot-crypt@2.0.1

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: "redacted"
  password: "redacted"

external_components:
  - source: github://reinisb/esphome-danfoss-eco-mod@v1.0.0
ble_client:
  - mac_address: 00:04:2f:06:32:18
    id: living_room_danfoss
climate:
  - platform: danfoss_eco
    name: "Living room Danfoss"
    ble_client_id: living_room_danfoss
    secret_key: 9f5de77f9c929090793264784a0ee361
    battery_level:
      name: "Living room Danfoss Battery Level"
    temperature:
      name: "Living room Danfoss Temperature"
    update_interval: 2min

## If I include the below lines then compile fails (there is nothing wrong with the below component)
output:
  - platform: ledc
    pin: 25
    frequency: 25000 Hz
    id: pwm_noctua_fan_5v
    
fan:
  - platform: speed
    output: pwm_noctua_fan_5v
    name: "Noctua Fan"
```

### Anything in the logs that might be useful for us?

```txt
#end of the log
Linking /data/esp32-devkitc-v4/.pioenvs/esp32-devkitc-v4/firmware.elf
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
*** [/data/esp32-devkitc-v4/.pioenvs/esp32-devkitc-v4/firmware.elf] Error 1
========================= [FAILED] Took 611.82 seconds =========================
```

reinisb, 2022-12-27:
cc @jesserockz ([ref](https://github.com/esphome/esphome/pull/3952))

ssieb, 2022-12-27:
Did it compile up to that point?  You probably just don't have enough memory for the final link.

reinisb, 2022-12-27:
Yep, everything seem to compile up to the very last point. Is there any way to free up some memory?

ssieb, 2022-12-27:
Maybe add some more swap? You should probably just do the compiling on something else.  But regardless, this isn't an esphome issue.  Support requests need to go to discord or one of the forums.

closed 2022-12-27 by ssieb

reinisb, 2022-12-28:
I don't know how to add swap. I created a HA OS VM with 4GB RAM, but compiling fails almost instantly.

[image]

ssieb, 2022-12-28:
Please come to the discord server.  It will be a lot easier to help you there.