esphome/issues #3977: ESPHOME stayed at last values if no communication with Peacefair PZEM-016.

mishomvp, 2022-12-27:
### The problem

Hello,

I use PZEM-016, but yesterday I found out that after the communication between the UART and the RS485 is broken, the last measured values remain and the only way to know about it is that you can see a "straight line in the graphs" or restart ESP32.
It would be good if communication between PZEM-016 and the UART, broken ESPHOME to show value "Unavailable" and stops writing data to the database.
Thanks in advance!

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP32

### Board

ESP32-EVB

### Component causing the issue

RS485, Peacefair PZEM-016

ssieb, 2022-12-27:
There's no way for esphome to tell HA that it's not available.  The closest option would be to publish `NaN` after a certain number of update intervals with no data.

mishomvp, 2022-12-28:
Okay. My idea is simply to not have wrong data being written to the database.

ssieb, 2022-12-28:
It's not being written.  HA is just showing the last value received.

closed 2023-05-22 by github-actions[bot]