esphome/issues #3979: climate_ir showing heat/cool option when heat isn't supported

nvx, 2022-12-28:
### The problem

When using the esphome climate_ir with fujitsu_general, setting `supports_heat: false` correctly disables the heat option, but the heat/cool option is still displayed.

I would expect the heat/cool option to only be enabled when both supports_heat and supports_cool are true 

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP8266

### Board

Mirabella Genio Smart IR

### Component causing the issue

climate_ir

### Example YAML snippet

```yaml
remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 33%

remote_receiver:
  id: ir_receiver
  pin:
    number: GPIO5
    inverted: True

climate:
  - platform: fujitsu_general
    name: "Great Room Cooler"
    supports_cool: true
    supports_heat: false
    receiver_id: ir_receiver
    visual:
      min_temperature: 18
      max_temperature: 30
      temperature_step: 1
```
```

### Additional information

I suspect the fix would be as simple as tweaking the code here:
https://github.com/esphome/esphome/blob/9a70bfa47187011ce9eccf7d2939bd067f53d8d6/esphome/components/climate_ir/climate_ir.cpp#L12

Replacing the line:
```
  traits.set_supported_modes({climate::CLIMATE_MODE_OFF, climate::CLIMATE_MODE_HEAT_COOL});
```

with:
```
  traits.set_supported_modes({climate::CLIMATE_MODE_OFF});
  if (this->supports_cool_ and this->supports_heat_)
    traits.add_supported_mode(climate::CLIMATE_MODE_HEAT_COOL);
```

would fix the issue, or perhaps a `supports_heat_cool` option in the yaml would be a better option? thoughts?

nvx, 2023-04-30:
still relevant

prajjwalburman, 2024-04-11:
Agreeing with @nvx, changing [climate_ir.cpp#L12](https://github.com/esphome/esphome/blob/6d480c5f058323933bc1f24de891928f86b36e95/esphome/components/climate_ir/climate_ir.cpp#L12) from 

`traits.set_supported_modes({climate::CLIMATE_MODE_OFF, climate::CLIMATE_MODE_HEAT_COOL});`

to 

`traits.set_supported_modes({climate::CLIMATE_MODE_OFF});
if (this->supports_cool_ and this->supports_heat_)
traits.add_supported_mode(climate::CLIMATE_MODE_HEAT_COOL);
`

shall fix this issue. but only if someone takes a look into this

prajjwalburman, 2024-04-11:
@OttoWinter @jesserockz @oxan @napieraj @CarlosGS @glmnet can you guys please take a look? I can raise a PR if needed

RonaldRegister, 2024-07-02:
I've come across this same issue. A little workaround that is working for me is to disable heat/cool (and fan only) from the card on my dashboards. Still tho; heat/cool shouldn't be present when `supports_heat: false` is set. 

See my screenshot:
[image]

glmnet, 2024-07-03:
Heat/cool is also what is know on many ac as auto

Since ACs which are cool only also has the auto function, then heat/cool mode is not hidden on any ac

petergerard, 2025-01-08:
Seems like extending the climate component to have a `supports_auto` (or `supports_heat_cool`) configuration option that defaults to `true` would be a good way to give more flexibility here.

It would resolve OP’e problem and also allow for ACs to have both cool and auto modes. Further, it would resolve my challenge - I have a multi-head heat pump where auto can cause problems since it would lead to one head calling for heat while another calls for cooling, which leads to an overall system problem.