esphome/issues #3980: DS18b20 timing issues with multiple sensors

messier433, 2022-12-28:
### The problem

Single wire communication to several DS18b20 temperature sensor are not working reliable and fail 90% of the time with CRC errors. The same setup with a custom arduino firmware using pstolarz/OneWireNg^0.10.0 does not show the issue.
The problem only started with connecting more than 1 sensor to the 1wire interface suggesting a change in timing due to added load on the bus. And indeed increasint the time_constant in the esphome function ESPOneWire::read_bit()  in esp_one_wire.cpp to 

`uint32_t timing_constant = 20;//12; `

fixed the issue. 

In the comments it states:
`  // note: for reading we'll need very accurate timing, as the
  // timing for the digital_read() is tight; according to the datasheet,
  // we should read at the end of 16µs starting from the bus low
  // typically, the ds18b20 pulls the line high after 11µs for a logical 1
  // and 29µs for a logical 0`

Therefore a time constant right in the middle of 11 and 29us seems to be the best choice with most margins to unknown bus delays.

### Which version of ESPHome has the issue?

 2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32s2 - custom PCB

### Component causing the issue

dallas

saschaludwig, 2023-01-05:
Is this related to this issue? https://github.com/esphome/issues/issues/3909

JamieR007, 2023-01-08:
I have four DS18B20 sensors in a star configuration at the end of ~6m of CAT5e, using a 3.3kOhm pullup resistor. I started seeing "scratch pad checksum invalid!" errors after upgrading to 2022.12.3 from a much earlier version.

I tried different pullup resistor values, but to no avail.

The 20µs change suggested above seemed to provide no discernable improvement.

However, the change to the interrupt lock does indeed appear to fix the issue.

JamieR007, 2023-01-08:
Also just noticed, comparing the implemenation of DallasTemperatureSensor::read_scratch_pad() with some earlier builds (prior to upgrading to 2022.12.3) there was previously no interrupt lock at all. Otherwise the implemenation looks identical.

craigueh, 2023-02-07:
I too am having this issue with 8 sensors on 4 different Dallas hubs since updating to 2022.12.8, seeing scratchpad checksum invalid errors every 15-20 mins on random sensors. This is a setup that never produced an error in 2 years or running prior to the update.

What do I need to do to implement the interrupt lock fix above?

craigueh, 2023-02-09:
Never mind, I re-read the thread and see I need to make the changes in esp_one_wire.cpp. 

I’ll try it and see how I get on.

JamieR007, 2023-07-10:
Summary of the discussion in the related issue #4543:

- Only certain hardware appears to exhibit the problem
- A workaround is to use separate hubs / pins for each sensor
- A solution has yet to be agreed or implemented for the original issue, described here.

I believe the root of the problem relates to the Interrupt Lock introduced in [PR 3181](https://github.com/esphome/esphome/pull/3181). This lock is not maintained for the duration of the operation in read_scratch_pad, hence why multiple sensors (on the same hub) collide. My proposed solution is described above and in #4543.

I am happy to test this or any alternative solutions as my setup can readily reproduce.

MasterCATZ, 2023-08-01:
So I guess I will stop looking for a hardware issue , system was running great for years reading all the fish tanks now getting crc scratch_pad messages being spammed over 100 sensors 

also seems like the esp keeps having to reboot its self

vdemidov, 2023-11-10:
Strange behaviour. The WT32-ETH01 board has 4 ds18b20 sensors connected. When connected to web interface only, errors are very rare. But when also connected to ESPHome Wireless Logs, more than 50% of updates fail with error "Scratch pad checksum invalid!"

jkosharek, 2025-04-17:
Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid.
Any fix for this, mine just started doing this again once I updated to ESPHome 2025.4.0

davebeckster, 2025-06-01:
Same here.  Noticed after 2025.4.0 installation

tablatronix, 2025-06-24:
I cannot seem to use more than 1 on bus, as soon as I add a second

```
Jun 23 21:25:37 waveshare-6ch-relay 192.168.20.146 dallas.temp.sensor: [C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
Jun 23 21:25:37 waveshare-6ch-relay 192.168.20.146 dallas.temp.sensor: [W][dallas.temp.sensor:031]:   Unable to select an address
```

tablatronix, 2025-06-24:
I expect it to enumerate the entire bus, how to I get the addresses, connect one at a time? Thats mental

tablatronix, 2025-06-24:
I see them , I was greping under 'dallas' not 'one_wire'.