esphome/issues #3983: Compiling fatal error

brewnut, 2022-12-28:
### The problem

When trying to add a new device to ESP home and generate a basic BIN file to install using ESP flasher the compile fails.
I have tried compiling with both the modern format and legacy format and both methods fail.

It always seems to fail on the first two api items of api_connection.cpp.o and api_frame_helper.cpp.o

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP32

### Board

MH ET LIVE ESP32 DevKIT

### Component causing the issue

api_connection.cpp.o

### Example YAML snippet

```yaml
esphome:
  name: deck-led

esp32:
  board: mhetesp32devkit
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Deck-Led Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/deck-led.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing deck-led (board: mhetesp32devkit; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/deck-led/.pioenvs/deck-led/src/esphome/components/api/api_connection.cpp.o
Compiling /data/deck-led/.pioenvs/deck-led/src/esphome/components/api/api_frame_helper.cpp.o
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/deck-led/.pioenvs/deck-led/src/esphome/components/api/api_connection.cpp.o] Error 1
========================= [FAILED] Took 26.99 seconds =========================
```

ssieb, 2022-12-28:
I expect that you're using a pi3 or something else with low memory.  Use the compile process limit, see https://esphome.io/components/esphome.html#configuration-variables.

closed 2022-12-28 by ssieb

brewnut, 2022-12-28:
My HA instance is actually running on a VM on an 12th gen i7. 
I set the compile process limit to 1 and it now compiles all the way to 'Linking /data/tc-led/.pioenvs/tc-led/firmware.elf' before failing.

I realised that the VM only had 2 cores and 1Gb of memory assigned so i have upped it to 4 cores and 8Gb of memory and it now compiles all the way and successfully creates the BIN file.