esphome/issues #3985: [BLE Tracker] TTGO T-Relay board: Bluetooth stops working after 5-7 days, resulting in NaN in HA

kformosa, 2022-12-29:
### The problem

As in the title, I have a TTGO T-Relay board and flashed ESPHome. Been using ESPHome for a while and enabled the BLE tracker when it was available. But since around the .11 update, I am getting NaN in HA after roughly 5-7 days. I just restart the board (pressing Reset) and everything is back to normal - until 5-7 days when I run into the same issue. Still having the same issue with the .12 update.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP32

### Board

TTGO T-Relay

### Component causing the issue

BLE Tracker

### Example YAML snippet

```yaml
esphome:
  name: terrarium-node

esp32:
  board: esp32dev
  framework:
    type: arduino
    
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
  
switch:
  - platform: gpio
    pin: 25
    inverted: true
    name: "LilyGo T-Relay Board"
    restore_mode: ALWAYS_ON
    
  - platform: gpio
    pin: 21
    name: "Relay 1"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/terrarium-node.yaml...
INFO Starting log output from terrarium-node.local using esphome API
INFO Successfully connected to terrarium-node.local
[11:35:07][I][app:102]: ESPHome version 2022.12.3 compiled on Dec 28 2022, 12:34:09
[11:35:07][C][wifi:504]: WiFi:
[11:35:07][C][wifi:362]:   Local MAC: 94:E6:86:DA:4B:40
[11:35:07][C][wifi:363]:   SSID: [redacted]
[11:35:07][C][wifi:364]:   IP Address: 192.168.1.145
[11:35:07][C][wifi:366]:   BSSID: [redacted]
[11:35:07][C][wifi:367]:   Hostname: 'terrarium-node'
[11:35:07][C][wifi:369]:   Signal strength: -36 dB ▂▄▆█
[11:35:07][C][wifi:373]:   Channel: 6
[11:35:07][C][wifi:374]:   Subnet: 255.255.255.0
[11:35:07][C][wifi:375]:   Gateway: 192.168.1.1
[11:35:07][C][wifi:376]:   DNS1: 192.168.1.210
[11:35:07][C][wifi:377]:   DNS2: 192.168.1.50
[11:35:07][C][logger:293]: Logger:
[11:35:07][C][logger:294]:   Level: DEBUG
[11:35:07][C][logger:295]:   Log Baud Rate: 115200
[11:35:07][C][logger:296]:   Hardware UART: UART0
[11:35:07][C][switch.gpio:076]: GPIO Switch 'LilyGo T-Relay Board'
[11:35:07][C][switch.gpio:084]:   Inverted: YES
[11:35:07][C][switch.gpio:099]:   Restore Mode: always ON
[11:35:07][C][switch.gpio:031]:   Pin: GPIO25
[11:35:07][C][switch.gpio:076]: GPIO Switch 'Relay 1'
[11:35:07][C][switch.gpio:099]:   Restore Mode: restore defaults to OFF
[11:35:07][C][switch.gpio:031]:   Pin: GPIO21
[11:35:07][C][switch.gpio:076]: GPIO Switch 'Relay 2'
[11:35:07][C][switch.gpio:099]:   Restore Mode: restore defaults to OFF
[11:35:07][C][switch.gpio:031]:   Pin: GPIO19
[11:35:07][C][switch.gpio:076]: GPIO Switch 'Relay 3'
[11:35:07][C][switch.gpio:099]:   Restore Mode: restore defaults to OFF
[11:35:07][C][switch.gpio:031]:   Pin: GPIO18
[11:35:07][C][switch.gpio:076]: GPIO Switch 'Relay 4'
[11:35:07][C][switch.gpio:099]:   Restore Mode: restore defaults to OFF
[11:35:07][C][switch.gpio:031]:   Pin: GPIO5
[11:35:07][C][bluetooth_proxy:065]: Bluetooth Proxy:
[11:35:07][C][bluetooth_proxy:066]:   Active: YES
[11:35:07][C][esp32_ble_tracker:870]: BLE Tracker:
[11:35:07][C][esp32_ble_tracker:871]:   Scan Duration: 300 s
[11:35:07][C][esp32_ble_tracker:872]:   Scan Interval: 1100.0 ms
[11:35:07][C][esp32_ble_tracker:873]:   Scan Window: 1100.0 ms
[11:35:07][C][esp32_ble_tracker:874]:   Scan Type: ACTIVE
[11:35:07][C][esp32_ble_tracker:875]:   Continuous Scanning: True
[11:35:07][C][captive_portal:088]: Captive Portal:
[11:35:07][C][mdns:103]: mDNS:
[11:35:07][C][mdns:104]:   Hostname: terrarium-node
[11:35:07][C][ota:093]: Over-The-Air Updates:
[11:35:07][C][ota:094]:   Address: terrarium-node.local:3232
[11:35:07][C][ota:097]:   Using Password.
[11:35:07][C][api:138]: API Server:
[11:35:07][C][api:139]:   Address: terrarium-node.local:6053
[11:35:07][C][api:143]:   Using noise encryption: NO
```

bdraco, 2022-12-30:
Use esp-idf instead for the framework type

kformosa, 2023-01-16:
I changed the platform to "ESP32" since I couldn't make that option work/compile.
Still, after 7 days the same problem occurred. Had to reset the board and all good until 7 days pass.

kformosa, 2023-02-03:
I updated to the lastest ESPHome but still have the same issue: basically the 2 bluetooth devices being proxied via this ESP board return "NaN" to HA. Sometimes after just after 1 day up-time.

Anyone else encountering same issue?

privatesam, 2023-02-04:
Got the same issue here - just need a restart switch in the bluetooth proxy YAML so I can automate a restart on the esp32 from HA.

bdraco, 2023-02-04:
The [ESP32 Platform](https://esphome.io/components/esp32.html) component should be configured to use the esp-idf framework, as the arduino framework uses significantly more memory and performs poorly with the Bluetooth proxy enabled. When switching from arduino to esp-idf, make sure to update the device with a serial cable as the partition table is different between the two frameworks as [OTA Update Component](https://esphome.io/components/ota.html) updates will not change the partition table.

https://esphome.io/components/bluetooth_proxy.html

privatesam, 2023-02-05:
@bdraco would this be the same if I flashed directly from Tasmota to esphome?

bdraco, 2023-02-05:
> @bdraco would this be the same if I flashed directly from Tasmota to esphome?

If you did tasmota to esphome you need to use the serial cable to flash it once as you likely have a tasmota partition table

privatesam, 2023-02-05:
> > @bdraco would this be the same if I flashed directly from Tasmota to esphome?
> 
> If you did tasmota to esphome you need to use the serial cable to flash it once as you likely have a tasmota partition table 

I used a USB cable and chrome to flash from ESPhome website so not OTA - should that have wiped the tasmota partition table?

bdraco, 2023-02-05:
> I used a USB cable and chrome to flash from ESPhome website so not OTA - should that have wiped the tasmota partition table?

yes

kformosa, 2023-02-07:
I flashed the board with the esp-idf framework using a cable over the weekend. Seems that the issue has been resolved. 
Thanks @bdraco