esphome/issues #3988: esp32s2 compile error: cannot convert 'USBCDC*' to 'HardwareSerial*' in assignment

Mrten, 2022-12-30:
### The problem

I'm trying to use an wemos lolin s2 pico for an co2 sensor, but it won't compile using the arduino framework. I'd like to add a neopixel so using esp-idf is out of the question as neopixelbus nor fastled are supported with esp-idf.

This is a minimal yaml:

### Which version of ESPHome has the issue?

2022.12.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

LayerCakeMakes, 2022-12-30:
This is definitively related to #3419

vidplace7, 2023-01-07:
@Mrten the platformio `lolin_s2_pico` board profile enables USB CDC by default -- this also applies to `lolin_s2_mini`
See: https://github.com/platformio/platform-espressif32/blob/v5.2.0/boards/lolin_s2_pico.json

You can try overriding this using `platformio_options` > `board_build.extra_flags`.

Example:
```yaml
esphome:
  name: co2
  platformio_options:
    board_build.extra_flags:
      - "-DARDUINO_USB_CDC_ON_BOOT=0"  # Override, defaults to '-DARDUINO_USB_CDC_ON_BOOT=1'
      
esp32:
  board: lolin_s2_pico
  framework:
    type: arduino

uart:
  - rx_pin: GPIO16
    tx_pin: GPIO17
    baud_rate: 9600
```

All I could find on USB CDC: https://docs.espressif.com/projects/esp-idf/en/v4.4.3/esp32s2/api-guides/usb-otg-console.html

Mrten, 2023-01-09:
Thanks for the suggestions and the pointer to the json!

However, If I do that it compiles but does not seem to do anything after upload and reboot. There's no port anymore to connect the esphome dashboard_install to (but that would be expected I guess) so I can't see any logs.

I then added a `status_led` on the correct GPIO but it does not light up after boot; if I add a `wifi` it won't connect (and still no status light).

zuk, 2023-01-20:
Adding

```
  platformio_options:
    board_build.extra_flags:
      - "-DARDUINO_USB_CDC_ON_BOOT=0"
```

to the `esphome` section did indeed make things work on my end. This is on a `esp32-qt-py-s2`. Thanks for the tip @vidplace7 !

Mrten, 2023-03-01:
Oh, and setting the wifipower to less than the default (17?) is also still mandatory: otherwise it crashes still.
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 10
```