esphome/issues #3989: fatal error: Arduino.h: No such file or directory when compiling for rp2040 on mac m1

gepatto, 2022-12-30:
### The problem

Esphome installed through pip on a Mac M1 (Ventura 13.1)
throws: ` fatal error: Arduino.h: No such file or directory`
when compiling for the rp2040

Tried reinstalling and installing with brew as an alternative, but gives the same error.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

pip

### What platform are you using?

RP2040

### Board

picow

### Example YAML snippet

```yaml
esphome:
  name: picow

rp2040:
  board: pico

output:
  - platform: gpio
    pin:
      number: 25
      mode: output
    id: LED

interval:
  - interval: 500ms
    then:
    - output.turn_on: LED
    - delay: 250ms
    - output.turn_off: LED
```

### Anything in the logs that might be useful for us?

```txt
esphome compile pico.yaml
INFO Reading configuration pico.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing picow (board: pico; framework: arduino; platform: platformio/raspberrypi @ 1.7.0)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduino-mbed @ 3.1.1
 - framework-arduinopico @ 1.20604.0 (2.6.4)
 - tool-rp2040tools @ 1.0.2
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0
|   |-- SocketWrapper @ 1.0
Compiling .pioenvs/picow/src/esphome/components/rp2040/core.cpp.o
Compiling .pioenvs/picow/src/esphome/components/rp2040/gpio.cpp.o
Compiling .pioenvs/picow/src/esphome/components/rp2040/preferences.cpp.o
Compiling .pioenvs/picow/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/picow/src/main.cpp.o
[... 161 lines cut ...]
lib_deps =
build_flags =
upload_flags =

; ========== AUTO GENERATED CODE BEGIN ===========
[env:picow]
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 1m
build_flags =
    -DUSE_ARDUINO
    -DUSE_RP2040
    -DUSE_RP2040_FRAMEWORK_ARDUINO
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -fno-exceptions
framework = arduino
lib_deps =
    ${common.lib_deps}
platform = platformio/raspberrypi @ 1.7.0
platform_packages =
    earlephilhower/framework-arduinopico @ https://github.com/earlephilhower/arduino-pico/releases/download/2.6.4/rp2040-2.6.4.zip
; =========== AUTO GENERATED CODE END ============
```

Deishelon, 2023-01-10:
Same issue on Linux:

```
(esp_venv) $:~/projects/pico-esphome$ esphome version
Version: 2022.12.3
(esp_venv) $:~/projects/pico-esphome$ python --version
Python 3.10.6
```

ssieb, 2023-01-10:
You need:
```yaml
rp2040:
  board: pico
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git
```

closed 2023-01-10 by ssieb

grokbeer, 2023-01-11:
Thanks for raising this issue @gepatto as I was struggling with this too and thought I'd tried everything.  As @ssieb commented, the fix is to override the platform_version.  This will be required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged.

It looks like you're using a Pico W so there's a couple of other changes to your YAML you'll need as well.  The "board" for Pico W is "rpipicow" and the on-board LED is accessed via GPIO pin 32 rather than 25 for the Pico.

See @geerlingguy's https://github.com/geerlingguy/pico-w-garage-door-sensor/blob/master/led-blink.yml for an updated led-blink.yml for the Pico W.