esphome/issues #3991: Dallas DS18B20 on Raspberry Pi Pico W - "Resetting bus for read failed"

tango7nz, 2023-01-01:
### The problem

First of all thanks for all the great work, and for bringing Raspberry Pi Pico support!

Just tried it for the first time and I can get the Pico up and running fine, but hitting an issue when trying to use a Dallas DS18B20 sensor.

It scans and picks up the sensor OK, but then when trying to read from it, it gives the following error:

[13:44:52][W][dallas.sensor:131]: 'Garage Temperature' - Resetting bus for read failed!

I have a 4.7K resister connected between 3.3V and the data line. Nothing else connected to the Pico. 

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Component causing the issue

Dallas

### Example YAML snippet

```yaml
esphome:
  name: temp-garage

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

dallas:
  pin: GPIO22

sensor:
  - platform: dallas
    address: 0xa40215718dabff28
    name: "Garage Temperature"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/temp-garage.yaml...
INFO Starting log output from temp-garage.local using esphome API
INFO Successfully connected to temp-garage.local
[13:44:31][I][app:102]: ESPHome version 2022.12.3 compiled on Jan  1 2023, 12:51:25
[13:44:31][C][logger:293]: Logger:
[13:44:31][C][logger:294]:   Level: DEBUG
[13:44:31][C][logger:295]:   Log Baud Rate: 115200
[13:44:31][C][logger:296]:   Hardware UART: USB_CDC
[13:44:31][C][dallas.sensor:075]: DallasComponent:
[13:44:31][C][dallas.sensor:076]:   Pin: GPIO22
[13:44:31][C][dallas.sensor:077]:   Update Interval: 60.0s
[13:44:31][D][dallas.sensor:082]:   Found sensors:
[13:44:31][D][dallas.sensor:084]:     0xa40215718dabff28
[13:44:31][C][dallas.sensor:089]:   Device 'Garage Temperature'
[13:44:31][C][dallas.sensor:089]:     Device Class: 'temperature'
[13:44:31][C][dallas.sensor:089]:     State Class: 'measurement'
[13:44:31][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[13:44:31][C][dallas.sensor:089]:     Accuracy Decimals: 1
[13:44:31][C][dallas.sensor:097]:     Address: 0xa40215718dabff28
[13:44:31][C][dallas.sensor:098]:     Resolution: 12
[13:44:31][C][mdns:103]: mDNS:
[13:44:31][C][mdns:104]:   Hostname: temp-garage
[13:44:31][C][ota:093]: Over-The-Air Updates:
[13:44:31][C][ota:094]:   Address: temp-garage.local:2040
[13:44:31][C][ota:097]:   Using Password.
[13:44:31][C][api:138]: API Server:
[13:44:31][C][api:139]:   Address: temp-garage.local:6053
[13:44:31][C][api:141]:   Using noise encryption: YES
[13:44:52][W][dallas.sensor:131]: 'Garage Temperature' - Resetting bus for read failed!
[13:44:52][D][sensor:126]: 'Garage Temperature': Sending state inf °C with 1 decimals of accuracy
```

faquin, 2023-01-25:
I am experiencing the same behavior.
Home Assistant 2022.12.9
ESPHome 2022.12.6

pss34, 2023-04-02:
The same here with esphome 2023.3.0:
```
[00:13:26][VV][scheduler:225]: Running interval 'update' with interval=60000 last_execution=4294964675 (now=57387)
[00:13:26][VV][scheduler:031]: set_timeout(name='0xbb01193cfbc1a928', timeout=94)
[00:13:27][VV][scheduler:225]: Running timeout '0xbb01193cfbc1a928' with interval=94 last_execution=57393 (now=57492)
[00:13:27][W][dallas.sensor:131]: 'Boiler temp' - Resetting bus for read failed!
[00:13:27][V][sensor:076]: 'Boiler temp': Received new state nan
[00:13:27][D][sensor:127]: 'Boiler temp': Sending state nan °C with 1 decimals of accuracy
[00:13:27][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[00:13:27][V][json:056]: Size after shrink 72 bytes
[00:13:27][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 1824070647
  state: nan
  missing_state: NO
}
[00:13:27][V][component:204]: Component dallas took a long time for an operation (0.05 s).
[00:13:27][V][component:205]: Components should block for at most 20-30ms.
```

KingJames2023, 2023-04-29:
i hve it up and running no problem however for me it shows the wrong temperature saying my room temperature is 21.375 celcius eventhough its more like 13 and it also mesures my body temperature at 21.9 celcius. is this a bad sensor or is something wrong with the dallas library?

austwhite, 2023-06-17:
I have to ask where you got your Dallas DS18B20 sensors from?  
I've got a couple that give the read fail error, but they were from a cheap Chinese source.    The ones I have from Element 14 and other reputable sources all work perfectly.
Maybe the sensors are not genuine and are not working right?