esphome/issues #3992: Modbus controller - Custom command lambda - Not processing received data

silicoon84, 2023-01-01:
### The problem

Hi all,

I've been trying to use the modbus controller to talk to a Daly bms over UART. [https://www.aliexpress.com/item/1005002205134036.html](url)

Below is an excerpt on how the UART communication works, which is pulled from here [https://github.com/maland16/daly-bms-uart](url) 
<html>
<body>
<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Here's what an outgoing packet looks like. It's always fixed 13 bytes

<html>
<body>

Start Byte | Host Address | Command ID | Data Length | Data | Checksum
-- | -- | -- | -- | -- | --
0xA5 | 0x80 | See below | 0x08 (fixed) | 0x0000000000000000 (8 bytes) | Sum of other 12 bytes

</body>
</html>

<html>
<body>
<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">This is what an incoming packet looks like. In this case it's the "Voltage, Current, and SOC" command. (I've included the response i get in the logs)</p>

Start Byte | Host Address | Command ID | Data Length | Data | Checksum
-- | -- | -- | -- | -- | --
0xA5 | 0x01 | 0x90 (one example) | 0x08 (fixed?*) | 0x023A0000753001ED (8 bytes) | 0x0D (Sum of other 12 bytes)

</body>
</html>

So i've been trying to transition my mqtt sensor using this code [https://github.com/maland16/daly-bms-uart](url) into ESPhome using modbus controller and custom command.
The command gets sent and the 13 byte response comes back, as you can see in the logs.
I then started working through building a lambda to process the response, but it does not seem like it gets triggered.
Even if i simply try a ESP_LOGD command in the lambda only, i dont see the output in the log, so it doesnt look like the response triggers the lambda. The example yalm from the documentation does not trigger either.

I've included the my yaml configuration, which i've pulled from the custom command example from the documentation [https://esphome.io/components/sensor/modbus_controller.html](url)

Any help would be epic.

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.4

### What platform are you using?

ESP32

### Board

esp32-wroom-32

### Component causing the issue

Modbus Controller

### Example YAML snippet

```yaml
esphome:
  name: cabin-bms-reader

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: debug

api:
  reboot_timeout: 0s
  password: !secret api_password
  
ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  - id: mod_bus
    tx_pin: 18
    rx_pin: 19
    baud_rate: 9600
    stop_bits: 1
    debug:
      direction: BOTH
      dummy_receiver: false
      after:
        delimiter: "\n"
      sequence:
        - lambda: UARTDebug::log_string(direction, bytes);

modbus:
 -  uart_id: mod_bus
    id: modbus1

modbus_controller:
  - id: bms
    ## the Modbus device addr
    address: 0x01
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 15s
    
sensor:
  - platform: modbus_controller
    modbus_controller_id: bms
    name: "BMS Test"
    id: total_test
    #   address: 0x158
    #   register_type: "read"
    custom_command: [0xa5, 0x80, 0x91, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xbe]
    value_type: U_DWORD_R
    ## the command returns an float value using 4 bytes
    lambda: |-
      ESP_LOGD("Modbus Sensor Lambda","Got new data" );
      union {
        float float_value;
        uint32_t raw;
      } raw_to_float;
      if (data.size() < 4 ) {
        ESP_LOGE("Modbus Sensor Lambda", "invalid data size %d",data.size());
        return NAN;
      }
      raw_to_float.raw =   data[0] << 24 | data[1] << 16 | data[2] << 8 |  data[3];
      ESP_LOGD("Modbus Sensor Lambda", "FP32 = 0x%08X => %f", raw_to_float.raw, raw_to_float.float_value);
      return raw_to_float.float_value;
    unit_of_measurement: w
    accuracy_decimals: 1
```

### Anything in the logs that might be useful for us?

```txt
[14:33:05][D][uart_debug:158]: >>> "\xA5\x80\x91\b\x00\x00\x00\x00\x00\x00\x00\x00\xBEuU"
[14:33:06][D][uart_debug:158]: <<< "\xA5\x01\x91\b\f\xF5\f\f\xD2\x02\x01|\xA9"
[14:33:06][D][uart_debug:158]: >>> "\xA5\x80\x91\b\x00\x00\x00\x00\x00\x00\x00\x00\xBEuU"
[14:33:06][D][uart_debug:158]: <<< "\xA5\x01\x91\b\f\xF5\f\f\xD2\x02\x01|\xA9"
[14:33:06][D][uart_debug:158]: >>> "\xA5\x80\x91\b\x00\x00\x00\x00\x00\x00\x00\x00\xBEuU"
[14:33:06][D][uart_debug:158]: <<< "\xA5\x01\x91\b\f\xF5\f\f\xD2\x02\x01|\xA9"
[14:33:06][D][uart_debug:158]: >>> "\xA5\x80\x91\b\x00\x00\x00\x00\x00\x00\x00\x00\xBEuU"
[14:33:06][D][uart_debug:158]: <<< "\xA5\x01\x91\b\f\xF5\f\f\xD2\x02\x01|\xA9"
[14:33:07][D][uart_debug:158]: >>> "\xA5\x80\x91\b\x00\x00\x00\x00\x00\x00\x00\x00\xBEuU"
[14:33:07][D][uart_debug:158]: <<< "\xA5\x01\x91\b\f\xF5\f\f\xD2\x02\x01|\xA9"
[14:33:07][D][modbus_controller:032]: Modbus command to device=1 register=0x5418 countdown=0 no response received - removed from send queue
```

silicoon84, 2023-01-01:
Closed this as I realised that there is already a sensor component for the Daly BMS [https://esphome.io/components/sensor/daly_bms.html](url) 
Great to see the speed at which things come out! Wasn't around when I first started trying to transition this a while back

closed 2023-01-01 by silicoon84

ssieb, 2023-01-02:
In case you haven't realized it already, your esphome version is really old.  You probably missed the transition to the new repository.

silicoon84, 2023-01-02:
> In case you haven't realized it already, your esphome version is really old. You probably missed the transition to the new repository.

Yeah I realised yesterday when I was trying to do some Bluetooth stuff. Weirdly broke on a few of my Ha instances and had to remove and install for it to get updates again. Cheers