esphome/issues #3993: ESP32 crashes on e-ink display update after enabling touchscreen

dannyyy, 2023-01-01:
### The problem

I use an external component to drive the e-ink display from the Lilygo-T5-4.7 board. Just using the `display` doesn't cause any issues. Even when the `i2c` component is present. But as soon as I add the `touchscreen` component, the ESP32 immediately crashes after a display component update is initiated (doesn't matter whether manually or automatically).

Version `2022.11.5` was working without any issues. Tested with the same snipped of code.

Could the update of the ESP32 platform caused this issue? What has been changed under the hood?

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

display/touchscreen

### Example YAML snippet

```yaml
substitutions:
  esp_name: Lilygo_T5_47
  esp_hostname: df-lilygo-t5-47

external_components:
  - source: github://ashald/esphome@lilygo-t5-47
    components:
      - lilygo_t5_47

logger:

esp32:
  board: esp32dev
  framework:
    type: arduino

esphome:
  name: ${esp_hostname}

i2c:
  sda: 15
  scl: 14

display:
  - platform: lilygo_t5_47
    id: eink

touchscreen:
  - platform: lilygo_t5_47
    interrupt_pin: GPIO13
    on_touch:
      then:
        - lambda: |-
            ESP_LOGI("check","Touched at x: %i y: %i ", touch.x, touch.y);
```

### Anything in the logs that might be useful for us?

```txt
[17:38:19]Rebooting...
[17:38:19]ets Jul 29 2019 12:21:46
[17:38:19]
[17:38:19]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:38:19]configsip: 0, SPIWP:0xee
[17:38:19]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:38:19]mode:DIO, clock div:2
[17:38:19]load:0x3fff0030,len:1184
[17:38:19]load:0x40078000,len:13132
[17:38:19]load:0x40080400,len:3036
[17:38:19]entry 0x400805e4
[17:38:19][I][logger:258]: Log initialized
[17:38:19][I][app:029]: Running through setup()...
[17:38:19][I][i2c.arduino:175]: Performing I2C bus recovery
[17:38:19][C][lilygo_t5_47.touchscreen:029]: Setting up Lilygo T5 4.7 Touchscreen...
[17:38:19][D][lilygo_t5_47.display:035]: Resetting periodic full update countdown to 0.
[17:38:19][I][app:062]: setup() finished successfully!
[17:38:19][D][lilygo_t5_47.display:051]: Periodic full update countdown 0.
[17:38:19][D][lilygo_t5_47.display:055]: Full update!
[17:38:24]E (10246) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[17:38:24]E (10246) task_wdt:  - loopTask (CPU 1)
[17:38:24]E (10246) task_wdt: Tasks currently running:
[17:38:24]E (10246) task_wdt: CPU 0: IDLE
[17:38:24]E (10246) task_wdt: CPU 1: loopTask
[17:38:24]E (10246) task_wdt: Aborting.
[17:38:24]
[17:38:24]abort() was called at PC 0x400e0635 on core 0
[17:38:24]
[17:38:24]
[17:38:24]Backtrace:0x40083a95:0x3ffbe81c |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40083a95: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
[17:38:24]
[17:38:24]
[17:38:24]
[17:38:24]
[17:38:24]ELF file SHA256: 0000000000000000
[17:38:24]
```

awulf, 2023-04-12:
I'm experiencing the same issue. Tried messing around with the source code of this component and have found that if commenting out line 34, where it attaches the interrupt, it stops crashing, but obviously doesn't function anymore.
`  this->interrupt_pin_->attach_interrupt(Store::gpio_intr, &this->store_, gpio::INTERRUPT_FALLING_EDGE);`

Comparing how this interrupt is setup to other components that use interrupts, I can't see whats going wrong here.

awulf, 2023-04-13:
Actually turns out adding any component with an interrupt handler will crash ESPHome. I tired `duty_cycle` for example.
This only happens when loading the custom component display driver (tried 3 variants), all based on EPDiy, which is required to drive the display. So I suspect something with EPDiy is clashing with ESPHome.

kvakulo, 2024-03-26:
I experienced the same issue when attempting to use the `remote_receiver` and the lilygo `display` simultaneously.

Your comment regarding interrupts led me to discover a compatibility issue related to the RMT module driver. With [this change in epdiy](https://github.com/kvakulo/epdiy/commit/1b129e9181305de567b7d0a2381fe1c0d0914aa9), I'm now able to use the `display` and the `remote_receiver` at the same time.

In [this branch](https://github.com/kvakulo/esphome-lilygo-t5-47/tree/lilygo-t5-47), I've adjusted the `lilygo_t5_47` component to refer to the version of epdiy with the change.

However, since I don’t have a touchscreen, I’m not certain if it will also resolve that issue.