esphome/issues #3994: magiquest wand IR protocol has been updated

tattonchantry, 2023-01-02:
### The problem

My wands work perfectly with arduino which uses:
https://github.com/Arduino-IRremote/Arduino-IRremote/blob/master/src/ir_MagiQuest.hpp

They only work some of the time with ESPHome using:
https://github.com/esphome/esphome/blob/dev/esphome/components/remote_base/lg_protocol.cpp

The rest of the time it is mis-diagnosed as a pronto signal.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2022.12.8

### What platform are you using?

ESP32

### Board

ESP32 DEVKIT V1, ESP8266 D1 MINI, and NODEMCU

### Component causing the issue

VS1838B

### Example YAML snippet

```yaml
binary_sensor:
  - platform: remote_receiver
    name: "Brogan Wand All"
    id: bw_all
    magiquest:
      wand_id: 0x022D3464
  - platform: remote_receiver
    name: "Pink Wand All"
    id: pw_all
    magiquest:
      wand_id: 0x00633400
  - platform: remote_receiver
    name: "Beni Wand All"
    id: benw_all
    magiquest:
      wand_id: 0x022BAA88
```

### Anything in the logs that might be useful for us?

```txt
[20:57:08][D][remote.magiquest:079]: Received MagiQuest: wand_id=0x022BAA88, magnitude=0x0817
[20:57:08][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 000A 0023 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 000A 0023 0009 0023 0014 0019 0008 0024 0014 0019 0008 0024 0014 0019 0013 0018 0014 0019 0008 0024 0013 001A
[20:57:08][D][remote.pronto:240]: 0008 0024 0013 001A 0008 0024 0013 001A 0009 0023 0013 001A 0008 0024 0009 0024 0009 0023 0013 001A 0008 0024 000A 0023 0008 0024 000A 0023 0009 0023 0009 0024 0009 0023 0013 001A 0009 0023 000A 0023 0008 0024 000A 0023 0009 0023 000A 0023 0013 0019 000A 0023 0013 0019 0013 001A 0012 001A 0013 001A 0009 0023 000A 0023 0009 0023 0014 0019 0013 06C3
[20:57:08][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 0007 06C3
[20:57:09][D][remote.magiquest:079]: Received MagiQuest: wand_id=0x022BAA88, magnitude=0x0C0B
[20:57:09][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 0008 0024 000A 0022 0008 0024 000A 0022 0008 0024 000A 0022 0008 0024 000A 0022 0014 0018 000A 0022 0008 0024 000A 0022 0014 0018 000B 0021 0013 0018 000B 0022 0013 0018 0014 0019 0015 0017 000A 0022 0014 0018
[20:57:09][D][remote.pronto:240]: 000A 0022 0014 0018 000B 0021 0014 0018 000A 0022 0013 0018 000A 0022 0008 0024 000A 0022 0014 0018 000A 0022 0008 0024 000A 0022 0008 0023 000A 0022 0008 0024 000A 0023 0014 0018 0014 0019 0008 0024 000A 0023 0008 0024 000A 0022 0008 0023 000A 0023 0014 0018 000A 0022 0014 0018 0013 001A 0013 0018 000A 0023 0008 0023 0014 0019 0008 0023 0014 06C3
[20:57:09][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 0005 06C3
[20:57:09][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 0004 06C3
[20:57:10][D][remote.magiquest:079]: Received MagiQuest: wand_id=0x022BAA88, magnitude=0x0817
[20:57:10][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 0008 0024 000B 0021 0008 0024 000B 0022 0008 0024 000A 0022 0008 0024 000B 0022 0013 0018 000A 0022 0008 0024 000B 0021 0014 0018 000A 0022 0013 0018 000A 0022 0014 0018 0014 0019 0015 0017 000A 0022 0013 0018
[20:57:10][D][remote.pronto:240]: 000A 0022 0014 0018 000B 0021 0013 0018 000B 0021 0013 0018 000A 0022 0008 0024 000A 0022 0014 0018 000A 0022 0008 0024 000A 0022 0008 0024 000A 0022 0008 0024 000A 0022 0014 0018 000A 0022 0008 0023 000A 0022 0008 0024 000A 0022 0008 0024 0013 001A 0008 0024 0014 0019 0013 0018 0014 0019 0013 0018 000A 0023 0008 0023 000A 0023 0013 0018 0014 06C3
[20:57:10][D][remote.magiquest:079]: Received MagiQuest: wand_id=0x022BAA88, magnitude=0x0C0B
[20:57:10][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 000A 0022 0008 0024 000A 0022 0008 0024 000A 0023 0009 0022 000B 001F 000C 0024 0014 0019 0009 0023 000A 0023 0009 0023 0014 0019 0009 0023 0014 0019 0009 0023 0014 0019 0013 0019 0014 0019 0009 0023 0014 0019
[20:57:10][D][remote.pronto:240]: 0009 0023 0014 0019 0009 0023 0014 001A 0008 0024 0013 001A 0008 0024 0009 0024 0008 0024 0013 001A 0008 0024 0009 0024 0008 0024 0009 0024 0008 0024 0009 0024 0008 0024 0013 001A 0012 001A 0009 0024 0008 0024 0009 0024 0008 0024 0009 0024 0008 0024 0013 001A 0008 0024 0013 001A 0012 001A 0013 001A 0008 0024 0009 0024 0012 001A 0009 0024 0012 06C3
[20:57:12][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 000A 0023 0008 0024 000A 0022 0008 0024 000A 0022 0008 0024 000A 0022 0008 0024 0014 0019 0008 0024 000A 0022 0008 0024 0014 0019 0008 0024 0014 0018 0008 0024 0013 0019 0014 0018 0014 0018 0008 0024 0014 0019
[20:57:12][D][remote.pronto:240]: 0008 0024 0014 0019 0008 0024 0014 0018 0008 0023 0013 0019 0008 0024 000B 0022 0008 0024 0014 0018 0008 0024 000C 0021 0008 0024 000B 0021 0008 0023 000C 0021 0008 0024 0014 0018 0015 0017 000C 0021 0008 0023 000C 0021 0008 0024 000C 0021 0008 0023 0014 0018 0007 0024 0014 0018 0015 0017 0014 0018 0008 0024 000B 0022 0015 0017 000B 0022 0014 06C3
[20:57:12][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 0004 06C3
[20:57:14][D][remote.magiquest:079]: Received MagiQuest: wand_id=0x022BAA88, magnitude=0x0817
[20:57:14][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 000A 0023 0008 0023 000A 0023 0008 0023 0009 0024 0008 0023 000A 0023 0008 0023 0014 0019 0009 0023 000A 0023 0008 0023 0014 0019 0008 0024 0013 001A 0008 0024 0013 001A 0013 0019 0013 001A 0008 0024 0013 001A
[20:57:14][D][remote.pronto:240]: 0008 0024 0013 001A 0008 0024 0014 0019 0009 0023 0013 001A 0009 0023 000A 0023 0009 0023 0013 001A 0009 0023 000A 0023 0009 0023 000A 0023 0009 0023 000A 0023 0009 0023 0014 0019 0008 0024 000A 0023 0009 0023 000A 0023 0009 0023 000A 0024 0012 001A 0009 0024 0012 0019 0014 001A 0012 001A 0013 001A 0008 0024 0009 0024 0008 0024 0013 001A 0012 06C3
```

ssieb, 2023-01-02:
Pronto is similar to raw, it's not a specific protocol.  The MagiQuest messages are getting decoded and it's also printing the pronto code for the same message.  The only one that might have been missed is the one at `20:57:12`.  I don't see a problem here.

tattonchantry, 2023-01-02:
I changed what it prints out and uploaded a video of it to youtube here:
https://youtu.be/lOKVb5zB1SY
Hopefully that helps. Thanks for looking into it.

INFO Reading configuration /config/esphome/irtester.yaml...
INFO Starting log output from irtester.local using esphome API
INFO Successfully connected to irtester.local
[08:18:59][I][app:102]: ESPHome version 2022.12.3 compiled on Jan  2 2023, 08:06:34
[08:18:59][C][wifi:504]: WiFi:
[08:18:59][C][wifi:362]:   Local MAC: EC:94:CB:6B:8F:68
[08:18:59][C][wifi:363]:   SSID: [redacted]
[08:18:59][C][wifi:364]:   IP Address: 192.168.1.158
[08:18:59][C][wifi:366]:   BSSID: [redacted]
[08:18:59][C][wifi:367]:   Hostname: 'irtester'
[08:18:59][C][wifi:369]:   Signal strength: -48 dB ▂▄▆█
[08:18:59][C][wifi:373]:   Channel: 1
[08:18:59][C][wifi:374]:   Subnet: 255.255.255.0
[08:18:59][C][wifi:375]:   Gateway: 192.168.1.1
[08:18:59][C][wifi:376]:   DNS1: 192.168.1.1
[08:18:59][C][wifi:377]:   DNS2: 0.0.0.0
[08:18:59][C][logger:293]: Logger:
[08:18:59][C][logger:294]:   Level: DEBUG
[08:18:59][C][logger:295]:   Log Baud Rate: 115200
[08:18:59][C][logger:296]:   Hardware UART: UART0
[08:18:59][C][remote_base:027]: Remote Receiver Binary Sensor 'Brogan Wand All'
[08:18:59][C][remote_base:027]: Remote Receiver Binary Sensor 'Pink Wand All'
[08:18:59][C][remote_base:027]: Remote Receiver Binary Sensor 'Beni Wand All'
[08:18:59][C][remote_receiver.esp32:054]: Remote Receiver:
[08:18:59][C][remote_receiver.esp32:055]:   Pin: GPIO3
[08:18:59][C][remote_receiver.esp32:060]:   Channel: 0
[08:18:59][C][remote_receiver.esp32:061]:   RMT memory blocks: 3
[08:18:59][C][remote_receiver.esp32:062]:   Clock divider: 80
[08:18:59][C][remote_receiver.esp32:063]:   Tolerance: 25%
[08:18:59][C][remote_receiver.esp32:064]:   Filter out pulses shorter than: 50 us
[08:18:59][C][remote_receiver.esp32:065]:   Signal is done after 10000 us of no changes
[08:18:59][C][captive_portal:088]: Captive Portal:
[08:18:59][C][mdns:103]: mDNS:
[08:18:59][C][mdns:104]:   Hostname: irtester
[08:18:59][C][ota:093]: Over-The-Air Updates:
[08:18:59][C][ota:094]:   Address: irtester.local:3232
[08:18:59][C][ota:097]:   Using Password.
[08:18:59][C][api:138]: API Server:
[08:18:59][C][api:139]:   Address: irtester.local:6053
[08:18:59][C][api:141]:   Using noise encryption: YES
[08:19:01][D][binary_sensor:036]: 'Beni Wand All': Sending state ON
[08:19:01][D][binary_sensor:036]: 'Beni Wand All': Sending state OFF
[08:19:08][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 000A 0024 0007 0024 000A 0023 0008 0024 000A 0023 0008 0024 000B 0022 0008 0024 0014 0019 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 0014 0019 0008 0024 0013 001A 0013 0018 0013 001A 0008 0024 0014 0019
[08:19:08][D][remote.pronto:240]: 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 000A 0023 0013 0018 000A 0023 0008 0024 0014 0019 0013 0019 0013 001A 0008 0024 000A 0023 0013 0019 000A 0023 0008 06C3
[08:19:10][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 0008 0024 000A 0023 0009 0023 000A 0023 0008 0024 000A 0023 0009 0023 000A 0023 0013 0019 000A 0023 0009 0023 000A 0023 0013 0019 000A 0023 0013 0019 000A 0023 0013 0019 0013 001A 0013 0019 000A 0023 0013 0018
[08:19:10][D][remote.pronto:240]: 000A 0023 0013 0019 000A 0023 0013 0019 000A 0023 0013 0019 000A 0022 0007 0025 000A 0023 0013 0018 000A 0023 0007 0025 000A 0023 0007 0025 000A 0023 0007 0025 000A 0023 0013 0019 000A 0023 0007 0025 000A 0023 0007 0025 000A 0023 0008 0024 0013 001A 0007 0025 000A 0023 0013 0019 0013 001A 0013 0019 000A 0023 0008 0024 0013 001A 0007 0025 000A 06C3
[08:19:11][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 0008 0024 000A 0023 0008 0024 000A 0023 0009 0023 000A 0023 0009 0023 000A 0023 0013 0019 000A 0023 0009 0023 000A 0023 0013 0019 000A 0023 0013 0019 000A 0023 0013 0019 0013 001A 0013 0019 000A 0023 0013 0019
[08:19:11][D][remote.pronto:240]: 000A 0023 0013 0018 000A 0023 0013 0018 000A 0023 0013 0018 000A 0022 0007 0025 000A 0023 0013 0019 000A 0023 0007 0025 000A 0023 0007 0025 000A 0023 0007 0025 000A 0023 0013 0019 000A 0023 0007 0025 0009 0024 0008 0024 000A 0023 0008 0024 000A 0023 0013 0019 0013 001A 0013 0019 0013 001A 0013 0019 0009 0024 0007 0025 0013 001A 0008 0024 0013 06C3
[08:19:12][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 000A 0023 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 0013 001A 0013 0019 0013 001A 0008 0024 0013 001A
[08:19:12][D][remote.pronto:240]: 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 000A 0023 0007 0025 0013 001A 0008 0024 000A 0022 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 0013 001A 0007 0025 000A 0023 0007 0025 000A 0022 0007 0025 000A 0022 0007 0025 0013 001A 0013 0018 0013 001A 0013 0019 0013 001A 0007 0025 000A 0022 0013 0019 000A 0023 0013 06C3
[08:19:13][D][remote.pronto:238]: Received Pronto: data=0000 006D 0038 0000 000A 0023 0008 0024 000A 0022 0008 0024 000A 0023 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 000A 0023 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 0013 001A 0013 0018 0013 001A 0008 0024 0013 001A
[... cut ...]
[08:26:41][D][remote.pronto:240]: 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 0013 001A 0008 0024 000A 0023 0008 0024 0013 001A 0009 0023 000A 0023 0008 0024 000A 0023 0008 0024 000A 0023 0007 0025 0013 001A 0008 0024 000A 0022 0008 0024 000A 0022 0007 0025 000A 0022 0013 0018 000A 0022 0007 0025 0013 001A 0013 0019 0013 001A 0007 0025 000A 0023 0013 0019 000A 0023 0007 06C3
[08:30:45][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 0004 06C3
[08:30:48][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 0006 06C3
[08:30:48][D][remote.pronto:238]: Received Pronto: data=0000 006D 0001 0000 0006 06C3

closed 2023-05-20 by github-actions[bot]