esphome/issues #3995: Add peak usage-parameters Fluvius

klenaers, 2023-01-02:
### The problem

The new Fluvius-parameters aren't available from the DSMR-library.
It is also not documented in https://esphome.io/components/sensor/dsmr.html

- 1-0:1.6.0 Maximum demand – Active energy import of the running month
- 0-0:98.1.0 Maximum demand – Active energy import of the last 13 months
- 1-0:1.4.0 Current average demand - Active energy import 

[image]

This is an example-message 
/FLU5\253770234_A
0-0:96.1.4(50217)
0-0:96.1.1(3153414731313030303030323331)
0-0:1.0.0(200512145552S)
1-0:1.8.1(000000.915*kWh)
1-0:1.8.2(000001.955*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.030*kWh)
1-0:1.4.0(02.351*kW)
1-0:1.6.0(200509134558S)(02.589*kW)
0-0:98.1.0(3)(1-0:1.6.0)(1-0:1.6.0)(200501000000S)(200423192538S)(03.695*kW)(2
00401000000S)(200305122139S)(05.980*kW)(200301000000S)(200210035421W)(04.318*k
W)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:32.7.0(234.6*V)
1-0:31.7.0(000.00*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313139303333373333)
0-1:24.4.0(1)
0-1:24.2.3(200512134558S)(00112.384*m3)
0-2:24.1.0(007)
0-2:96.1.1(3853414731323334353637383930)
0-2:24.2.1(200512134558S)(00872.234*m3) 

[File specs](https://maakjemeterslim.be/rails/active_storage/disk/eyJfcmFpbHMiOnsibWVzc2FnZSI6IkJBaDdDRG9JYTJWNVNTSWhlblJ4YW14cU5XdHRhalZsWkRWbmJIaDNNak4zYmpReGFUazRkd1k2QmtWVU9oQmthWE53YjNOcGRHbHZia2tpVjJsdWJHbHVaVHNnWm1sc1pXNWhiV1U5SW1VdFRWVkRVMTlRTVY5RlpGOHhYemRmTVM1d1pHWWlPeUJtYVd4bGJtRnRaU285VlZSR0xUZ25KMlV0VFZWRFUxOVFNVjlGWkY4eFh6ZGZNUzV3WkdZR093WlVPaEZqYjI1MFpXNTBYM1I1Y0dWSkloUmhjSEJzYVdOaGRHbHZiaTl3WkdZR093WlUiLCJleHAiOiIyMDIzLTAxLTAyVDEwOjExOjA3LjI0MloiLCJwdXIiOiJibG9iX2tleSJ9fQ==--660e63b52dded4c87d934c5d838e33afb2079aa5/e-MUCS_P1_Ed_1_7_1.pdf?content_type=application%2Fpdf&disposition=inline%3B+filename%3D%22e-MUCS_P1_Ed_1_7_1.pdf%22%3B+filename%2A%3DUTF-8%27%27e-MUCS_P1_Ed_1_7_1.pdf)

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.12.8

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

klenaers, 2023-01-06:
Hi,
I tried to fix it myself and cloned the esphome & Arduino-dsmr-repository and created a branch "peak-power".

https://github.com/klenaers/arduino-dsmr/tree/peak-power
https://github.com/klenaers/esphome/tree/peak-power

following 2 values are being parsed correctly.
1-0:1.6.0 Maximum demand – Active energy import of the running month
1-0:1.4.0 Current average demand - Active energy import

The last one is to advanced for me at the moment.
0-0:98.1.0 Maximum demand – Active energy import of the last 13 months

Can somebody help me to get this merged in the esphome-code? And maybe add the 3th field?

Thanks in advance

mhendriks, 2023-01-07:
The composition of fields makes it difficult. The easiest way is to parse it as a string. Then you have the value as a string... a bit more difficult to process in HA but doable. It is a little step closer to the perfect result but in this case shareable.

ADD 
fields.h
DEFINE_FIELD(highest_peak_pwr_13mnd, String, ObisId(0, 0, 98, 1, 0), RawField);

fields.cpp
constexpr ObisId highest_peak_pwr_13mnd::id;
constexpr char highest_peak_pwr_13mnd::name_progmem[];
//constexpr const __FlashStringHelper *highest_peak_pwr_13mnd::name;

RESULT parsing to json 
"highest_peak_pwr_13mnd": {
    "value": "(1)(1-0:1.6.0)(1-0:1.6.0)(230101000000W)(221203113000W)(04.839*kW)"
  },

klenaers, 2023-01-08:
Hi @mhendriks ,
I had the same idea as you proposed and to parse it in Node-red afterwards.
The field added in my code is "energy_max_demand_import_last_13_months".

mhendriks, 2023-01-08:
Super! I think the string is easier to parse on the client side. Often certainly more CPU power :-)

cosmos1978, 2023-02-08:
> Hi @mhendriks , I had the same idea as you proposed and to parse it in Node-red afterwards. The field added in my code is "energy_max_demand_import_last_13_months".

Could you do a merge request to the esphome official code ? It seems someone else pushed your code to the official arduino-dsmr github

For the 13th month I solved it in home assistant by using a simple templated sensor like this.

  peak_current_year:
    friendly_name: Peak current year
    value_template: >-
      {{ states('sensor.peak_power_consumption_last_13_months_belgium').split(')(')[-1] | replace('*kW)',' kW') }}

mhendriks, 2023-02-16:
The PR has already been made : https://github.com/esphome/esphome/pull/4412

bwynants, 2023-02-16:
I also added a parser for the third field see https://github.com/glmnet/arduino-dsmr/pull/15

cosmos1978, 2023-02-16:
> I also added a parser for the third field see [glmnet/arduino-dsmr#15](https://github.com/glmnet/arduino-dsmr/pull/15)

Did you take into account the fact it can hold values for multiple years ?

like this.
(2)(1-0:1.6.0)(1-0:1.6.0)(230101000000W)(221207221500W)(12.948kW)(230201000000W)(230126000000W)(11.745kW)

bwynants, 2023-02-16:
> Did you take into account the fact it can hold values for multiple years ?

nope, is it always the last value we need?

bwynants, 2023-02-16:
changed it to grab the last value

cosmos1978, 2023-02-16:
> 

For me the answer is yes. I am only interested in the value of the current year. But at the moment I import the value as a string.
This gives me the possibility to do the parsing in home assistant. That way I can create 2 sensor templates one for the current year and one for last year.

So for me the most flexible way would be to just parse it as a string. and dont do any parsing on the esp since it is nearly impossible to correctly parse all values. I would take to much code changes to do that.
If you do it you would need to look at the first value that lists how many arrays there are.

in the example above there are 2

Then use that value to generate 2 new fields with a custom name like  active_energy_import_maximum_demand_last_13_months_2022 and
active_energy_import_maximum_demand_last_13_months_2023

The value for the year can be taken from the second date field 221207221500W and 230126000000W

But again it is much easier to just parse it in home assistant and leave it as a text string.

bwynants, 2023-02-16:
if you take the sensor and plot it in a graph it will plot multiple years (for the years HA was running.)

cosmos1978, 2023-02-16:
> if you take the sensor and plot it in a graph it will plot multiple years (for the years HA was running.)

You got a point there. Did not think about that. Last value would be ok then.

bwynants, 2023-07-18:
https://github.com/esphome/esphome/pull/5081 is merged and will be in ESPHome 2023.7 i think

klenaers, 2024-08-27:
ok, issue can be closed

closed 2024-08-27 by klenaers