esphome/issues #3996: Esphome Dashboard cant update, install or show logs

lweberru, 2023-01-02:
### The problem

Via command line I can do all kind of esphome command, like, compile, run, getting logs via usb and over air.
But via the web based dashboard the devices are shown and I can also edit them, but all further actions, like adopt, update, install or getting logs fails, without any error message. The log windows is opening but just stays empty and nothing further happens. I tried it with different docker images (stable, beta or dev).

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.0.dev20221227

### What platform are you using?

ESP32-IDF

### Board

nodemcu

### Component causing the issue

dashboard

### Example YAML snippet

```yaml
Independend of any specific yaml
```

### Anything in the logs that might be useful for us?

```txt
2023-01-02 20:03:38,642 INFO 101 GET /compile (172.19.0.2) 0.71ms
2023-01-02 20:03:39,076 INFO 101 GET /compile (172.19.0.2) 0.70ms
2023-01-02 20:03:39,472 INFO 101 GET /compile (172.19.0.2) 0.76ms
2023-01-02 20:13:01,405 INFO 200 GET /edit?configuration=esp32-1.yaml (172.19.0.2) 13.70ms
2023-01-02 20:13:01,458 INFO 101 GET /ace (172.19.0.2) 0.73ms
2023-01-02 20:13:06,599 INFO 101 GET /ace (172.19.0.2) 0.75ms
2023-01-02 20:16:29,979 INFO Starting dashboard web server on http://0.0.0.0:6052 and conf
2023-01-02 20:16:43,621 INFO 101 GET /compile (172.19.0.2) 1.09ms
2023-01-02 20:16:43,898 INFO 101 GET /compile (172.19.0.2) 0.82ms
2023-01-02 20:16:44,125 INFO 101 GET /compile (172.19.0.2) 0.73ms
2023-01-02 20:16:44,657 INFO 200 GET / (172.19.0.2) 5.49ms
2023-01-02 20:16:44,851 INFO 101 GET /compile (172.19.0.2) 0.72ms
2023-01-02 20:16:44,956 INFO 101 GET /compile (172.19.0.2) 0.78ms
2023-01-02 20:16:45,119 INFO 101 GET /compile (172.19.0.2) 0.70ms
2023-01-02 20:17:07,651 INFO 200 GET /static/js/esphome/c.284caccc.js (172.19.0.2) 21.21ms
2023-01-02 20:17:12,829 INFO 200 GET /static/js/esphome/c.8ffd08fb.js (172.19.0.2) 41.76ms
2023-01-02 20:17:12,877 INFO 101 GET /logs (172.19.0.2) 0.82ms
```

### Additional information

Via command line inside the container all is working fine.

Here is my docker-compose.yaml:
```
version: '3.2'

services:
  esphome:
    container_name: esphome
    image: esphome/esphome:beta
    command: dashboard --username admin --password cleared /config
    volumes:
      - /volume1/docker/Esphome/config:/config
      - /etc/localtime:/etc/localtime:ro
    restart: always
    privileged: true
    network_mode: host
```

lweberru, 2023-01-03:
I have a setup with a nginx reverse proxy in front. Now recognized, that if I access the dashboard without the reverse proxy in front, it is working. 
My nginx.conf looks like that:

```
server {
  listen 80;
  listen [::]:80;
  server_name esphome.klue.de;

  listen 443 ssl;
  listen [::]:443 ssl;

  ssl_certificate /etc/letsencrypt/live/esphome.klue.de/fullchain.pem;
  ssl_certificate_key /etc/letsencrypt/live/esphome.klue.de/privkey.pem;

  include /etc/nginx/conf.d/includes/site-defaults.conf;
  include /etc/nginx/conf.d/includes/cert_bot.conf;
  expires $expires;

  location / {
      proxy_pass  http://dsm.clue.de:6052;  
      proxy_set_header Upgrade $http_upgrade;
      proxy_set_header Connection "Upgrade";
      proxy_set_header Host $http_host;
      proxy_set_header X-Real-IP $remote_addr;
      proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
      proxy_set_header X-Forwarded-Proto $scheme;
  }
  if ($scheme != "https") {
      return 301 https://$host$request_uri;
  }
}

```

Any idea, what to change here?
My home assistant nginx conf has the same schema and is working.

Lukas

lweberru, 2023-01-03:
Ok I googled a lot and found a working solution. Not sure which setting did it:

```
upstream backend_esphome.dsm.clue.de {
      # esphome
      server dsm.clue.de:6052;
}

server {
  listen 80;
  listen [::]:80;
  server_name esphome.klue.de;

  listen 443 ssl;
  listen [::]:443 ssl;

  ssl_certificate /etc/letsencrypt/live/esphome.klue.de/fullchain.pem;
  ssl_certificate_key /etc/letsencrypt/live/esphome.klue.de/privkey.pem;

  include /etc/nginx/conf.d/includes/site-defaults.conf;
  include /etc/nginx/conf.d/includes/cert_bot.conf;
  expires $expires;

  location / {
      proxy_pass  http://backend_esphome.dsm.clue.de;
      proxy_http_version 1.1;
      proxy_set_header Upgrade $http_upgrade;
      proxy_set_header Connection "Upgrade";
      proxy_set_header Host $http_host;
      proxy_set_header X-Real-IP $remote_addr;
      proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
      proxy_set_header X-Forwarded-Proto $scheme;
      proxy_set_header X-NginX-Proxy true;
      proxy_ssl_name $host;
      proxy_ssl_server_name on;
  }
  if ($scheme != "https") {
      return 301 https://$host$request_uri;
  }
}

```

ImmortalScientist, 2023-03-07:
Thanks - I played around with mine and managed to get it working with a more minimal set of location headers :)

```
                        proxy_http_version 1.1;
                        proxy_set_header Upgrade $http_upgrade;
                        proxy_set_header Connection "upgrade";
                        proxy_set_header Host $http_host;
```

oddlama, 2023-03-17:
This is the same as #2677. I've added a comment on that issue why this is happening. https://github.com/esphome/issues/issues/2677#issuecomment-1474520959