esphome/issues #3999: waveshare_epaper 1.54inv2

jwillmer, 2023-01-02:
### The problem

I only like to update the display once and then go to deep sleep. But if I use `component.update: epaper_display` it will only do a partial refresh and does look grainy. It will also not remember the `full_update_every` property so that it will never do a full display update. When putting the update interval to: `update_interval: 2s` it works and looks good but then I don't have much control over it and it renders the image way to often - energy consumption on battery!

### Which version of ESPHome has the issue?

v2022.12.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

D1 Mini Pro

### Component causing the issue

MH-ET LIVE 1.54in E-Paper

### Example YAML snippet

```yaml
script:
  - id: process
    then:
    - ..
    - ...
    - logger.log: "Update display"
    - component.update: epaper_display
    - logger.log: "Sleep for 15 sec to let the e-paper complete the update process"
    - delay: 15s
    - if: 
        condition: 
          lambda: |-
            ...
            return isDayTime;
        then:
          - logger.log: "Sleep for 5 min during the sun is up"  
          - deep_sleep.enter:
              id: deep_sleep_1
              sleep_duration: 5min
        else:
          - logger.log: 
              format: "Sleep for %d hours during the sun is down" 
              args: [ 'id(night_sleep_hours)' ]
          - deep_sleep.enter:
              id: deep_sleep_1
              sleep_duration: !lambda return id(night_sleep_hours) * 3600000;
```

### Anything in the logs that might be useful for us?

```txt
...
[D][main:113]: Update display
[D][main:203]: Sleep for 15 sec to let the e-paper complete the update process
[VV][scheduler:026]: set_timeout(name='', timeout=15000)
[V][component:200]: Component script took a long time for an operation (0.13 s).
[V][component:201]: Components should block for at most 20-30ms.
[VV][api.service:470]: on_ping_request: PingRequest {}
[VV][api.service:043]: send_ping_response: PingResponse {}
[VV][scheduler:195]: Running timeout '' with interval=15000 last_execution=40888 (now=55894)
[D][main:134]: Sleep for 10 hours during the sun is down
[I][deep_sleep:116]: Beginning Deep Sleep
[I][deep_sleep:118]: Sleeping for 36000000000us
[E][waveshare_epaper:119]: Timeout while displaying image!
[VV][api.service:025]: send_disconnect_request: DisconnectRequest {}
```

### Additional information

[image]

unhold, 2023-01-06:
Does the display get ugly over time because it never receives a full update?

The display update counter is reset after deep sleep. The reason is that the ESP32 loses it's main memory during deep sleep. You can think of deep sleep as a scheduled reset.

A small memory in the RTC unit is preserved during deep sleep. ESPHome is built on top of Arduino for ESP32. [RTC memory can be used in Arduino this way](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/deepsleep.html), but I'm not aware of a way to do that in ESPHome. You could manually modify the Arduino/Platformio project that ESPHome generates in <your-yaml-folder>/.esphome.

jwillmer, 2023-01-06:
@unhold 
> Does the display get ugly over time because it never receives a full update?

I'm not 100% sure. It could be that it starts a full update but does not complete it. I use a very long timeout of 15sec in the hope it will complete but seams like a full update consists of multiple screen refresh and a single call wont execute this behavior -  subsequent manual calls to refresh also does not work to get the same effect.

I can life with refreshing the screen every time I wake up but as I said I'm currently not able to get a full refresh when triggering the display manually. Maybe the implementation did not account for it?

jwillmer, 2023-01-06:
I'm also not sure what value I have to set in order to trigger a full update on every call. I looked at [the code](https://github.com/esphome/esphome/blob/8e3af515c9941d4cd9535cb4388ce31c724fce18/esphome/components/waveshare_epaper/waveshare_epaper.cpp#L257) at it seams like that the first call to the display will always trigger a full update? Or maybe I have to set `full_update_every` to `1`?

unhold, 2023-01-06:
Yes, there is always a full update after the first call, I read that from the code and have experienced it on my 2.73" epaper. The behavior of 0 is not documented, but from the code, I see that 0 and 1 for `full_update_every` update the display every time. (Maybe not so great). In the YAML you gave it was commented out, so defaults to 30. Anyways, your display should not mess up.

unhold, 2023-01-06:
> @unhold
>
>
> I'm not 100% sure. It could be that it starts a full update but does not complete it. I use a very long timeout of 15sec in the hope it will complete but seams like a full update consists [...]

If I understand you correctly, it already looks messed up after a power-on-reset, where a full update should happen. But it does not happen if you do not deep-sleep.

Does your display go to all black and white and then draws the content? This is how a full update looks on my 2.13" epaper.

Is there enough time from when the display stops changing to the "Beginning Deep Sleep" message appearing in the log?

To further debug it, you could play with the values or try to create a minimal YAML that still has the problem. Another idea for troubleshooting, does the problem go away if you update it two times (by component.update) before going to sleep?

jwillmer, 2023-01-06:
Corresponding thread in discord: https://discord.com/channels/429907082951524364/1061037309778198590
My code: https://gist.github.com/jwillmer/daba74022aa523203043e24bd9c43b71
Latest test (video): 10 manual refreshes with 2 sec delay in between - also tested with 10 sec but same effect.

https://user-images.githubusercontent.com/1503577/211116400-9510fc25-8895-4476-ad5a-43964d403dac.mp4

unhold, 2023-01-08:
Instead of turning all-black then white, like it should for a full update, the display only turns slightly more then less grey for a moment. Check your wiring and power supply. If that does not help, add some delay between wake-up and  display update.

ssieb, 2023-01-08:
@unhold the issue is that if it is configured for normal updates with the update_interval, it works fine.  But not if using the component.update action.

jwillmer, 2023-01-08:
@unhold as @ssieb mentions the panel is working fine when I set the `update_interval: 2s`. So I don't think the wiring is wrong. I also acquire all my sensor data before updating the display (takes usually 5-10sec) so the display should have enough time to initialize.

unhold, 2023-01-09:
My thinking was that the first update after reset or deep sleep may not be working. Maybe because of (a) some electrical issues (charge-up) or (b) some initialization effects.

Anyways, maybe others can help.

jwillmer, 2023-01-09:
> some initialization effects

Any idea what it could be / what I could do to mitigate?

ssieb, 2023-01-09:
> My thinking was that the first update after reset or deep sleep may not be working. Maybe because of (a) some electrical issues (charge-up) or (b) some initialization effects.

That doesn't fit, because the screen updates correctly after deep sleep if using the update_interval.  Also, he tried repeatedly updating the screen as you can see in the video and it still doesn't work.

jwillmer, 2023-01-09:
It is working now. And I can't explain why. What I did:
- I removed all code except of the display and made it auto update every 2 sec to see that the full update works
- Then I added a script that updates the display manually and removed auto update - full update works
- Then I added back my sensors
- Then more and more parts of my script
- At the end I ended up with the same script as before ([diff](https://gist.github.com/jwillmer/daba74022aa523203043e24bd9c43b71/revisions)) and all is working

I applied all updates as OTA and even tried to update it by erasing/installing via usb to recreate the issue but the code continued working. On one hand I'm happy that it is working on the other I'm very unsatisfied that I can't recreate the update issue. 

Thank you very much for your support @ssieb and @unhold . If you have any guess what could have fixed it then let me know. Marking the issue as closed but will revert in case I manage to recreate it.

closed 2023-01-09 by jwillmer