esphome/issues #4001: Assumed state (template) switch not working on webserver UI

RoboMagus, 2023-01-04:
### The problem

When declaring an `assumed_state` template switch, the webserver UI will continue to show a toggle switch. 
This will remove the option of having the user select the 'on' or 'off' commands.

### Which version of ESPHome has the issue?

2022.11.x-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/a

### What platform are you using?

ESP32

### Component causing the issue

webserver

### Example YAML snippet

```yaml
switch:
  - platform: template
    name: IR Candles
    assumed_state: true
    optimistic: false
    turn_on_action:
      - script.execute: candles_on
    turn_off_action:
      - script.execute: candles_off
```

closed 2023-04-30 by jesserockz

RoboMagus, 2023-04-30:
@jesserockz, not actually closed yet untill https://github.com/esphome/esphome-webserver/pull/25 is merged.
But it seem communication has stalled there..

jesserockz, 2023-04-30:
Its auto closed when a linked PR is merged. I have merged the other one anyway now