esphome/issues #4005: Error attempting to update existing ESPHome bluetooth proxies

Gferretta, 2023-01-05:
### The problem

Hi

I have this error running ESPHOME in a RPi3+ with an SD
No can update existing BT Proxies ESP32 (Generic) over wifi or cable.
No compile at all.

ESPHome 2022.12.3
Home Assistant 2023.1.0
Supervisor 2022.12.1
Operating System 9.4
Frontend 20230104.0 - latest

[image]

Attached complete compile error log
[logs_esp32-bluetooth-proxy-059bf0_compile.txt](https://github.com/esphome/issues/files/10352776/logs_esp32-bluetooth-proxy-059bf0_compile.txt)

The error only occurs if I want to update the firmware using ESPHome.
If I load the firmware from https://esphome.github.io/bluetooth-proxies/ it does it perfectly.
I already cleaned temporary and compilation files.

Regards

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.0

### What platform are you using?

ESP32

### Board

nodemcu esp32

crackha, 2023-01-09:
Hi @Gferretta 
Same problem for me too. 
I've the same HA / ESPHome setup, with different proxy hardware but same errors.
Any solution yet?
Links to my posts FYI:

- https://community.home-assistant.io/t/bluetooth-proxies-updates-fail/513901
- https://github.com/esphome/bluetooth-proxies/issues/59
- https://github.com/home-assistant/core/issues/85320

If you get a solution I'd appreciate hearing of it
Cheers :-)

Gferretta, 2023-05-10:
No solution at all.
Not working in HA 2023.5.2   EspHome 2023.4.4
Others users had same problem reported.
🤷‍♂️

dabepp, 2023-05-19:
I also have this error...