esphome/issues #4007: rgbww light not sending temperature information to homeassistant

clinta, 2023-01-06:
### The problem

I have an RGBWW light with the temeratures for warm and cold set, but that temperature information is not in home assistant, resulting in a UI that provides separate sliders for cold and white brightness, rather than a temperature slider. If I create an CWWW light with the same configs and pins, I get all the necessary teperature information to have the nicer temperature slider interface.

```yaml
light:
  - platform: cwww
    name: office cwww
    cold_white: pin_18
    warm_white: pin_19
    warm_white_color_temperature: 2700 K
    cold_white_color_temperature: 6500 K
  - platform: rgbww
    name: Office Light
    id: rgbww_light
    red: pin_0
    blue: pin_1
    green: pin_3
    cold_white: pin_18
    warm_white: pin_19
    warm_white_color_temperature: 2700 K
    cold_white_color_temperature: 6500 K
    restore_mode: always_on
```

[image]
[image]
[image]

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.0

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

rgbww

### Example YAML snippet

```yaml
light:
  - platform: rgbww
    name: Office Light
    id: rgbww_light
    red: pin_0
    blue: pin_1
    green: pin_3
    cold_white: pin_18
    warm_white: pin_19
    warm_white_color_temperature: 2700 K
    cold_white_color_temperature: 6500 K
    restore_mode: always_on
```

### Additional information

This could be related to https://github.com/esphome/issues/issues/2321, unsure.

bkaufx, 2023-01-06:
I think this is working as intended.  WW indicates two white channels.  Have you tried the RGBCT component?  I haven't but I think it will do what you are looking for.  https://esphome.io/components/light/rgbct.html

clinta, 2023-01-06:
RGBCT would require different hardware, with a pin to control the temperature and white brightness. My hardware has a cold and warm white LED.

The reason I think this is not working as intended is because the CWWW light, which is the same thing, two white LEDs, just without the RGB component, sends it's temperature data to home assistant and gets the temperature slider interface.

bkaufx, 2023-01-06:
Good points.  RGBWW has always presented as two white channels as far back as I can remember so I just assumed it was intended that way.  I haven't thought about it much though.  I do think it is intentional but can't say for sure.  Maybe an option to present as color temp / white brightness for the RGBWW component would be helpful.

clinta, 2023-01-06:
Yeah, it has always worked this way since I made these lights, I only recently wondered if it could work differently, which prompted me to try CWWW and examine the state to see if `cold_white_color_temperature` and `warm_white_color_temperature` ever actually made it to home assistant. That's how I discovered that they do with CWWW, but not with RGBWW.

bkaufx, 2023-01-06:
One thing I've seen is someone just made two separate light entities, one for CWWW and one for RGB.

clinta, 2023-01-06:
That's what I'm considering as a workaround until this can be fixed.

vittrup88, 2023-02-28:
Try set `constant_brightness: true`
It only influences on cold and warm white channels. My UI in home assistant then changed to temperature instead of the 2 sliders.

clinta, 2023-02-28:
My understanding is that `constant_brightness` should only be required if the power supply can't drive cw and ww at full brightness.

If this does fix the issue, then it's still a bug. Because a cwww light does not require `constant_brightness`

hobbsAU, 2025-06-21:
Adding the following line to your ESPHome config seems to solve this by exposing both entity color_modes (color_temp, rgb) in Home Assistant.

```yaml  
color_interlock: true
```

```yaml
light:
  - platform: rgbww
    name: Office Light
    id: rgbww_light
    red: pin_0
    blue: pin_1
    green: pin_3
    cold_white: pin_18
    warm_white: pin_19
    warm_white_color_temperature: 2700 K
    cold_white_color_temperature: 6500 K
    restore_mode: always_on
    constant_brightness: true
    color_interlock: true
```

[image]

[image]