esphome/issues #4008: Wifi Scan Sensor Barfs on strange AP names

geiseri, 2023-01-06:
### The problem

I have the AP scan text sensor enabled on my devices but it seems there is a SSID that really annoys it.  In the unifi console it comes back as `Charlie’s_House` but for some reason I see this: 

```
2023-01-06 09:44:31.946 INFO (MainThread) [aioesphomeapi.connection] atom1 @ 10.0.11.43: Invalid protobuf message: b'\r\x8dQL\xc1\x12\xff\x017bcda75ea091: -67dB\n7bcda75ea091: -73dB\n7bcda75ea091: -76dB\nbf04dd96885c: -64dB\nFios-1G1CH: -64dB\ngeekyguests: -66dB\ngeekyguests: -71dB\nbf04dd96885c: -72dB\nbf04dd96885c: -75dB\ngeekyguests: -76dB\nCharlie\xe2\x80\x99s_House: -77dB\nAntanta: -77dB\nDesk: -78dB\nCharlie\xe2': 'utf-8' codec can't decode byte 0xe2 in position 254: 'utf-8' codec can't decode byte 0xe2 in position 254: unexpected end of data in field: TextSensorStateResponse.state
```

on the home-assistant logs.  I can't see why the `'` is coming back odd, but I am wondering if the text sensor should sanitize the text before it returns.  I am not sure if it is crashing them or what since it causes the device to disconnect and reconnect hundreds of times a minute.   Once they are disconnected I cant connect to them via the esphome component to view the logs.  

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.1

### What platform are you using?

ESP32

### Board

tinypico,m5stack-atom,esp32dev,wio_link,d1_mini

### Component causing the issue

wifi_info.scan_results

### Example YAML snippet

```yaml
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${friendly_devicename} IP Address
    mac_address:
      name: ${friendly_devicename} MAC Address
    scan_results:
      name: ${friendly_devicename} Scan Results
  - platform: restart_reason
    name: "${friendly_devicename} Wake-up Reason"
    id: wakeup_reason
```

### Anything in the logs that might be useful for us?

```txt
2023-01-06 09:44:46.415 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to atom1 @ 10.0.11.43
2023-01-06 09:44:47.498 INFO (MainThread) [aioesphomeapi.connection] atom1 @ 10.0.11.43: Invalid protobuf message: b'\r\x8dQL\xc1\x12\xff\x017bcda75ea091: -67dB\n7bcda75ea091: -73dB\n7bcda75ea091: -76dB\nbf04dd96885c: -64dB\nFios-1G1CH: -64dB\ngeekyguests: -66dB\ngeekyguests: -71dB\nbf04dd96885c: -72dB\nbf04dd96885c: -75dB\ngeekyguests: -76dB\nCharlie\xe2\x80\x99s_House: -77dB\nAntanta: -77dB\nDesk: -78dB\nCharlie\xe2': 'utf-8' codec can't decode byte 0xe2 in position 254: 'utf-8' codec can't decode byte 0xe2 in position 254: unexpected end of data in field: TextSensorStateResponse.state
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 583, in _process_loop
    msg.ParseFromString(pkt.data)
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/message.py", line 202, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/python_message.py", line 1128, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/python_message.py", line 1195, in InternalParse
    pos = field_decoder(buffer, new_pos, end, self, field_dict)
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/decoder.py", line 591, in DecodeField
    field_dict[key] = _ConvertToUnicode(buffer[pos:new_pos])
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/decoder.py", line 553, in _ConvertToUnicode
    value = str(byte_str, 'utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 254: 'utf-8' codec can't decode byte 0xe2 in position 254: unexpected end of data in field: TextSensorStateResponse.state
2023-01-06 09:44:47.499 ERROR (MainThread) [aioesphomeapi.connection] Unexpected exception in process task: 'utf-8' codec can't decode byte 0xe2 in position 254: 'utf-8' codec can't decode byte 0xe2 in position 254: unexpected end of data in field: TextSensorStateResponse.state
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 151, in _do_cleanup
    await self._process_task
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/connection.py", line 583, in _process_loop
    msg.ParseFromString(pkt.data)
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/message.py", line 202, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/python_message.py", line 1128, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/python_message.py", line 1195, in InternalParse
    pos = field_decoder(buffer, new_pos, end, self, field_dict)
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/decoder.py", line 591, in DecodeField
    field_dict[key] = _ConvertToUnicode(buffer[pos:new_pos])
  File "/usr/local/lib/python3.10/site-packages/google/protobuf/internal/decoder.py", line 553, in _ConvertToUnicode
    value = str(byte_str, 'utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 254: 'utf-8' codec can't decode byte 0xe2 in position 254: unexpected end of data in field: TextSensorStateResponse.state
2023-01-06 09:44:47.500 INFO (MainThread) [aioesphomeapi.reconnect_logic] Disconnected from ESPHome API for atom1 @ 10.0.11.43
```
```

### Additional information

I am trying to track down the owner of the accesspoint on the off chance it is just the fact there are so many AP that its truncating.

geiseri, 2023-01-06:
I am also trying to get the logs from one of the devices, but the AP doesn't show up where my desk is located.  I will post them when I can get a laptop to get the logs via serial port.

nohn, 2023-03-05:
Problem is likely the use of string.substr in https://github.com/esphome/esphome/blob/df3f13ded884682dfbcbba542d79aedb1e100b0a/esphome/components/wifi_info/wifi_info_text_sensor.h#L45.

string.substr is not aware of multibyte-charsets.

See also https://github.com/home-assistant/core/issues/88431

nohn, 2023-03-07:
@geiseri something like https://github.com/esphome/esphome/compare/dev...nohn:esphome:remove_incomplete_multibyte_chars_from_wifi_info?

geiseri, 2023-03-08:
TL;DR
All adventures that begin with Unicode end up with you being eaten by a grue...

So, UTF-8 is "interesting."  interesting the way that uncle we don't talk about anymore. Better still the 802.11 authors hate nice things and specify "it can be 0-32 octets with an unspecified or UTF8 coding." 802.11-2012 (Section 6.3.11.2.2),  Fun fact at one point you could crash an iPhone using malformed encoding for an SSID. 

What should cover most cases I think is to walk back from the end of the string until you find a value smaller than `0x80` (128), and discard anything after that.   In an ideal world, this will cover 99.9% of cases.  The other cases can be covered by using an EMP to revoke their privilege to use computers.

nohn, 2023-03-08:
From a US point of view, that 99.9% may be true, but the world uses multibyte. And maybe even inside the US. At least when most Apple users type a hyphen, that‘s not 0x2D but some multibyte thing.

geiseri, 2023-03-08:
Like you said about the hyphen my neighbor typed in `'` but the rich text input decides `U+2019` looks prettier.  

I mean according to Wikipedia [UTF-8 has good adoption on the internet](https://en.wikipedia.org/wiki/UTF-8#:~:text=UTF%2D8%20is%20the%20dominant,8%20encodings%20on%20the%20web.), but we live in a world where the United States still refuses IPv6 and the Metric system ;)  As long as they are using a UTF-* based encoding you should be okay, But in the end you will be eaten by a grue ;)

geiseri, 2023-03-08:
Oh, I apologize.  I completely misunderstood your point.  Yeah, if the user was in CN, or JP then it is a very real possibility that all SSIDs would be some sort of non-ASCII encoding.  Luckly in that case the SSID list always contains `: ` so it would stop there.  Personally, I would rather have it walk back until it gets to the last `\n` in the string just so it's not truncated halfway through.  That is more aesthetic, but it would ideally address most cases since what are the chances that the user could have a UTF-* SSID that happens to lie in the middle of byte 255?   I hear the grue coming!

nohn, 2023-03-08:
Canada, Latin America, Europe, Africa. You don't have to go to Asia to have non ASCII characters in daily usage that are multibye characters in UTF-8. Even in Europe you can find 100% multibyte alphabets like Cyrillic or Greek. It's the norm, not the exception. Chances that byte 255 is half of a multibye character is certainly below 50%, but still quite probable.

geiseri, 2023-03-08:
I might an error on the side checking if the last byte is larger than 127 (must be for it to be non-ascii) and if so, `find_last_of`('\n' ) and use that result for the substr.