esphome/issues #4011: ledc component NOT working with esp32 s2 mini

soubhik-khan, 2023-01-07:
### The problem

I’m unable to get the ledc component working with esp32 s2 mini. I tested with esp8266 d1 mini and esp32 d1 mini, it works well, but on s2 mini, there is no output on mosfet. I tried with GPIO 13, 14, 15,16 - non of the pins provide the voltage to the mosfet.

Logs on esphome web doesn't show any errors! 

### Which version of ESPHome has the issue?

v2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2023.1.1

### What platform are you using?

ESP32

### Board

Esp32 S2 mini

### Component causing the issue

ledc

### Example YAML snippet

```yaml
output:
  - platform: ledc
    pin: GPIO14
    id: pwm_output
light:
  - platform: monochromatic
    output: pwm_output
    name: "Living Room LED"  
    id: led_1
```

randybb, 2023-01-07:
Duplicate of https://github.com/esphome/issues/issues/3128 and fixed in https://github.com/esphome/esphome/pull/3820

closed 2023-01-07 by randybb

soubhik-khan, 2023-01-08:
@randybb thanks for looking into it. but the bug https://github.com/esphome/esphome/pull/3820 you linked is for c3 variant. I tried the s2 variant 2 days back with the latest esphome release and it doesn't work. Am I missing anything in my config?

ssieb, 2023-01-08:
The fix is for all the variants, but it's only in dev right now.  Either use dev or wait for the next release.

soubhik-khan, 2023-01-08:
> The fix is for all the variants, but it's only in dev right now. Either use dev or wait for the next release.

gotcha, thanks!

soubhik-khan, 2023-01-08:
btw quick question, can we install the dev version along with the stable release on homeassistant?

ssieb, 2023-01-08:
yes

xvil, 2023-01-18:
Hello,

I tried to use the ledc component from dev and it's not working for me.

I am running ESPHome 2022.12.3 and overriding the ledc component like this:

`esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: arduino
      
external_components:
  - source: github://esphome/esphome@dev
    components: [ ledc ]`

The behaviour is that changing the output value has not effect.
The fan that is hooked to the pwm output seems to always run at the same speed (looks like 100%).

Using the esp-idf, it works as expected (but then I cannot use the captive portal or the web server).

Any idea of what might be preventing to work using the arduino framework?