esphome/issues #4012: TVOC unit discrepency between home assistant and esphome

xconverge, 2023-01-07:
### The problem

Home assistant warns about an unsupported unit when using ppb for volatile organic compound 

https://github.com/home-assistant/core/issues/85170

The ccs811 sensor and a few others typically report ppb https://github.com/esphome/esphome/blob/dev/esphome/components/ccs811/sensor.py#L43

I can simply multiply by 4.5 and manually set the units to get out of this situation using this reference in section 2.3

https://www2.dmu.dk/atmosphericenvironment/expost/database/docs/ppm_conversion.pdf

My question is should I just do this and move on or is there a PR that more people could benefit from here

Other recent references of people who might want to see this:
https://github.com/home-assistant/core/issues/85266#issuecomment-1373575626
https://github.com/home-assistant/core/issues/85344

### Which version of ESPHome has the issue?

2022.12.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.0

### What platform are you using?

ESP8266

xconverge, 2023-01-08:
This is what I am using now as a workaround/fix, but if this issue had any traction we might want to make this part of the tvoc specific sensors or documentation

```
  - platform: ccs811
    eco2:
      name: "CCS811 eCO2 Value"
    tvoc:
      name: "CCS811 Total Volatile Organic Compound"
      unit_of_measurement: "µg/m³"
      filters:
        - multiply: 4.5
    address: 0x5A
    update_interval: 60s
    # https://esphome.io/components/sensor/ccs811.html#calibrating-baseline
    baseline: 0x2AC5
```

Vrobenmat, 2023-01-10:
If HA is saying that µg/m3 is on the only allowable unit of measurement for VOC, then esphome should be providing that unit out of the box, no?

xconverge, 2023-01-11:
> If HA is saying that µg/m3 is on the only allowable unit of measurement for VOC, then esphome should be providing that unit out of the box, no?

I am sure there are some esphome users who dont use HA though (not sure why but dont want to fully couple the 2 projects)

I think the sensors reading ppb is the most "accurate" according to the datasheets which is nice and clean from the esphome side. I think the conversion should be optional or documented (I am happy with the snippet that solves it for my case) and maybe even defaulted to on for VOC specific sensors though

Vrobenmat, 2023-01-11:
Ah, indeed see that the sensor's native unit is ppb, makes sense to keep providing that. Bit annoying that HA only considers µg/m3 as valid - conversion seems fuzzy as an assumption is made about molecular mass. Have implemented your workaround.

ssieb, 2023-05-12:
The device class can be changed now: https://github.com/home-assistant/core/pull/92973

jamesmyatt, 2023-05-12:
@jesserockz , Can this issue be left open until https://github.com/home-assistant/core/pull/92973 has made its way into ESPHome, please?

ssieb, 2023-05-12:
It will, but it also needs a PR to change the sensor class of the components in esphome.

jamesmyatt, 2023-05-12:
@ssieb . Exactly. But I didn't see it had already been unmarked stale. 👍

closed 2023-06-21 by jesserockz