esphome/issues #4014: esp32_camera i2c boot loop on t-camera S3

justlikeef, 2023-01-08:
### The problem

A boot loop occurs on with the t-camera s3 when the esp32_camera is configured.  I have verified that the pins match the documentation and the silkscreen on the board.

### Which version of ESPHome has the issue?

 v2023.1.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.1

### What platform are you using?

ESP32

### Board

ESP32-S3-WROOM-1

### Component causing the issue

esp32_camera

### Example YAML snippet

```yaml
esphome:
  name: doorbell-dev

esp32:
  board: esp32s3box

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "W7n180Mr0kg2Gd/LzaPQDc1fpLDTk9NspTnTrFPsV6s="

ota:
  password: "32444409df9426d8181f139b95c75497"

wifi:
  ssid: myssid
  password: "mypass"

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.0.219
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Doorbell-Dev Fallback Hotspot"
    password: "QjhVKvHNbf4G"

captive_portal:

web_server:
  include_internal: true

time:
  - platform: homeassistant
    id: homeassistant_time

i2c:
  sda: GPIO07
  scl: GPIO06

font:
  - file: "fonts/times.ttf"
    id: tnr1
    size: 20
  - file: "fonts/times.ttf"
    id: tnr2
    size: 40

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    rotation: 180
    address: 0x3C
    id: doorbell_display
    pages:
    # Name page
      - id: name_page
        lambda: |-
          it.print(34, 0, id(tnr1), "Peter");
          it.print(29, 20, id(tnr1), "Marcus");
          it.print(8, 43, id(tnr1), "Havard Olai");
    # Ding Dong page
      - id: ding_page
        lambda: |-
            it.print(64, 0, id(tnr2), TextAlign::TOP_CENTER, "DING");
            it.print(64, 64, id(tnr2), TextAlign::BASELINE_CENTER, "DONG");
    # Welcome in page
      - id: come_in_page
        lambda: |-
            it.print(64, 0, id(tnr2), TextAlign::TOP_CENTER, "KOM");
            it.print(64, 64, id(tnr2), TextAlign::BASELINE_CENTER, "INN");
text_sensor:
  - platform: template
    name: uptime_human_readable
    id: uptime_human_readable
    icon: mdi:clock-start
    update_interval: 60s

sensor:
  - platform: uptime
    name: uptime_sensor
    id: uptime_sensor
    update_interval: 15s
    internal: true
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human_readable
            state: !lambda |-
                      int seconds = round(id(uptime_sensor).raw_state);
                      int days = seconds / (24 * 3600);
                      seconds = seconds % (24 * 3600);
                      int hours = seconds / 3600;
                      seconds = seconds % 3600;
                      int minutes = seconds /  60;
                      seconds = seconds % 60;
                      return (
                        (days ? to_string(days)+":" : "00:") +
                        (hours ? to_string(hours)+":" : "00:") +
                        (minutes ? to_string(minutes)+":" : "00:") +
                        (to_string(seconds))
                      ).c_str();
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 10s

binary_sensor:
  - platform: gpio
    pin: GPIO17
    name: PIR
    device_class: motion

  - platform: gpio
    pin:
      number: GPIO16
      mode: INPUT_PULLUP
      inverted: True
    name: Doorbell Button
    # On press show ding dong page and activate doorbell script in Home Assistant
    on_press:
      then:
        - display.page.show: ding_page
        - component.update: doorbell_display
        - delay: 2s
        - display.page.show: name_page
        - component.update: doorbell_display

esp32_camera:
  external_clock:
    pin: GPIO38
    frequency: 10MHz
  i2c_pins:
    sda: GPIO05
    scl: GPIO04
  data_pins: [GPIO14, GPIO47, GPIO48, GPIO21, GPIO13, GPIO11, GPIO10, GPIO09]
  vsync_pin: GPIO8
  href_pin: GPIO18
  pixel_clock_pin: GPIO12
  vertical_flip: false
  horizontal_mirror: false
  name: Camera
```

### Anything in the logs that might be useful for us?

```txt
[I][logger:258]: Log initialized
[C][ota:469]: There have been 2 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:175]: Performing I2C bus recovery
[D][binary_sensor:034]: 'PIR': Sending initial state ON
[D][binary_sensor:034]: 'Doorbell Button': Sending initial state OFF
E (10458) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10458) task_wdt:  - loopTask (CPU 1)
E (10458) task_wdt: Tasks currently running:
E (10458) task_wdt: CPU 0: IDLE
E (10458) task_wdt: CPU 1: IDLE
E (10458) task_wdt: Aborting.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037731c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
[I][logger:258]: Log initialized
```

justlikeef, 2023-01-08:
May be a regression of this one, but I don't remember how to compile without re-generating code...

https://github.com/esphome/issues/issues/405

Jason2866, 2023-01-15:
Have you checked if the cam component has this commit https://github.com/espressif/esp32-camera/commit/77d71b03f3ac47803fd66df7387bceaf27d6969d
without it will crash when a i2c device is configured and used.

rici44, 2023-03-02:
> May be a regression of this one, but I don't remember how to compile without re-generating code...
> 
> 
> 
> https://github.com/esphome/issues/issues/405

Did you manage this board to work with the camera? My is not restarting but camera is not wotkíng. I have 2 boards.

martijnbuts, 2023-03-10:
The ESP32-S3 cam works with the following config and latest ESPhome beta 
esphome:
  name: esp32-s3-cam
  friendly_name: ESP32-S3 cam

esp32:
  board: lolin_s3
  framework:
    type: arduino
    version: 2.0.7
    platform_version: 6.0.1

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-S3-Cam Fallback Hotspot"
    password: 

captive_portal:

psram: 

web_server:
  version: 2

sensor:
  - platform: template
    name: Free psram
    lambda: return ESP.getFreePsram();
    icon: "mdi:memory"
    entity_category: diagnostic
    state_class: measurement
    unit_of_measurement: "b"
    update_interval: 60s

  - platform: template
    name: Free HEAP
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
    icon: "mdi:memory"
    entity_category: diagnostic
    state_class: measurement
    unit_of_measurement: "b"
    update_interval: 60s
  - platform: uptime
    name: Uptime

esp32_camera:
  external_clock:
    pin: GPIO38
    frequency: 10MHz
  i2c_pins:
    sda: GPIO05
    scl: GPIO04
  data_pins: [GPIO14, GPIO47, GPIO48, GPIO21, GPIO13, GPIO11, GPIO10, GPIO09]
  vsync_pin: GPIO8
  href_pin: GPIO18
  pixel_clock_pin: GPIO12
  vertical_flip: false
  horizontal_mirror: false
  name: Camera

maxpersg, 2023-03-11:
> The ESP32-S3 cam works with the config below and latest ESPhome beta
> 
> esphome: name: esp32-s3-cam friendly_name: ESP32-S3 cam
> 
> esp32: board: esp32s3box framework: type: arduino version: 2.0.7 platform_version: 6.0.1
> 
> logger:
> 
> api: encryption: key:
> 
> ota: password:
> 
> wifi: ssid: !secret wifi_ssid password: !secret wifi_password
> 
> ap: ssid: "Esp32-S3-Cam Fallback Hotspot" password:
> 
> captive_portal:
> 
> psram:
> 
> web_server: version: 2
> 
> sensor:
> 
> * platform: template
>   name: Free psram
>   lambda: return ESP.getFreePsram();
>   icon: "mdi:memory"
>   entity_category: diagnostic
>   state_class: measurement
>   unit_of_measurement: "b"
>   update_interval: 60s
> * platform: template
>   name: Free HEAP
>   lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
>   icon: "mdi:memory"
>   entity_category: diagnostic
>   state_class: measurement
>   unit_of_measurement: "b"
>   update_interval: 60s
> * platform: uptime
>   name: Uptime
> 
> esp32_camera: external_clock: pin: GPIO38 frequency: 10MHz i2c_pins: sda: GPIO05 scl: GPIO04 data_pins: [GPIO14, GPIO47, GPIO48, GPIO21, GPIO13, GPIO11, GPIO10, GPIO09] vsync_pin: GPIO8 href_pin: GPIO18 pixel_clock_pin: GPIO12 vertical_flip: false horizontal_mirror: false name: Camera

It worked for the first instance only. after subsequent resets, i went back to restarting

rici44, 2023-03-11:
Thx for update

joopdo, 2023-06-02:
@mrdis solved my issues at: https://github.com/esphome/issues/issues/4168

pmannk, 2023-07-11:
I appreciate this is an old thread, but have you tried this with your GPIO16 button commented out?
Specifically this bit:
```
  - platform: gpio
    pin:
      number: GPIO16
      mode: INPUT_PULLUP
      inverted: True
    name: Doorbell Button
    # On press show ding dong page and activate doorbell script in Home Assistant
    on_press:
      then:
        - display.page.show: ding_page
        - component.update: doorbell_display
        - delay: 2s
        - display.page.show: name_page
        - component.update: doorbell_display
```

I updated a camera board this week from a very old ESPHome 2022.6 release to 2023.6 and it was triggering the watchdog timer. Like you, I was using GPIO16.
Digging through the [pin notes](https://github.com/raphaelbs/esp32-cam-ai-thinker/blob/master/docs/esp32cam-pin-notes.md) for my board (ai-thinker) I found a note that GPIO16 is connected to the PSRAM.
As soon as I commented out that button the board would boot and run fine. I've since switched to a different GPIO.