esphome/issues #4015: Wrong switch/button combo leads to start-up exception

jov58, 2023-01-08:
### The problem

I have a [LilyGO TTGO TO ESP8266 OLED](https://templates.blakadder.com/lilygo_ESP8266_OLED_SH1106.html) board which has three physical buttons (momentary push-down switches). I wanted to use one of these as a switch toggle, This switch should be present in Home Assistant. I didn't know right away how to do this, so after some trial-and-error I ended up with this code (full yaml example [here](https://github.com/jov58/esphome/blob/main/switch-button-combo/lab-test.yaml)):
```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12
      inverted: true
      mode:
        input: true
        pullup: true
    id: btn1
    on_press:
      - switch.toggle: sw1

switch:
  - platform: gpio
    pin:
      number: GPIO12
      inverted: true
      mode:
        input: true
        pullup: true
    id: sw1
    name: $device_name Switch
```
That didn't work out. It resulted in an exception/reboot loop on start-up (full log output [here](https://github.com/jov58/esphome/blob/main/switch-button-combo/lab-test-error-log)):
```
Showing logs:
[14:08:17][I][logger:258]: Log initialized
[14:08:17][I][app:029]: Running through setup()...
[14:08:17][I][i2c.arduino:175]: Performing I2C bus recovery
[14:08:17][D][binary_sensor:034]: 'btn1': Sending initial state OFF
[14:08:17][C][switch.gpio:011]: Setting up GPIO Switch 'Lab Control Switch'...
[14:08:17][D][switch:017]: 'Lab Control Switch' Turning OFF.
[14:08:17][D][switch:056]: 'Lab Control Switch': Sending state OFF
[14:08:17]
[14:08:17]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[14:08:17]
[14:08:17]Exception (29):
[14:08:17]epc1=0x40209443 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[14:08:17]
[14:08:17]>>>stack>>>
```

### Which version of ESPHome has the issue?

2022.12 (esphome/esphome-armv7:latest)

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Board

LilyGO TTGO TO ESP8266 OLED

### Component causing the issue

switch and/or button

### Example YAML snippet

```yaml
This solved the problem:

switch:
  - platform: template
    optimistic: true
    id: sw1
    name: $device_name Switch
```
```

### Anything in the logs that might be useful for us?

```txt
See https://github.com/jov58/esphome/blob/main/switch-button-combo/lab-test-error-log
```

### Additional information

I obviously programmed this wrong the first time, but I still would like to report it here because it leads to an almost unresponsive device. It took me a while to figure this out and only discovered the problem after flashing with esphome-flasher and examining the log output. It would be nice if such a programming mistake could be captured somehow by the compiler.

ssieb, 2023-01-08:
I can't even reproduce this issue, maybe it needs more of your config.  And without a decoded stack trace, I can't tell what actually crashed.  But the "compiler" can't know that you've coded it wrong.

closed 2023-01-08 by ssieb

jov58, 2023-01-09:
As mentioned above, the complete [config is here](https://github.com/jov58/esphome/blob/main/switch-button-combo/lab-test.yaml) and the [stack trace is here](https://github.com/jov58/esphome/blob/main/switch-button-combo/lab-test-error-log).

I can reproduce it as often as I want. Only tried on this board though, perhaps it's something odd on the board that causes this.

About the compiler: isn't it odd to have an input binary sensor and output switch defined on the same GPIO pin?