esphome/issues #4016: device_class preciptitation is missing

hmorkemo, 2023-01-08:
### The problem

precipitation is listed as a device class in https://www.home-assistant.io/integrations/sensor/ ,
but specifying a sensor device_class: precipitation gives an error message when compiling an esphome device.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.1.1

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

hydreon_rgxx

### Example YAML snippet

```yaml
sensor:
  - platform: hydreon_rgxx
    model: "RG_15"
    update_interval: 60s
    acc:
      name: "Regn"
      device_class: precipitation
```

ssieb, 2023-01-08:
Already done, will be in the next release.
https://github.com/esphome/esphome/pull/4219

closed 2023-01-08 by ssieb

javilopezalarcon, 2023-01-15:
in platform: pulse_counter , integration and total_daily_energy the device_class: precipitation doesn't work either

ssieb, 2023-01-15:
It will in the next release or if you use dev.

javilopezalarcon, 2023-01-15:
> It will in the next release or if you use dev.

perfect, thank you very much, I look forward to the next update 👍🏻 it was to expand the issue that apparently failed in all sensors