esphome/issues #4021: Automations with multiple remote receivers - too many candidates found

nixmaster2000, 2023-01-10:
### The problem

I'm trying to build a multi-zone AV controller with three IR receivers, one for each room, with automations under each receiver. Which receiver receives the signal determines which TV performs the action. (Have not used binary sensor in this project due to the large number of sensors required and not enough memory).

I'm unable to validate the YAML with the error "Too many candidates found for 'receiver_id'". YAML validates successfully if I remove two receivers and leave only one.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

Olimex ESP32-PoE-ISO

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
remote_receiver:
  - id: lounge_receiver
    pin:
      number: 36
      inverted: true
    tolerance: 55%
    on_nec:
      - if:
          condition:
            and:
              - lambda: return x.address == 0xBF00;
              - lambda: return x.command == 0xF20D; # Power button
          then:
            if:
              condition:
                - text_sensor.state:
                    id: lounge_activity
                    state: "Off"
              then:
                - switch.turn_off: av_power
              else:
                - script.execute: power_off_lounge

  - id: kitchen_receiver
    pin:
      number: 13
      inverted: true
    tolerance: 55%
    on_nec:
      - if:
          condition:
            and:
              - lambda: return x.address == 0xBF00;
              - lambda: return x.command == 0xF20D; # Power button
          then:
            if:
              condition:
                - text_sensor.state:
                    id: kitchen_activity
                    state: "Off"
              then:
                - switch.turn_off: av_power
              else:
                - script.execute: power_off_kitchen

  - id: bedroom_receiver
    pin:
      number: 2
      inverted: true
    tolerance: 55%
    on_nec:
      - if:
          condition:
            and:
              - lambda: return x.address == 0xBF00;
              - lambda: return x.command == 0xF20D; # Power button
          then:
            if:
              condition:
                - text_sensor.state:
                    id: bedroom_activity
                    state: "Off"
              then:
                - switch.turn_off: av_power
              else:
                - script.execute: power_off_bedroom
```

### Anything in the logs that might be useful for us?

```txt
Failed config

remote_receiver: [source av-control.yaml:447]
  - id: lounge_receiver
    pin:
      number: 36
      inverted: True
      mode:
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
      drive_strength: 20.0
    tolerance: 55
    on_nec:

      Too many candidates found for 'receiver_id' type 'remote_base::RemoteReceiverBase' Some are 'lounge_receiver', 'kitchen_receiver', 'bedroom_receiver'.
      - then:
          - if:
              condition:
                and:
                  - lambda: !lambda |-
                      return x.address == 0xBF00;
                  - lambda: !lambda |-
                      return x.command == 0xF20D;
              then:
                - if:
                    condition:
                      and:
                        - text_sensor.state:
                            id: lounge_activity
                            state: Off
                    then:
                      - switch.turn_off:
                          id: av_power
                    else:
                      - script.execute:
                          id: power_off_lounge
    dump: []
    buffer_size: 10000
    filter: 50us
    idle: 10ms
    memory_blocks: 3
    rmt_channel: 2
  - id: kitchen_receiver
    pin:
      number: 13
      inverted: True
      mode:
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
      drive_strength: 20.0
    tolerance: 55
    on_nec:

      Too many candidates found for 'receiver_id' type 'remote_base::RemoteReceiverBase' Some are 'lounge_receiver', 'kitchen_receiver', 'bedroom_receiver'.
      - then:
          - if:
              condition:
                and:
                  - lambda: !lambda |-
                      return x.address == 0xBF00;
                  - lambda: !lambda |-
                      return x.command == 0xF20D;
              then:
                - if:
                    condition:
                      and:
                        - text_sensor.state:
                            id: kitchen_activity
                            state: Off
                    then:
                      - switch.turn_off:
                          id: av_power
                    else:
                      - script.execute:
                          id: power_off_kitchen
    dump: []
    buffer_size: 10000
    filter: 50us
    idle: 10ms
    memory_blocks: 3
    rmt_channel: 2
  - id: bedroom_receiver
    pin:
      number: 2
      inverted: True
      mode:
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
      drive_strength: 20.0
    tolerance: 55
    on_nec:

      Too many candidates found for 'receiver_id' type 'remote_base::RemoteReceiverBase' Some are 'lounge_receiver', 'kitchen_receiver', 'bedroom_receiver'.
      - then:
          - if:
              condition:
                and:
                  - lambda: !lambda |-
                      return x.address == 0xBF00;
                  - lambda: !lambda |-
                      return x.command == 0xF20D;
              then:
                - if:
                    condition:
                      and:
                        - text_sensor.state:
                            id: bedroom_activity
                            state: Off
                    then:
                      - switch.turn_off:
                          id: av_power
                    else:
                      - script.execute:
                          id: power_off_bedroom
    dump: []
    buffer_size: 10000
    filter: 50us
    idle: 10ms
    memory_blocks: 3
    rmt_channel: 2
```

ale1800, 2023-02-21:
same problem, hope it will be fixed soon

ssieb, 2023-02-21:
As it says: `"Too many candidates found for 'receiver_id'"`, you need to add the undocumented `receiver_id:` parameter.

nixmaster2000, 2023-02-21:
If I replace

`remote_receiver:
  - id: lounge_receiver
    pin:
      number: 36
      inverted: true`

with
`remote_receiver:
  - receiver_id: lounge_receiver
    pin:
      number: 36
      inverted: true`

I get
` [receiver_id] is an invalid option for [0]. Please check the indentation.`

Can you tell me where the `receiver_id:` parameter should go?

ssieb, 2023-02-21:
In the automation.
```yaml
on_nec:
  receiver_id: ...
```

ale1800, 2023-02-22:
Ok, so this is my configuration and it throws me these errors:

livingroom.yaml: In lambda function:
livingroom.yaml:85:25: error: 'data' was not declared in this scope
livingroom.yaml: In lambda function:
livingroom.yaml:86:25: error: 'data' was not declared in this scope
livingroom.yaml: In lambda function:
livingroom.yaml:87:25: error: 'data' was not declared in this scope
livingroom.yaml: In lambda function:
livingroom.yaml:88:25: error: 'data' was not declared in this scope

I don't understand what should I change

```
remote_receiver:
  - id: RF_RX
    pin: 
      number: GPIO5
      inverted: True
    dump: 
      - rc_switch
    tolerance: 35%
    filter: 200us
    idle: 4ms
    buffer_size: 10kb
    on_rc_switch:
      receiver_id: RF_RX
      then:
        - homeassistant.event:
            event: esphome.rf_code_received
            data:
              sync: !lambda 'return format_hex(data.sync);'
              low: !lambda 'return format_hex(data.low);'
              high: !lambda 'return format_hex(data.high);'
              code: !lambda 'return format_hex(data.code);'      
  - id: IR_RX
    pin: 
      number: GPIO19
      inverted: True
    dump: all
```

nixmaster2000, 2023-02-22:
After some messing around this has successfully compiled for me. Not tested on actual hardware yet.
````yaml
remote_receiver:
  - id: lounge_receiver
    pin:
      number: 36
      inverted: true
    tolerance: 55%
    on_nec:
      receiver_id: lounge_receiver
      then:
        - if:
            condition:
              and:
                - lambda: return x.address == 0xBF00;
                - lambda: return x.command == 0xF20D; # Power button
            then:
              if:
                condition:
                  - text_sensor.state:
                      id: lounge_activity
                      state: "Off"
                then:
                  - switch.turn_off: av_power
                else:
                  - script.execute: power_off_lounge

  - id: kitchen_receiver
    pin:
      number: 13
      inverted: true
    tolerance: 55%
    on_nec:
      receiver_id: kitchen_receiver
      then:
        - if:
            condition:
              and:
                - lambda: return x.address == 0xBF00;
                - lambda: return x.command == 0xF20D; # Power button
            then:
              if:
                condition:
                  - text_sensor.state:
                      id: kitchen_activity
                      state: "Off"
                then:
                  - switch.turn_off: av_power
                else:
                  - script.execute: power_off_kitchen

  - id: bedroom_receiver
    pin:
      number: 2
      inverted: true
    tolerance: 55%
    on_nec:
      receiver_id: bedroom_receiver
      then:
        - if:
            condition:
              and:
                - lambda: return x.address == 0xBF00;
                - lambda: return x.command == 0xF20D; # Power button
            then:
              if:
                condition:
                  - text_sensor.state:
                      id: bedroom_activity
                      state: "Off"
                then:
                  - switch.turn_off: av_power
                else:
                  - script.execute: power_off_bedroom
````

ssieb, 2023-02-22:
I just realized how silly it is that it requires the `receiver_id` to be set when it's already within a specific receiver.  I wonder how that can be fixed.

jesserockz, 2023-02-22:
Please test the linked PR with:

```yaml
external_components:
  - source: github://pr#4477
    components: [remote_base, remote_receiver]
    refresh: 0s
```

closed 2023-02-23 by jesserockz

ale1800, 2023-02-23:
Hi, sorry but I can't succeed at configuring it. I also updated ESPHome

It throws me the same errors:

```
In lambda function:
/config/esphome/soggiorno-remote.yaml:84:25: error: 'data' was not declared in this scope
             sync: !lambda 'return format_hex(data.sync);'
                         ^~~~
/config/esphome/soggiorno-remote.yaml:84:25: note: suggested alternative: 'atan'
             sync: !lambda 'return format_hex(data.sync);'
                         ^~~~
                         atan
/config/esphome/soggiorno-remote.yaml: In lambda function:
/config/esphome/soggiorno-remote.yaml:85:25: error: 'data' was not declared in this scope
             low: !lambda 'return format_hex(data.low);'
                         ^~~~
/config/esphome/soggiorno-remote.yaml:85:25: note: suggested alternative: 'atan'
             low: !lambda 'return format_hex(data.low);'
                         ^~~~
                         atan
/config/esphome/soggiorno-remote.yaml: In lambda function:
/config/esphome/soggiorno-remote.yaml:86:25: error: 'data' was not declared in this scope
             high: !lambda 'return format_hex(data.high);'
                         ^~~~
/config/esphome/soggiorno-remote.yaml:86:25: note: suggested alternative: 'atan'
             high: !lambda 'return format_hex(data.high);'
                         ^~~~
                         atan
/config/esphome/soggiorno-remote.yaml: In lambda function:
/config/esphome/soggiorno-remote.yaml:87:25: error: 'data' was not declared in this scope
             code: !lambda 'return format_hex(data.code);'
                         ^~~~
/config/esphome/soggiorno-remote.yaml:87:25: note: suggested alternative: 'atan'
             code: !lambda 'return format_hex(data.code);'
                         ^~~~
                         atan
```

This is the configuration
```
remote_receiver:
  - id: RF_RX
    pin: 
      number: GPIO5
      inverted: True
    dump: 
      - rc_switch
    tolerance: 35%
    filter: 200us
    idle: 4ms
    buffer_size: 10kb
    on_rc_switch:
        - homeassistant.event:
            event: esphome.rf_code_received
            data:
              sync: !lambda 'return format_hex(data.sync);'
              low: !lambda 'return format_hex(data.low);'
              high: !lambda 'return format_hex(data.high);'
              code: !lambda 'return format_hex(data.code);'      
  - id: IR_RX
    pin: 
      number: GPIO19
      inverted: True
    dump: all

external_components:
  - source: github://pr#4477
    components: [remote_base, remote_receiver]
    refresh: 0s
```

ssieb, 2023-02-23:
I don't know why you think that should work.  Where did you find an example like that?  Check the [docs](https://esphome.io/components/remote_receiver.html#configuration-variables) again for that trigger.

ale1800, 2023-02-23:
Sorry, I thought it was fixed
I found the example in the RF bridge component https://esphome.io/components/rf_bridge.html and I thought that I could re-use the "event" part in my configuration. Obviously not on_code_received but on_rc_switch

ssieb, 2023-02-23:
No, the rf_bridge parameters are different.  You have to use the docs for the component you're using.