esphome/issues #4025: Can't compile 2022.12.3 under windows

silentgecko, 2023-01-11:
### The problem

I'm not able to compile esphome with 2022.12.3 anymore, previous version (2022.11.1) worked.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

lolin_d32

### Example YAML snippet

```yaml
substitutions:
    name: "Banana 1"
    update_interval: 360s

esphome:
  name: banana_1
  platform: ESP32
  board: lolin_d32
  on_shutdown:
    then:
      - switch.turn_off: spower

logger:

api:
  password: "xxxx"

ota:
  password: "xxx"

wifi:
  ssid: "xxx"
  password: "xxx"
  
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.55
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$name Fallback Hotspot"
    password: "e388ZTjA9god"

captive_portal:
bluetooth_proxy:
  active: true

esp32_ble_tracker:
  scan_parameters:
    interval: 500ms
    window: 500ms
    active: true

switch:
  - platform: gpio
[... 298 lines cut ...]
In file included from src/esphome/core/automation.h:5,
                 from src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.h:4,
                 from src/esphome/components/esp32_ble_client/ble_client_base.h:5,
                 from src/esphome/components/bluetooth_proxy/bluetooth_connection.h:5,
                 from src/esphome/components/bluetooth_proxy/bluetooth_connection.cpp:1:
src/esphome/core/helpers.h:14:10: fatal error: esp_heap_caps.h: No such file or directory

***********************************************************************
* Looking for esp_heap_caps.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_heap_caps.h"
* Web  > https://registry.platformio.org/search?q=header:esp_heap_caps.h
*
***********************************************************************

 #include <esp_heap_caps.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\banana_1\src\esphome\components\bluetooth_proxy\bluetooth_connection.cpp.o] Error 1
============================================================================================================================== [FAILED] Took 1.98 seconds ==============================================================================================================================
```

### Additional information

Tried as admin cmd in windows with global esphome install as well as user esphome install in userdir, same result

yousaf465, 2023-01-11:
why not try WSL ubuntu for compiling?

silentgecko, 2023-02-10:
Sure, wsl ubuntu works fine for me now, but what is the sense in supporting windows then anyway, when the answer will be: use wsl?

rekcodocker, 2023-10-30:
Same error here. 
I am compiling on a Raspberry PI with homeassistant, latest version.

It reads like really basic components are missing in the compile environment.