esphome/issues #4028: Component pzemac - Error message in log when resetting energy

tibouh, 2023-01-12:
### The problem

Hi,

I have 3 pzem 004t v3 connected on a wemos d1 mini (on same pins).
The data reporting is working but I have an issue when clicking on any of the 3 buttons in HA to reset a specific pzem energy.
No matter the button I click, there is an error logged and the energy does not seem to be reset. See 3 lines at 13:15:12 in log below.

### Which version of ESPHome has the issue?

2022.12.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.2

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

pzemac

### Example YAML snippet

```yaml
esphome:
  name: main-power-meter

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: <secret>

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: <secret>
    password: <secret>

captive_portal:

uart:
  - id: uart1
    rx_pin: D5
    tx_pin: D6
    baud_rate: 9600

modbus:
  send_wait_time: 500ms
  id: mod_bus_pzem
  uart_id: uart1

sensor:
  - platform: pzemac
    id: pzemac_1
    current:
      name: "Phase 1 Current"
    voltage:
      name: "Phase 1 Voltage"
    energy:
      name: "Phase 1 Energy"
      filters:
        multiply: 0.57735
    power:
      name: "Phase 1 Power"
      filters:
        multiply: 0.57735
    frequency:
      name: "Phase 1 Frequency"
    power_factor:
      name: "Phase 1 Power Factor"
    update_interval: 60s
    address: 0x10
    modbus_id: mod_bus_pzem
  - platform: pzemac
    id: pzemac_2
    current:
      name: "Phase 2 Current"
    voltage:
      name: "Phase 2 Voltage"
    energy:
      name: "Phase 2 Energy"
      filters:
        multiply: 0.57735
    power:
      name: "Phase 2 Power"
      filters:
        multiply: 0.57735
    frequency:
      name: "Phase 2 Frequency"
    power_factor:
      name: "Phase 2 Power Factor"
    update_interval: 60s
    address: 0x11
    modbus_id: mod_bus_pzem
  - platform: pzemac
    id: pzemac_3
    current:
      name: "Phase 3 Current"
    voltage:
      name: "Phase 3 Voltage"
    energy:
      name: "Phase 3 Energy"
      filters:
        multiply: 0.57735
    power:
      name: "Phase 3 Power"
      filters:
        multiply: 0.57735
    frequency:
      name: "Phase 3 Frequency"
    power_factor:
      name: "Phase 3 Power Factor"
    update_interval: 60s
    address: 0x12
    modbus_id: mod_bus_pzem

button:
  - platform: template
    name: "Phase 1 kWh reset"
    on_press:
      - pzemac.reset_energy: pzemac_1
  - platform: template
    name: "Phase 2 kWh reset"
    on_press:
      - pzemac.reset_energy: pzemac_2
  - platform: template
    name: "Phase 3 kWh reset"
    on_press:
      - pzemac.reset_energy: pzemac_3
```

### Anything in the logs that might be useful for us?

```txt
[13:14:30][D][pzemac:049]: PZEM AC: V=230.3 V, I=0.680 A, P=146.8 W, E=27512.0 Wh, F=50.0 Hz, PF=0.94
[13:14:30][D][sensor:126]: 'Phase 1 Voltage': Sending state 230.30000 V with 1 decimals of accuracy
[13:14:30][D][sensor:126]: 'Phase 1 Current': Sending state 0.68000 A with 3 decimals of accuracy
[13:14:30][D][sensor:126]: 'Phase 1 Power': Sending state 84.75498 W with 2 decimals of accuracy
[13:14:30][D][sensor:126]: 'Phase 1 Energy': Sending state 15884.05371 Wh with 0 decimals of accuracy
[13:14:30][D][sensor:126]: 'Phase 1 Frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[13:14:30][D][sensor:126]: 'Phase 1 Power Factor': Sending state 0.94000  with 2 decimals of accuracy
[13:15:12][D][button:013]: 'Phase 1 kWh reset' Pressed.
[13:15:12][D][modbus:077]: Modbus user-defined function 42 found
[13:15:12][W][pzemac:015]: Invalid size for PZEM AC!
[13:15:18][D][pzemac:049]: PZEM AC: V=230.0 V, I=1.546 A, P=304.0 W, E=35381.0 Wh, F=50.0 Hz, PF=0.85
[13:15:18][D][sensor:126]: 'Phase 3 Voltage': Sending state 230.00000 V with 1 decimals of accuracy
[13:15:18][D][sensor:126]: 'Phase 3 Current': Sending state 1.54600 A with 3 decimals of accuracy
[13:15:18][D][sensor:126]: 'Phase 3 Power': Sending state 175.51440 W with 2 decimals of accuracy
[13:15:18][D][sensor:126]: 'Phase 3 Energy': Sending state 20427.22070 Wh with 0 decimals of accuracy
[13:15:18][D][sensor:126]: 'Phase 3 Frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[13:15:18][D][sensor:126]: 'Phase 3 Power Factor': Sending state 0.85000  with 2 decimals of accuracy
```

MusienkoAnton, 2023-04-20:
I have the same error, to0.
Home Assistant:
Supervisor 2023.04.0
Operating System 10.0
Інтерфейс: 20230411.1 - latest
on Raspberry
Esphome 2023.4.0

cjcr, 2023-05-10:
Confirmed. Same here.

tibouh, 2024-07-01:
This issue is still active.

As per documentation, the correct reply for command 0x42 is 4 bytes long.
So in code, instead of erroring if reply is less than 20 bytes (actually should be less than 21 bytes if correct reply from command 0x04), the code should exclude the case when reply if 4 bytes long also.

I can look into creating a pull request.

puravida13, 2024-09-10:
Is this issue resolved? Same here

keniji, 2024-12-31:
It's 2024.12.31 now and this issue seems still exist...