esphome/issues #4032: restore_from_flash bricks ESP

floriandotorg, 2023-01-13:
### The problem

Hello together, I noticed that when you use `restore_from_flash: true` it’s going to brick your device.

Steps to reproduce:
1. Connect fresh ESP via USB. Do the initial setup, paste YAML provided and click Install
2. Check that the Wi-Fi connection works
3. Click Install again
4. Notice that installation is stuck at `INFO Starting log output from /dev/ttyUSB1 with baud rate 115200`and Wi-Fi connection is not attempted

As far as I can tell, this procedure bricks the ESP. I found no way to revive them. Even a fresh setup with the base YAML could not make it work again.

* You might need to wait a few minutes until the value is written to flash. For obvious reasons, I didn't experiment a lot ;)

If anyone knows how I can bring back my dead ESPs, please let me know :)

### Which version of ESPHome has the issue?

v2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.4

### What platform are you using?

ESP8266

### Board

D1 Mini

### Example YAML snippet

```yaml
esphome:
  name: bug

esp8266:
  board: d1_mini
  restore_from_flash: false

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key:

ota:
  password:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid:
    password:

captive_portal:
    
globals:
  - id: total_pulses
    type: int
    restore_value: true
    initial_value: '0'
```

ssieb, 2023-01-13:
You can't brick an ESP like that.  Your config does nothing, what are you expecting to happen?  People have been using `restore_from_flash: true` for a long time.

floriandotorg, 2023-01-13:
Hey @ssieb, thanks for your answer!

As far as I can tell the ESP is bricked. It's not connecting to Wi-Fi anymore, doesn't log anything even with `VERY_VERBOSE`.

It can be flashed but that's it, no reaction whatsoever. I even tried to flash the LED, but nothing.

Maybe it's trapped in an infinite loop, because the restore doesn't work. Even though, setting `restore_from_flash: false` doesn't bring it back to life.

What can I do? How can I debug this?

ssieb, 2023-01-13:
> It doesn't log anything not even with `VERY_VERBOSE`.

If it was bricked, you wouldn't be able to even change that, so I'm not sure what you're doing.

Please come to the discord server so we can have a more interactive discussion.

closed 2023-01-13 by ssieb