esphome/issues #4033: Remote Receiver not working on RP2040

seanco1, 2023-01-14:
### The problem

I seem to be getting a compile error when trying to enable the "remote_receiver:" option in the yaml config using the RP2040 platform.
`remote_receiver:`
`  pin: GPIO20`
`  dump: all`
`  buffer_size: 1kb`

If I remove the above code snippit or hash it out the YAML will compile and upload to the RP2040

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### What platform are you using?

RP2040

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
esphome:
  name: liam-led

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .homea.local

remote_receiver:
  pin: GPIO20
  dump: all
  buffer_size: 1kb

light:
  - platform: rgb
    name: "Liam LED Strip"
    red: output_component2
    green: output_component3
    blue: output_component1
    effects:
      - pulse:
      - random:
      - strobe:

output:
  - platform: rp2040_pwm
    id: output_component1
    pin: GPIO15
  - platform: rp2040_pwm
    id: output_component2
    pin: GPIO13
  - platform: rp2040_pwm
    id: output_component3
    pin: GPIO10

sensor:
  - platform: wifi_signal
    name: "Liam LED WiFi"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/liam-led.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing liam-led (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- MD5Builder @ 1.0.0
Linking .pioenvs/liam-led/firmware.elf
/config/.esphome/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pioenvs/liam-led/src/main.cpp.o: in function `setup':
main.cpp:(.text.setup+0x46c): undefined reference to `_ZTVN7esphome15remote_receiver23RemoteReceiverComponentE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/liam-led/firmware.elf] Error 1
========================= [FAILED] Took 134.60 seconds =========================
```

tomsride, 2023-01-20:
Same issue here; using ESPHome 2022.12.3; picow

Thanks for your support & all your great work!

YAML: esphome:
  name: pico-beamer

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "8JPMJKYBOztHWSSMuTqbkfLgtZbOlOTRqViSYVueuvU="

ota:
  password: "b8d3baa3f30ec95ff6a36c86cc5e0b95"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Pico-Beamer Fallback Hotspot"
    password: "del"

remote_receiver:
  pin: GPIO1
  dump: all
  buffer_size: 1kb

Respnse:
INFO Reading configuration /config/esphome/pico-beamer.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing pico-beamer (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- MD5Builder @ 1.0.0
Linking /data/pico-beamer/.pioenvs/pico-beamer/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /data/pico-beamer/.pioenvs/pico-beamer/src/main.cpp.o: in function `setup':
main.cpp:(.text.setup+0x480): undefined reference to `_ZTVN7esphome15remote_receiver23RemoteReceiverComponentE'
collect2: error: ld returned 1 exit status
*** [/data/pico-beamer/.pioenvs/pico-beamer/firmware.elf] Error 1
========================= [FAILED] Took 11.82 seconds =========================

Nicknakin, 2023-01-24:
I'm experiencing the same problem
```yaml
esphome:
  name: picoremote

# Example configuration entry
rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

logger:

# Enable Home Assistant API
api:
  password: "password"

ota:
  password: "password"

wifi:
  ssid: "internet"
  password: "password"

remote_receiver:
  pin: GPIO22
  dump: all
  buffer_size: 512b
```

```
NFO Reading configuration picoremote.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing picoremote (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
----------------------------------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- MD5Builder @ 1.0.0
Linking .pioenvs/picoremote/firmware.elf
/home/nick/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pioenvs/picoremote/src/main.cpp.o: in function `setup':
main.cpp:(.text.setup+0x468): undefined reference to `_ZTVN7esphome15remote_receiver23RemoteReceiverComponentE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/picoremote/firmware.elf] Error 1
======================================= [FAILED] Took 1.17 seconds =======================================
```

kuriburi, 2023-02-20:
I also experience the same error:
esphome v2023.2.3

```
INFO Reading configuration /config/esphome/room-ir-remote.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing room-ir-remote (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- MD5Builder @ 1.0.0
Linking /data/room-ir-remote/.pioenvs/room-ir-remote/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /data/room-ir-remote/.pioenvs/room-ir-remote/src/main.cpp.o: in function `setup':
main.cpp:(.text.setup+0x404): undefined reference to `_ZTVN7esphome18remote_transmitter26RemoteTransmitterComponentE'
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: main.cpp:(.text.setup+0x408): undefined reference to `_ZTVN7esphome18remote_transmitter26RemoteTransmitterComponentE'
collect2: error: ld returned 1 exit status
*** [/data/room-ir-remote/.pioenvs/room-ir-remote/firmware.elf] Error 1
```

Mjarred, 2023-04-26:
I’m having the same issue, is there a way to resolve this?

PropaneDragon, 2023-06-27:
For anyone interested, I did a bit of snooping and it appears there's no RP2040 implementation of remote_receiver so it's never going to compile for a RP2040 board. It appears all boards need their own custom implementation writing in C++.

freddiemare, 2023-12-24:
Any new updates around the implementation of the RemoteTransmitterComponent / RemoteReceiverComponent on the RP2040?

lsellens, 2023-12-25:
@freddiemare I submitted a PR for this last week. Works fine for me, but would love to have others test.

```
external_components:
  - source: github://pr#5974
    components: [remote_transmitter]
    refresh: always
```

seanco1, 2023-12-26:
I am not able to get this to work.

I updated ESPHome to version 2024.1.0-dev.
Added this near the top of the config
`external_components:`
`   - source: github://pr#5974`
`    components: [remote_transmitter]`
`    refresh: always`

unhashed this from my config
`remote_receiver:`
`  pin: GPIO20`
`  dump: all`
`  buffer_size: 1kb`

but I get this message "undefined reference to `_ZTVN7esphome15remote_receiver23RemoteReceiverComponentE'
collect2: error: ld returned 1 exit status" during the install to the 2040 module. Which seems to be the same message as before. I guess I have not done something correctly although in the logs below you can see the ESPHome version and the Git PR5974.

`
INFO ESPHome 2024.1.0-dev
INFO Reading configuration /config/esphome/liam-led.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5974/head
INFO Generating C++ source...
INFO Compiling app...
Processing liam-led (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30600.0 (3.6.0) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
|-- SPI @ 1.0
Compiling .pioenvs/liam-led/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/addressable_light.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/automation.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/esp_color_correction.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/esp_range_view.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/light_call.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/light_json_schema.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/light_output.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/light/light_state.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/ota/ota_backend_arduino_libretiny.cpp.o
[... 15 lines cut ...]
Compiling .pioenvs/liam-led/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/wifi/wifi_component_libretiny.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/wifi/wifi_component_pico_w.cpp.o
Compiling .pioenvs/liam-led/src/esphome/components/wifi_signal/wifi_signal_sensor.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/application.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/component.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/controller.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/entity_base.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/log.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/string_ref.cpp.o
Compiling .pioenvs/liam-led/src/esphome/core/util.cpp.o
Compiling .pioenvs/liam-led/src/main.cpp.o
Linking .pioenvs/liam-led/firmware.elf
/data/cache/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: .pioenvs/liam-led/src/main.cpp.o: in function `setup':
main.cpp:(.text.setup+0x494): undefined reference to `_ZTVN7esphome15remote_receiver23RemoteReceiverComponentE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/liam-led/firmware.elf] Error 1
========================= [FAILED] Took 78.38 seconds =========================
`

lsellens, 2023-12-26:
@seanco1 I didn't add the portion for the remote_receiver just the transmitter. I honestly didn't think about it since I don't use it but I'll look at it when I get home tonight to see if its just as easy to implement.

lsellens, 2023-12-27:
@seanco1 try now. I added the remote_receiver code.
```
external_components:
   - source: github://pr#5974
    components: [remote_transmitter, remote_receiver]
    refresh: always
```

seanco1, 2023-12-27:
@lsellens This is now working. Thanks for getting this to work on the RPi2040

Here is some of the debug when I pressed a button on a random remote
`
[11:37:44][I][remote.pronto:237]: Received Pronto: data=0000 006D 0002 0000 0160 0052 001A 0180 06C3
[11:37:44][I][remote.pronto:237]: Received Pronto: data=0000 006D 0002 0000 0160 0052 001A 0180 06C3`