esphome/issues #4036: Pulse Meter docs: Are the values used by `total` applied with the filter?

l3d00m, 2023-01-14:
### The problem

When using `pulse_meter` with a multiply filter, it's unclear whether the filter applies to the values collected by `total`.

Example: I have a power meter that does 375 pulses per kWh, which is why I use a `multiply: 160` filter inside `pulse_meter`.  

I'm now struggling whether I have to include those 375 pulses/kWh into the calculation for the `total` `multiply` filter or if the values used by `total` already have the previous filter from `pulse_meter` applied (so I don't need to apply any filter to `total`).

It would be great if this could be included into the documentation for `pulse_meter` somehow, maybe as a short sentence in the [Counting total pulses](https://esphome.io/components/sensor/pulse_meter.html#counting-total-pulses) section?

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

pulse_meter

### Example YAML snippet

```yaml
sensor:
  - platform: pulse_meter
    pin: D5
    name: "Pulse meter"
    unit_of_measurement: "W"
    filters:
      - multiply: 160 # (60s / 375 pulses per kWh) * 1000 W
    total:
      name: "Total pulse meter"
      unit_of_measurement: 'kWh'
      # filters:
      #  - multiply: 0.0027 #    <---  is this required?? (1/375 pulses per kWh)
```

l3d00m, 2023-05-15:
Not stale

TrentHouliston, 2023-09-13:
To answer the question, no the total sensor is entirely independent from any filters you add to the rate sensor