esphome/issues #4039: Unable to create firmware for the Pico W

Dr-Passmore, 2023-01-15:
### The problem

Trying to download the install manually or installing via usb or wireless provides this error message: 

Most  likely a config issue with the docker container, but I have been unable to solve the issue. 

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### What platform are you using?

RP2040

### Board

Pico W

### Example YAML snippet

```yaml
esphome:
  name: test-pico-w

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "zfw4Lr5sb6jaMh3LAuPT8uqprFpT8JfTKEG4B/MQHO4="

ota:
  password: "cb09f7b9a864501a2a7acfb0824bf125"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Test-Pico-W Fallback Hotspot"
    password: "qfvVu9dBkK8P"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/test-pico-w.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test-pico-w (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- MD5Builder @ 1.0.0
Compiling .pioenvs/test-pico-w/FrameworkArduinoVariant/picow_digital.cpp.o
Compiling .pioenvs/test-pico-w/FrameworkArduinoVariant/picow_init.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/FrameworkArduinoBootloader/boot2_w25q080_2_padded_checksum.S.o
sh: 1: arm-none-eabi-gcc: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/api_connection.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/api_pb2.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/api_server.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/list_entities.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/proto.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/subscribe_state.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/api/user_services.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/logger/logger.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/md5/md5.cpp.o
sh: 1: arm-none-eabi-g++: Permission denied
Compiling .pioenvs/test-pico-w/src/esphome/components/mdns/mdns_component.cpp.o
*** [.pioenvs/test-pico-w/FrameworkArduinoVariant/picow_digital.cpp.o] Error 126
Compiling .pioenvs/test-pico-w/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
*** [.pioenvs/test-pico-w/FrameworkArduinoVariant/picow_init.cpp.o] Error 126
*** [.pioenvs/test-pico-w/FrameworkArduinoBootloader/boot2_w25q080_2_padded_checksum.S.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/api_connection.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/api_frame_helper.cpp.o] Error 126
sh: 1: arm-none-eabi-g++: Permission denied
*** [.pioenvs/test-pico-w/src/esphome/components/api/api_pb2.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/api_pb2_service.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/api_server.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/list_entities.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/proto.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/subscribe_state.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/api/user_services.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/logger/logger.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/md5/md5.cpp.o] Error 126
*** [.pioenvs/test-pico-w/src/esphome/components/mdns/mdns_component.cpp.o] Error 126
sh: 1: arm-none-eabi-g++: Permission denied
*** [.pioenvs/test-pico-w/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o] Error 126
========================== [FAILED] Took 1.63 seconds ==========================
```

seanco1, 2023-01-16:
I copied and pasted your YAML snippit and the file compiles just fine. I am also using ESPHome inside docker (on Unraid).

If I was to guess to me it seems perhaps your docker container is not able to write to the disk it has mapped. The host path also may not have the correct permissions  at the OS level.

Dr-Passmore, 2023-01-16:
Thanks for the response @seanco1. I think you are correct. I will have to dig into the file permissions my docker container has.

closed 2026-03-12 by bdraco