esphome/issues #4040: Pulse Meter Sensor - wrong pulse counting

derwolff, 2023-01-15:
### The problem

I configured the Puls Meter Sensor for a TCR5000 infrared device to count the revolutions of a water meter with a reflective area (nearly a half circle) on the 0.001 m^3 pointer (one complete turn are 10L or 0.01 m^3).
Due to the slow frequency of the sensor I'm only interessted in the total value. 
I noticed radom counts and increasing of the total value. So I configured a binray_input sensor to track the switching of the infrared device and found no bouncing or random level change.
If have switch of any multiply filters and asume a increasing of the total value by 1 with every level change from OFF to ON of the sensor. But as you can see, there is a random increasing by 1,2 or 4. 

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2023.1.4

### What platform are you using?

ESP8266

### Board

Wemos Mini

### Component causing the issue

Pulse Meter Sensor

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    pin:     
      number: GPIO15
      inverted: true
      mode:
        input: true
    name: "Water Sensor"
    id: water_sensor
    filters:
      - delayed_on: 100ms
      
sensor:
  - platform: pulse_meter
    pin: 
      number: GPIO15
      inverted: true
      mode:
        input: true
    internal_filter: 100ms
    internal_filter_mode: edge   
    device_class: WATER
    name: 'Water Pulse Meter'
    filters:
        - multiply: 10
    unit_of_measurement: "l/min"
    icon: "mdi:water"
    total:
      name: "Water Meter Total"
      unit_of_measurement: "m³"
      id: water_meter_total
      accuracy_decimals: 3
      device_class: water
      state_class: total_increasing
      #filters:
      #  - multiply: 0.01 # 1l = 0.001 m^3
```

### Additional information

[image]
[image]

freijn, 2023-01-18:
Hi Wolff,

Can you test something?   
Make the watermeter to stop, so that the reflectionarea is under the sensor so the sensor gets a
trigger and keeps it high. 
I can see in your config you have edge as a filter, but I can't get it to work.
Please check/confirm this is the case with you as well.
If so, I have a ( dirty h/w ) solution for you :-)

Cheers,

  Frank

derwolff, 2023-01-19:
I changed the config to "pulse" and it's looks better, but in some periods the sensor counts without level change. 
What is your solution?

```
sensor:
  - platform: pulse_meter
    pin: 
      number: GPIO15
      inverted: true
      mode:
        input: true
    internal_filter: 5s
    internal_filter_mode: pulse
    device_class: WATER
    name: 'Water Pulse Meter'
    filters:
        - multiply: 10
    unit_of_measurement: "l/min"
    icon: "mdi:water"
    total:
      name: "Water Meter Total"
      unit_of_measurement: "m³"
      id: water_meter_total
      accuracy_decimals: 3
      device_class: water
      state_class: total_increasing
      filters:
        - multiply: 0.01 # 1l = 0.001 m^3
```

[image]

freijn, 2023-01-20:
Have you tested the issue starts when the refelction is under the sensor and hold ?

Watermeter puls is on GPIO12
I have hardwired the two pins together.   GPIO04 <->GPIO2

So using the binary sensor as edge detection and the puls sensor as counter.
Of course you can change the pin numbers as you like.
Please let me know the result.

#  watermeter counter  #
  - platform: pulse_counter
    pin: GPIO04
    id: watermeter_pulse
    name: "watermeter pulse"
    state_class: measurement
    unit_of_measurement: "l/min"    
    accuracy_decimals: 1
    icon: "mdi:water-pump"

#  Totaal water meter  #      
    total:
      id: sensor_pulse_meter_total
      name: "Watermeter Total"
      unit_of_measurement: "m³"
      state_class: total_increasing
      device_class: water
      accuracy_decimals: 3

# Watermeter Puls direct from sensor with edge detection by 'onpress'
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12
      mode: INPUT_PULLUP
    name: Pulse Sensor Water
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    on_press:
      then:
        output.turn_on: gpio_02
    on_release:
      then:
        output.turn_off: gpio_02
output:
  - platform: gpio
    pin:
      number: GPIO02
    id: gpio_02

`

derwolff, 2023-01-22:
In my case the issue happens due to the very slow transition from the mirror surface. Your solution is a good option but I decide to optimize the binary sensor setting and count the waterflow in Home Assistant. I create a counter helper and a automation for increasing the counter triggered by the binary sensor.
The output is way better than before.
I think the ESP Pulse Meter Sensor implementation needs all option from the binary sensor for correct triggering.

```
binary_sensor:
  - platform: gpio
    pin:     
      number: GPIO15
      inverted: true
      mode:
        input: true
    name: "Water Sensor"
    id: water_sensor
    filters:
      - delayed_on_off: 10s
```

fu-zhou, 2023-03-10:
I have the same situation: The sensor is sending a single pulse (4 seconds long though as the mirror turns slowly, but a clear edge is being recognized), so no multiple pulses while the mirror is under the sensor. But in between two pulses (about 90 seconds) the pulse counter is counting like 24 pulses which I find strange. What can be done about it?

```
14:04:38 | [D] | [binary_sensor:036] | 'TCRT5000_DI': Sending state ON
14:04:42 | [D] | [binary_sensor:036] | 'TCRT5000_DI': Sending state OFF
14:05:19 | [D] | [pulse_counter:174] | 'Pulse_Counter': Retrieved counter: 12.50 pulses/min
14:05:19 | [D] | [pulse_counter:180] | 'Pulse_Counter': Total : 218 pulses
14:06:13 | [D] | [binary_sensor:036] | 'TCRT5000_DI': Sending state ON
14:06:17 | [D] | [binary_sensor:036] | 'TCRT5000_DI': Sending state OFF
14:07:19 | [D] | [pulse_counter:174] | 'Pulse_Counter': Retrieved counter: 12.00 pulses/min
14:07:19 | [D] | [pulse_counter:180] | 'Pulse_Counter': Total : 242 pulses
```

The configuration looks like:
```
# TCRT5000
binary_sensor:
  - platform: gpio
    pin: D2
    name: "TCRT5000_DI"
    filters:
      invert

sensor:
  - platform: pulse_counter
    pin: D2
    name: "Pulse_Counter"
    count_mode:
      rising_edge: DISABLE
      falling_edge: INCREMENT
    update_interval: 120s

    total:
```

freijn, 2023-03-10:
Have you tried my solution?  ( see above for code)

fu-zhou, 2023-03-11:
I actually figured out that I was using "pulse_counter" while derwolff was using "pulse_meter". So I did not try your suggestion but I moved to "pulse_meter" and it now seems to work with the internal_filer:
```
sensor:
  - platform: pulse_meter
    pin: D2
    id: gaszaehler
    internal_filter: 500ms
    internal_filter_mode: PULSE
    total:
      name: "Gaszaehler_gesamt"
      unit_of_measurement: "m3"
      accuracy_decimals: 2
      filters:
        - multiply: 0.01
```

Qhilm, 2024-03-13:
> ### The problem
>
> I configured the Puls Meter Sensor for a TCR5000 infrared device to count the revolutions of a water meter with a reflective area (nearly a half circle) on the 0.001 m^3 pointer [...]

I find it very interesting that the total is increasing **before** the binary_sensor state is changing. Does someone have an explanation for this?