esphome/issues #4041: BLE notify truncated to 20 bytes

jdeath, 2023-01-15:
### The problem

I have BLE notify that should return 68 bytes. After a HA restart, the first update retrieves 68 bytes, but then all others only return 20 bytes (data I want is beyond 20). Integration works fine with USB Bluetooth adapter, but not an ESPHome proxy.

Proxies are ESPHome 2022.12.3 using esp-idf.

Also tried:
```
framework:
    type: esp-idf
    version: 4.4.3
    platform_version: 5.3.0
```

### Which version of ESPHome has the issue?

ESPHome 2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.4

### What platform are you using?

ESP32-IDF

### Board

ESP32

### Component causing the issue

https://github.com/jdeath/rd200v2

### Additional information

Maybe an MTU issue? Similar issues mentioned here: https://github.com/espressif/esp-idf/issues/3315

bdraco, 2023-01-15:
I think we can fix it with something like this
```
diff --git a/homeassistant/components/esphome/bluetooth/client.py b/homeassistant/components/esphome/bluetooth/client.py
index 6aa21c315a..cd62723f43 100644
--- a/homeassistant/components/esphome/bluetooth/client.py
+++ b/homeassistant/components/esphome/bluetooth/client.py
@@ -342,6 +342,9 @@ class ESPHomeClient(BaseBleakClient):
             await self.disconnect()
             raise
 
+        if has_cache:
+            await self._client.bluetooth_device_set_mtu(self._address_as_int, self._mtu)
+
         self._disconnected_event = asyncio.Event()
         return True
 

```

But we need an API on the esphome side

bdraco, 2023-01-16:
If that doesn't work we might need to make our own api to tell bluedroid the MTU which should be possible since `esp_ble_gatt_set_local_mtu` is relatively simple

closed 2023-01-20 by jesserockz