esphome/issues #4042: BME680 stops working when pressure sensor is configured

lindenaar, 2023-01-15:
### The problem

Hi,

I am using the bm680_bsec sensor platform, which does not work (i.e. I get nog sensor values) when I enable the pressure sensor.

When I do enable the pressure sensor I do not get any readings anymore, but my ESP8266 (ESP-01 module) only reports:

```
[20:30:37][W][bme680_bsec.sensor:237]: Failed to get sensor mode (BME680 Error Code -2)
[20:30:38][W][bme680_bsec.sensor:251]: Failed to get sensor data (BME680 Error Code 2)
```

oddly enough the other sensors work correctly when I do comment out the `pressure` sensor.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

223.1.4

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

bme680_bsec

### Example YAML snippet

```yaml
esp8266:
  board: esp01_1m

# Disable logging
logger:
  baud_rate: 0

uart:
  rx_pin: GPIO03 #GPIO2
  baud_rate: 9600

i2c:
  sda: GPIO00
  scl: GPIO02
  scan: true

bme680_bsec:
    address: 0x77
    sample_rate: ulp

sensor:
  - platform: pm1006
    pm_2_5:
      name: "Particulate Matter 2.5µm Concentration"

  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
[... 181 lines cut ...]
[20:53:17][D][pm1006:091]: Got PM2.5 Concentration: 1 µg/m³
[20:53:17][D][sensor:126]: 'Particulate Matter 2.5µm Concentration': Sending state 1.00000 µg/m³ with 0 decimals of accuracy
[20:53:19][D][pm1006:091]: Got PM2.5 Concentration: 1 µg/m³
[20:53:19][D][sensor:126]: 'Particulate Matter 2.5µm Concentration': Sending state 1.00000 µg/m³ with 0 decimals of accuracy
[20:53:19][W][bme680_bsec.sensor:237]: Failed to get sensor mode (BME680 Error Code -2)
[20:53:19][W][bme680_bsec.sensor:251]: Failed to get sensor data (BME680 Error Code 2)
[20:53:21][D][pm1006:091]: Got PM2.5 Concentration: 1 µg/m³
[20:53:21][D][sensor:126]: 'Particulate Matter 2.5µm Concentration': Sending state 1.00000 µg/m³ with 0 decimals of accuracy
[20:53:22][W][bme680_bsec.sensor:237]: Failed to get sensor mode (BME680 Error Code -2)
[20:53:22][W][bme680_bsec.sensor:251]: Failed to get sensor data (BME680 Error Code 2)
[20:53:23][D][pm1006:091]: Got PM2.5 Concentration: 1 µg/m³
[20:53:23][D][sensor:126]: 'Particulate Matter 2.5µm Concentration': Sending state 1.00000 µg/m³ with 0 decimals of accuracy
[20:53:25][W][bme680_bsec.sensor:237]: Failed to get sensor mode (BME680 Error Code -2)
[20:53:25][W][bme680_bsec.sensor:251]: Failed to get sensor data (BME680 Error Code 2)
[20:53:26][D][pm1006:091]: Got PM2.5 Concentration: 1 µg/m³
[20:53:26][D][sensor:126]: 'Particulate Matter 2.5µm Concentration': Sending state 1.00000 µg/m³ with 0 decimals of accuracy
[20:53:28][D][pm1006:091]: Got PM2.5 Concentration: 1 µg/m³
[20:53:28][D][sensor:126]: 'Particulate Matter 2.5µm Concentration': Sending state 1.00000 µg/m³ with 0 decimals of accuracy
[20:53:28][W][bme680_bsec.sensor:237]: Failed to get sensor mode (BME680 Error Code -2)
[20:53:28][W][bme680_bsec.sensor:251]: Failed to get sensor data (BME680 Error Code 2)
```

### Additional information

When I comment out the `pressure` sensor in tye `YAML` config, the other sensors work as expected (so far).

renanqts, 2024-01-21:
same here, it stopped since I bumped my esphome to 2023.12 and upgraded the firmware

gouppe, 2025-06-04:
same for me; stopped to work after any update, currently esphome 2025.5.2