esphome/issues #4044: On_value_range doesn't support "FOR" condition

Tryfos, 2023-01-17:
### The problem

Whenever I add the "for" condition (https://esphome.io/guides/automations.html#for-condition) to a on_value_range automation, all the conditions do not work (not just the "for" condition). The action is always triggered.
Is the "for" condition supported in a on_value_range automation?
This is similar to unresolved issue #3255 

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.5

### What platform are you using?

ESP8266

### Board

Flashed smart plug Moesrgb (esp01_1m)

### Component causing the issue

on_value_range condition

### Example YAML snippet

```yaml
sensor:
  - platform: hlw8012
    <<: !include common/devices/plugs/moesrgb/hlw8012_base.yaml
    power:
      name: $friendly_name power
      unit_of_measurement: W
      id: power
      on_value_range:
        - below: 7.5
          then:
            - if:
                condition:
                  - for:
                      time: 2s
                      condition:
                        sensor.in_range:
                          id: power
                          below: 7.5
                  - light.is_on: rgb_light
                then:
                  light.turn_off: rgb_light
```

### Additional information

If I remove the "for" condition and/or add any other conditions to the automation, it works as expected.

ssieb, 2023-01-17:
It can't work like that.  `on_value_range:` is called **once** when the value goes in range.  At that point, the sensor is obviously not going to have been in that range for 2s.  So the condition will never be true.

closed 2023-01-17 by ssieb

Tryfos, 2023-01-18:
Ok, fair enough, I thought it was the same case with #3255, where lambda was used (it's not clear if that was resolved).
So, if I use another sensor for a FOR condition nested within an on_value_range, this will work? Is the "for" condition normally supported for this automation?

ssieb, 2023-01-18:
That issue is the exact same thing that you're trying to do here.  A `for:` is not likely going to work in `on_value_range:` for any condition though because of what I explained earlier.  If you need more help, come to the discord server.  GitHub issues are not for support.