esphome/issues #4045: Problem compiling

berdugorn, 2023-01-18:
### The problem

First time trying to install ESPhome to ESP32 and it fails.
From Home Assisntant I went to ESPhome tab,
I chose the option "Plug into this computer",
I chose the correct COM port,
It showed "Connecting..." then "Preparing Installation.." and then "Failed to prepare configuration, See what went wrong."

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.5

### What platform are you using?

ESP32

### Board

LOLIN32

### Example YAML snippet

```yaml
esphome:
  name: door-switch

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Lns+SkYmKVL0lTbdYoLqQWBAqS7zDwr+HmxNukX5TUY="

ota:
  password: "f095b3816ad4ef68821227f3464eed50"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Door-Switch Fallback Hotspot"
    password: "eBGdWbHXuCeI"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/door-switch.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing door-switch (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/door-switch/.pioenvs/door-switch/src/esphome/components/api/api_connection.cpp.o
Compiling /data/door-switch/.pioenvs/door-switch/src/esphome/components/api/api_frame_helper.cpp.o
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:116:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/door-switch/.pioenvs/door-switch/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:116:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/door-switch/.pioenvs/door-switch/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 1.72 seconds ==========================
```

azzwiz, 2023-01-18:
I'm getting the same error trying to install a new esp32

silvioped, 2023-01-20:
FYI, I solved uninstalling and then reinstalling the esp32 addon on ha (you don't loose the configured boards)

berdugorn, 2023-01-20:
> FYI, I solved uninstalling and then reinstalling the esphome addon on ha (you don't loose the configured boards)

I confirm that this works! 👍 
thank you!

closed 2023-01-20 by berdugorn