esphome/issues #4048: IR Receiver not parsing Magiquest results

mikerenna, 2023-01-19:
### The problem

I have two different Magiquest wands for my kids and am attempting to leverage the ESPHome support for the protocol. However, I an not seeing any magiquest protocol results. With an "all" setting for the IR receiver, i see only pronto labeled results. if i set it to magiquest i get nothing, and if set to raw, i see plenty of data rows labeled as raw. Any idea what might be the issue here? I recognize that perhaps the wands are the problem. I have no way to prove they are sending maqiquest protocol beyond that they are marketed as Magiquest by the Great Wolf Lodge where my kids got them. Any ideas on how to troubleshoot this would be appreciated. Thanks in advance. -Mike
[logs_maqiquest-1_upload.txt](https://github.com/esphome/issues/files/10458554/logs_maqiquest-1_upload.txt)

### Which version of ESPHome has the issue?

2.0.6

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

IR Receiver

### Example YAML snippet

```yaml
remote_receiver:
  pin:
    number: GPIO32
    inverted: true
    mode:
      input: true
  dump: all
```

F0rged, 2023-04-01:
I am fighting with the same issue. Did you ever figure this out?

NeoMatrixJR, 2023-11-29:
Same, following.

mikerenna, 2023-12-01:
> I can confirm that MagiQuest used to work nicely but broke at some point, now only showing pronto codes - not sure what version.
> I have old wands and successfully used this to get two different wand IDs and was using them for triggering automations for the kids.
> Broke with an update but don't recall what old version the board was on before I updated it :-(

Thanks Blake. I gave up on the idea. Would have been cool, but I was ultimately defeated. ☹ Thanks for the information. At least I know I’m not crazy.

sreknob, 2023-12-01:
Following up here.
I was able to get it reliably working again on the current version just by telling the controller to stop dumping other codes.

Relevant config:
```
remote_receiver:
  pin:
    number: D5
    inverted: true
    mode:
      input: true
      pullup: true
  dump:
    - magiquest

binary_sensor:
  - platform: remote_receiver
    name: "Grant's Wand"
    id: grants_wand
    magiquest:
      wand_id: 0x020DF054

  - platform: remote_receiver
    name: "Luke's Wand"
    id: lukes_wand
    magiquest:
      wand_id: 0x020EA194
```

```
[13:01:21][I][app:102]: ESPHome version 2023.11.6 compiled on Dec  1 2023, 12:58:58
[13:06:07][I][ota:117]: Boot seems successful, resetting boot loop counter.
[16:09:12][I][remote.magiquest:079]: Received MagiQuest: wand_id=0x020DF054, magnitude=0x0C0B
[16:09:15][I][remote.magiquest:079]: Received MagiQuest: wand_id=0x020EA194, magnitude=0x0815
[16:09:15][I][remote.magiquest:079]: Received MagiQuest: wand_id=0x020EA194, magnitude=0x0C09
```

jordandrako, 2024-02-28:
I just got a couple new Magiquest wands for the kids and I'm running into the same issue. I know Great Wolf Lodge recently expanded/updated MagiQuest and heard they got new wands, so I'm not sure if the decoder works for the new wands yet. I'm only getting pronto codes with dump: all, and nothing with dump: magiquest.

Good to know @sreknob was able to get it working for known wand IDs but I have no idea how to get the ID for my new wands.

sreknob, 2024-02-29:
@jordandrako - I only have old wands and the MagiQuest wand ID is output with every IR receive in the ESPHome logs, as designed.

Have you tried with another library like https://github.com/Arduino-IRremote/Arduino-IRremote to see if you can get codes out of it? If so, it could be in the implementation of the protocol in ESPHome.

Briefly looking over the Arduino-IRremote files, the only slight difference I see in the protocol is that the `MAGIQUEST_UNIT` that ESPHome uses is 288 rather than the calculated 287.5 that other libraries use, like the project mentioned above:

```
#define MAGIQUEST_PERIOD         1150   // Time for a full MagiQuest "bit" (1100 - 1200 usec)
#define MAGIQUEST_UNIT          (MAGIQUEST_PERIOD / 4) // 287.5
```

and ESPHome uses `static const uint32_t MAGIQUEST_UNIT = 288;  // us`

There may be other differences in parsing, would need to look deeper. I'm not sure this is going to make a huge difference though in the IR world.

Anyway, thought that maybe you could try that small change out in magiquest_protocol.cpp and/or verify that Arduino-IRremote works with your wand. If changing the definition of the unit work, that's a simple fix. If not and Arduino-IRremote works, then it's obviously ESPHome's implementation to be looked at further.

Good luck!

mzakharo, 2024-03-02:
Just got back from Great Wolf Lodge and apparently got the `latest` wands. We have 2 of them and one seems to work better than the other. One thing that helped me is setting `tolerance` to `45%` in the yaml.

Siege36, 2024-04-02:
Experiencing same issue here. 3 brand new wands. I am unsure of their wand IDs and am unable to figure out what they are. 

current config:
remote_receiver:
  pin:
    number: GPIO4
    inverted: true
  dump:
    - magiquest

Setting dump to magiquest gives me no log messages. Changing dump to -all gives me the below pronto raw data:
Log here:

[I][remote.pronto:233]: Received Pronto: data=
[I][remote.pronto:235]: 0000 006D 0020 0000 0007 0051 0006 0053 0005 0053 0005 0053 0013 0045 0012 0046 0006 0053 0006 0027 0013 0018 0005 0027 0012 001A 0011 0047 0005 0027 0012 00CB 0008 0025 0012 0072 0012 001B 0012 0072 0005 0053 0006 0053 0013 001A 
[I][remote.pronto:235]: 0012 001A 0005 0053 0007 0051 0007 0051 0013 0045 0011 001C 0012 001A 0011 0048 0005 0053 0013 001A 0012 0181 06C3
[I][remote.pronto:233]: Received Pronto: data=
[I][remote.pronto:235]: 0000 006D 0024 0000 000A 004F 000A 004F 0008 0051 0008 0051 0014 0044 0014 0044 000A 004F 0008 0025 0015 0017 0008 0025 0014 0018 0014 0044 000A 0023 0014 0018 0008 0051 0008 0051 0009 0023 0015 0017 0007 0051 0013 001A 0015 0017 
[I][remote.pronto:235]: 0008 0051 0008 0051 0009 004F 0014 0018 0014 0018 000A 004F 0008 0051 0008 0051 0014 0044 0014 0018 0014 0018 0013 0045 0007 0051 0014 0018 0015 0181 06C3
[I][remote.pronto:233]: Received Pronto: data=
[I][remote.pronto:235]: 0000 006D 0022 0000 0006 0053 0006 0053 0008 004E 000A 0025 0013 0045 0012 001A 0005 0053 0006 0053 0013 0045 0012 001B 0012 0072 0012 0046 0005 0053 0007 0051 0013 0045 0007 0025 0012 0019 0012 0047 0006 0053 0006 0053 000B 0021 
[I][remote.pronto:235]: 0013 0018 0012 0046 0006 0053 0009 004F 0008 0025 0012 0046 0012 001A 0011 001C 0012 001A 0005 0053 0005 0027 0012 001A 0012 0181 06C3
[I][remote.pronto:233]: Received Pronto: data=
[I][remote.pronto:235]: 0000 006D 0022 0000 0008 0051 0008 0051 0008 0051 0006 0027 0014 0018 0006 0027 0013 0071 0006 0053 0013 0045 0013 001A 0012 0072 0013 0045 0006 0053 0007 0051 0014 0044 0006 0027 0012 0019 0013 0045 0005 0053 0007 0051 0007 0025 
[I][remote.pronto:235]: 0013 0018 0013 0045 0007 0051 0006 0053 0007 0025 0014 0018 0006 0027 0012 001A 0011 001C 0012 0072 0006 0027 0013 0018 0013 0181 06C3

christ, 2024-11-13:
YMMV but here's what I used to get consistent data from the new magiquest wands. I have no idea why it works, but I just started stepping through tolerance levels starting at 30% until I got a proper wand_id decoded.

remote_receiver:
  pin: 
    number: GPIO16
    inverted: true
  dump: magiquest
  tolerance: 40%

gqkevyn, 2025-03-21:
Just got new wands in early 2025 and like many magiquest wand_id didn't show. I found two solutions. One listed here by christ and was to use add tolerance: 40% after dump:magiquest. Did this on 3 different set ups no issues.

The other was using a 36khz IR receiver. Most of the ones in the market are 38khz. With the 36khz IR receiver I could remove the tolerance: 40% and still get the wand id.

Spectre5, 2025-04-16:
The solution for me was the 36khz IR receiver.  The 38khz was spotty at best, even with large tolerances.  The 36khz has never shown a wrong wand_id and consistently triggers every time.  I'm using the TSOP38236 for anyone interested.