esphome/issues #4049: Got invalid frame from camera!

andreclemente, 2023-01-19:
### The problem

I'm trying to use a ESP32-CAM with the cameras DCX-OV2 640-V2 and OV2640. None of them work, I keep getting "Got invalid frame from camera!"

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.5

### What platform are you using?

ESP32

### Board

ESP-32S + ESP32-CAM-MB

### Example YAML snippet

```yaml
substitutions:
  esphome_name: espcam
  camera_name: espcam camera

esphome:
  name: "${esphome_name}"
  platform: ESP32
  board: node32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
  reboot_timeout: 60min
  encryption:
    key: "key"

ota:

time:
  - platform: homeassistant
  
logger:

sensor:
  # General
  - platform: uptime
    name: "${esphome_name} - Uptime Sensor"
  - platform: wifi_signal
    name: "${esphome_name} - WiFi Signal"
    update_interval: 60s

esp32_camera:
  name: "${camera_name}"
  external_clock:
    pin: GPIO0
    frequency: 10MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  # Image settings
  max_framerate: 5 fps    # default: 10 fps, max 60
  idle_framerate: 0.2 fps # default: 0.1 fps - framerate for 'picture' in HA dashboard
  resolution: 800x600     # default: 640x480 (VGA) - higher res requires more memory
  jpeg_quality: 63        # 10 (best) to 63 (worst)

switch:
  - platform: gpio
    name: "${camera_name} flash"
    pin: 4
    #inverted: True
  - platform: restart
    name: "${esphome_name} - Restart"
    id: restart_switch
```

### Anything in the logs that might be useful for us?

```txt
[22:01:45][C][esp32_camera:048]: ESP32 Camera:
[22:01:45][C][esp32_camera:049]:   Name: espcam camera
[22:01:45][C][esp32_camera:050]:   Internal: NO
[22:01:45][C][esp32_camera:052]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[22:01:45][C][esp32_camera:053]:   VSYNC Pin: 25
[22:01:45][C][esp32_camera:054]:   HREF Pin: 23
[22:01:45][C][esp32_camera:055]:   Pixel Clock Pin: 22
[22:01:45][C][esp32_camera:056]:   External Clock: Pin:0 Frequency:10000000
[22:01:45][C][esp32_camera:060]:   I2C Pins: SDA:26 SCL:27
[22:01:45][C][esp32_camera:062]:   Reset Pin: -1
[22:01:45][C][esp32_camera:083]:   Resolution: 800x600 (SVGA)
[22:01:45][C][esp32_camera:105]:   JPEG Quality: 63
[22:01:45][C][esp32_camera:107]:   Contrast: 0
[22:01:45][C][esp32_camera:108]:   Brightness: 0
[22:01:45][C][esp32_camera:109]:   Saturation: 0
[22:01:45][C][esp32_camera:110]:   Vertical Flip: ON
[22:01:45][C][esp32_camera:111]:   Horizontal Mirror: ON
[22:01:45][C][esp32_camera:112]:   Special Effect: 0
[22:01:45][C][esp32_camera:113]:   White Balance Mode: 0
[22:01:45][C][esp32_camera:116]:   Auto Exposure Control: 1
[22:01:45][C][esp32_camera:117]:   Auto Exposure Control 2: 0
[22:01:45][C][esp32_camera:118]:   Auto Exposure Level: 0
[22:01:45][C][esp32_camera:119]:   Auto Exposure Value: 300
[22:01:45][C][esp32_camera:120]:   AGC: 1
[22:01:45][C][esp32_camera:121]:   AGC Gain: 0
[22:01:45][C][esp32_camera:122]:   Gain Ceiling: 0
[22:01:45][C][esp32_camera:128]:   Test Pattern: NO
[22:01:45][C][restart:076]: Restart Switch 'espcam - Restart'
[22:01:45][C][restart:078]:   Icon: 'mdi:restart'
[22:01:45][C][restart:099]:   Restore Mode: restore defaults to OFF
[22:01:45][C][psram:020]: PSRAM:
[22:01:45][C][psram:021]:   Available: YES
[22:01:45][C][psram:024]:   Size: 1 MB
[22:01:45][C][mdns:103]: mDNS:
[22:01:45][C][mdns:104]:   Hostname: espcam
[22:01:45][C][ota:093]: Over-The-Air Updates:
[22:01:45][C][ota:094]:   Address: espcam.local:3232
[22:01:45][C][api:138]: API Server:
[22:01:45][C][api:139]:   Address: espcam.local:6053
[22:01:45][C][api:141]:   Using noise encryption: YES
[22:01:45][C][wifi_signal.sensor:009]: WiFi Signal 'espcam - WiFi Signal'
[22:01:45][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[22:01:45][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[22:01:45][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[22:01:45][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[22:01:47][W][esp32_camera:166]: Got invalid frame from camera!
[22:01:51][W][esp32_camera:166]: Got invalid frame from camera!
[22:01:55][W][esp32_camera:166]: Got invalid frame from camera!
[22:01:59][W][esp32_camera:166]: Got invalid frame from camera!
[22:02:03][W][esp32_camera:166]: Got invalid frame from camera!
[22:02:07][W][esp32_camera:166]: Got invalid frame from camera!
[22:02:11][W][esp32_camera:166]: Got invalid frame from camera!
[22:02:19][W][esp32_camera:166]: Got invalid frame from camera!
[22:02:23][D][sensor:127]: 'espcam - WiFi Signal': Sending state -68.00000 dBm with 0 decimals of accuracy
[22:02:23][W][esp32_camera:166]: Got invalid frame from camera!
```

terveer, 2023-02-21:
Same here: (I had to use the PSRAM Framebuffer workaround though - see #3908)

INFO Reading configuration /config/freenove-s3.yaml...
INFO Starting log output from 192.168.*.* using esphome API
INFO Successfully connected to 192.168.*.*
[23:04:41][I][app:102]: ESPHome version 2023.2.2 compiled on Feb 21 2023, 22:58:44
[23:04:41][C][wifi:504]: WiFi:
[23:04:41][C][wifi:367]:   Hostname: 'freenoves3'
[23:04:41][C][wifi:369]:   Signal strength: -59 dB ▂▄▆█
[23:04:41][C][logger:293]: Logger:
[23:04:41][C][logger:294]:   Level: DEBUG
[23:04:41][C][logger:295]:   Log Baud Rate: 115200
[23:04:41][C][logger:296]:   Hardware UART: UART0
[23:04:41][C][logger:298]:   Level for 'esp32_ble_tracker': DEBUG
[23:04:41][C][logger:298]:   Level for 'wifi': DEBUG
[23:04:41][C][logger:298]:   Level for 'i2c': DEBUG
[23:04:41][C][logger:298]:   Level for 'esp32_camera': DEBUG
[23:04:41][C][switch.gpio:076]: GPIO Switch 'LED'
[23:04:41][C][switch.gpio:099]:   Restore Mode: restore defaults to OFF
[23:04:41][C][switch.gpio:031]:   Pin: GPIO2
[23:04:41][C][uptime.sensor:031]: Uptime Sensor 'freenoves3 Uptime'
[23:04:41][C][uptime.sensor:031]:   Device Class: 'duration'
[23:04:41][C][uptime.sensor:031]:   State Class: ''
[23:04:41][C][uptime.sensor:031]:   Unit of Measurement: 's'
[23:04:41][C][uptime.sensor:031]:   Accuracy Decimals: 0
[23:04:41][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[23:04:41][C][restart:076]: Restart Switch 'Restart freenoves3'
[23:04:41][C][restart:078]:   Icon: 'mdi:restart'
[23:04:41][C][restart:099]:   Restore Mode: restore defaults to OFF
[23:04:41][C][esp32_camera:048]: ESP32 Camera:
[23:04:41][C][esp32_camera:049]:   Name: ESP32-Kamera
[23:04:41][C][esp32_camera:050]:   Internal: NO
[23:04:41][C][esp32_camera:052]:   Data Pins: D0:16 D1:17 D2:18 D3:8 D4:9 D5:10 D6:11 D7:12
[23:04:41][C][esp32_camera:053]:   VSYNC Pin: 6
[23:04:41][C][esp32_camera:054]:   HREF Pin: 7
[23:04:41][C][esp32_camera:055]:   Pixel Clock Pin: 13
[23:04:41][C][esp32_camera:056]:   External Clock: Pin:15 Frequency:20000000
[23:04:41][C][esp32_camera:060]:   I2C Pins: SDA:4 SCL:5
[23:04:41][C][esp32_camera:062]:   Reset Pin: -1
[23:04:41][C][esp32_camera:083]:   Resolution: 800x600 (SVGA)
[23:04:41][C][esp32_camera:105]:   JPEG Quality: 12
[23:04:41][C][esp32_camera:107]:   Contrast: 0
[23:04:41][C][esp32_camera:108]:   Brightness: 0
[23:04:41][C][esp32_camera:109]:   Saturation: 0
[23:04:41][C][esp32_camera:110]:   Vertical Flip: ON
[23:04:41][C][esp32_camera:111]:   Horizontal Mirror: ON
[23:04:41][C][esp32_camera:112]:   Special Effect: 0
[23:04:41][C][esp32_camera:113]:   White Balance Mode: 0
[23:04:41][C][esp32_camera:116]:   Auto Exposure Control: 1
[23:04:41][C][esp32_camera:117]:   Auto Exposure Control 2: 0
[23:04:41][C][esp32_camera:118]:   Auto Exposure Level: 0
[23:04:41][C][esp32_camera:119]:   Auto Exposure Value: 300
[23:04:41][C][esp32_camera:120]:   AGC: 1
[23:04:41][C][esp32_camera:121]:   AGC Gain: 0
[23:04:41][C][esp32_camera:122]:   Gain Ceiling: 0
[23:04:41][C][esp32_camera:128]:   Test Pattern: NO
[23:04:41][C][copy.sensor:015]: Copy Sensor 'freenoves3 WiFi Signal Percent'
[23:04:41][C][copy.sensor:015]:   Device Class: 'signal_strength'
[23:04:41][C][copy.sensor:015]:   State Class: 'measurement'
[23:04:41][C][copy.sensor:015]:   Unit of Measurement: '%'
[23:04:41][C][copy.sensor:015]:   Accuracy Decimals: 0
[23:04:41][C][mdns:108]: mDNS:
[23:04:41][C][mdns:109]:   Hostname: freenoves3
[23:04:41][C][ota:093]: Over-The-Air Updates:
[23:04:41][C][ota:094]:   Address: 192.168.189.147:3232
[23:04:41][C][ota:097]:   Using Password.
[23:04:41][C][api:138]: API Server:
[23:04:41][C][api:139]:   Address: 192.168.189.147:6053
[23:04:41][C][api:141]:   Using noise encryption: YES
[23:04:41][C][wifi_info:009]: WifiInfo IPAddress 'freenoves3 IP address'
[23:04:41][C][wifi_signal.sensor:009]: WiFi Signal 'freenoves3 WiFi Signal'
[23:04:41][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[23:04:41][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[23:04:41][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[23:04:41][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[23:04:44][W][esp32_camera:166]: Got invalid frame from camera!
[23:04:48][W][esp32_camera:166]: Got invalid frame from camera!
[23:04:52][W][esp32_camera:166]: Got invalid frame from camera!
[23:04:56][W][esp32_camera:166]: Got invalid frame from camera!
[23:04:59][D][esp32.preferences:113]: Saving 1 preferences to flash...
[23:04:59][D][esp32.preferences:142]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[23:05:00][W][esp32_camera:166]: Got invalid frame from camera!

Alan-K2, 2023-03-10:
I had the same issue. For me the solution was to remove the ref to the power down pin GPIO32 from the YAML.

andreclemente, 2023-08-08:
Tried both and it didn't work. Still have the same "[W][esp32_camera:190]: Got invalid frame from camera!"

yeyeto2788, 2024-03-18:
Did you guys managed to make it work? My board has a `8225N V2.0 171026` camera and I get the same error.

Trying to remove the power actually made it worse as the camera never initialized.

timnolte, 2025-06-01:
So frustrated as I had this all working then updated to the latest ESPHome Builder (2025.5.1) in Home Assistant then was prompted up update the firmware on the device and the result was this error over and over with no camera working any more. The device had been working without issue for months. I've tried power cycling the device and reinstalled the configuration multiple times to no avail. Other functions of the board work just fine.

timnolte, 2025-06-01:
> So frustrated as I had this all working then updated to the latest ESPHome Builder (2025.5.1) in Home Assistant then was prompted up update the firmware on the device and the result was this error over and over with no camera working any more. The device had been working without issue for months. I've tried power cycling the device and reinstalled the configuration multiple times to no avail. Other functions of the board work just fine.
> 
> So even after making sure that I also updated HA to the latest version which I hadn’t yet. Then wiping all of the build files and doing a reinstall I still end up with:
> 
> ```
> [W][esp32_camera:187]: Got invalid frame from camera!
> ``
> ```

Well, I unplugged the power and plugged it back in and now it's working again. I guess I'll probably need to just chalk this up to cheap flaky hardware. :man_shrugging:

sapphirezero89, 2026-01-30:
Same problem after updating. Anyone find a fix?