esphome/issues #4051: CANbus on ESP32 crashes on a busy bus

wjcloudy, 2023-01-20:
### The problem

Trying to read a single ID from a fairly busy (around 100 messages a second) 500kbps CANbus. 

The ESP handles about 700 messages before crashing and rebooting over and over. I've cut the config right down so there are no actions being performed (and works fine when unplugged from the bus) Is it possible to implement the ESP32's hardware canbus filter to avoid processing every single message?

[22:51:30][D][canbus:070]: received can message (#684) std can_id=0x521 size=8
[22:51:30][D][canbus:070]: received can message (#685) std can_id=0x600 size=8
[22:51:30][D][canbus:070]: received can message (#686) std can_id=0x355 size=8
[22:51:30][D][canbus:070]: received can message (#687) std can_id=0x621 size=8
[22:51:30][D][canbus:070]: received can message (#688) std can_id=0x600 size=8
[22:51:30][D][canbus:070]: received can message (#689) std can_id=0x110 size=8
[22:51:30][D][canbus:070]: received can message (#690) std can_id=0x521 size=8
[22:51:30][D][canbus:070]: received can message (#691) std can_id=0x200 size=8
[22:51:30][D][canbus:070]: received can message (#692) std can_id=0x621 size=8
[22:51:30][D][canbus:070]: received can message (#693) std can_id=0x110 size=8
[22:51:30][D][canbus:070]: received can message (#694) std can_id=0x310 size=8
[22:51:30][D][canbus:070]: received can message (#695) std can_id=0x210 size=8
[22:51:30][D][canbus:070]: received can message (#696) std can_id=0x631 size=8
INFO 192.168.1.85: Ping timed out!
INFO Disconnected from ESPHome API for 192.168.1.85
WARNING Disconnected from API

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.6

### What platform are you using?

ESP8266

### Board

esp-wrover-kit

### Component causing the issue

esp32_can

### Example YAML snippet

```yaml
canbus:
  - platform: esp32_can
    tx_pin: GPIO21
    rx_pin: GPIO22
    can_id: 4
    bit_rate: 500kbps
    on_frame:
    - can_id: 0x373 #BMS_High_Low
      then:
        - light.toggle:
            id: led_buitin
```

### Anything in the logs that might be useful for us?

```txt
[22:51:30][D][canbus:070]: received can message (#683) std can_id=0x310 size=8
[22:51:30][D][canbus:070]: received can message (#684) std can_id=0x521 size=8
[22:51:30][D][canbus:070]: received can message (#685) std can_id=0x600 size=8
[22:51:30][D][canbus:070]: received can message (#686) std can_id=0x355 size=8
[22:51:30][D][canbus:070]: received can message (#687) std can_id=0x621 size=8
[22:51:30][D][canbus:070]: received can message (#688) std can_id=0x600 size=8
[22:51:30][D][canbus:070]: received can message (#689) std can_id=0x110 size=8
[22:51:30][D][canbus:070]: received can message (#690) std can_id=0x521 size=8
[22:51:30][D][canbus:070]: received can message (#691) std can_id=0x200 size=8
[22:51:30][D][canbus:070]: received can message (#692) std can_id=0x621 size=8
[22:51:30][D][canbus:070]: received can message (#693) std can_id=0x110 size=8
[22:51:30][D][canbus:070]: received can message (#694) std can_id=0x310 size=8
[22:51:30][D][canbus:070]: received can message (#695) std can_id=0x210 size=8
[22:51:30][D][canbus:070]: received can message (#696) std can_id=0x631 size=8
INFO 192.168.1.85: Ping timed out!
INFO Disconnected from ESPHome API for 192.168.1.85
WARNING Disconnected from API
```

ivoh95, 2023-05-14:
Thank you for this! 

Ive been troubleshooting why my esphome canbus device stopped working: i changed the logger level to debug to debug an issue. But another issue is caused by the debug log,  making it impossible to debug. Canbus must cause a huge amount of log messages to crash the device! A note in the documentation needs to be added about this problem. n my use case, I am connected to a vehicle canbus network, easily multiple 100s of messages every second. 

I believe the log level on a per component basis can be set. As seen here https://esphome.io/components/logger.html#manual-tag-specific-log-levels

If the correct fix is to add a note in the documentation about debugging with can, or if something can be done on a lower level im not sure.

randyoo, 2024-07-17:
As of today, this issue seems to continue to exist. Default log level is "debug" for canbus, and after about message #685, the entire ESP hangs. It looks like setting the log level to "INFO" keeps it from hanging. But it would be preferable to be able to get those debug level messages.