esphome/issues #4053: Wifi connection not working error Auth expired

riker65, 2023-01-21:
### The problem

After reflashing the esphome device error occured with wifi connection
`Auth expired `

I did reflash the ESP8266 with a plain firmware via arduino ide serial connection, as no wifi was working.

deleted the whole flash area and config data and sketch

device connects normally to same wifi with arduino sketch and tasmota

from tasmota I did load new firmware esphome  -which worked formerly and also an a new device -  and got the error again

`Auth rexpired.`

issue accured again when 
`esphome run testtr.yaml`

### Which version of ESPHome has the issue?

2022.12.13

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

none

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

only mqtt, switch

### Example YAML snippet

```yaml
esphome:
  name: trtestnew

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"

wifi:
#  use_address: trtest.local
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Trtest Fallback Hotspot"
    password: "Q5McJVNgRbEf"

captive_portal:

# Example configuration entry
sensor:
  - platform: dht
    pin: D2
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s
switch:
  - platform: gpio
    name: "LED_Int"
    pin: 2
    inverted: true

# Example configuration entry
web_server:
  port: 80
  
# Example configuration entry
mqtt:
  broker: 192.168.0.9
  username: !secret mqtt_user
  password: !secret mqtt_password
  
binary_sensor:
  - platform: gpio
    name: "Living Room Window"
    pin:
      number: 16
      inverted: true
      mode:
        input: true
#        pullup: true
```

### Anything in the logs that might be useful for us?

```txt
07:13:00][W][wifi_esp8266:482]: Event: Disconnected ssid='TR7272' bssid=06:18:D6:27:CA:64 reason='Auth Expired'
[07:13:00][W][wifi:549]: Error while connecting to network.

_esp8266:482]: Event: Disconnected ssid='TR7272' bssid=06:18:D6:27:CA:64 reason='Handshake Failed'
[07:13:00][W][wifi:585]: Restarting WiFi adapter...
[07:13:00][I][wifi:257]: WiFi Connecting to 'TR7272'...
[07:13:05][W][wifi_esp8266:482]: Event: Disconnected ssid='TR7272' bssid=06:18:D6:27:CA:64 reason='Auth Expired'
[07:13:05][W][wifi:549]: Error while connecting to network.
[07:13:05][W][wifi:585]: Restarting WiFi adapter...
[07:13:05][W][wifi_esp8266:482]: Event: Disconnected ssid='TR7272' bssid=06:18:D6:27:CA:64 reason='Authentication Failed'
```

MallocArray, 2023-02-21:
I've been seeing this off and on as well.  I saw some discussion about increasing the update interval on my OLED display, but it didn't reliably resolve it.  If I let the device sit for 1-5 minutes, it seems to eventually connect, but while I'm watching the logs, I see the 'Auth Expired" message and then it randomly connects.

In another Issue https://github.com/esphome/issues/issues/3944 the requester mentioned changing the power_output which I haven't tried yet.

riker65, 2023-02-24:
I did set output_power: 8.5 but this does not solve the issue

Diaoul, 2023-09-23:
This worked for me
```
  output_power: 8.5dB
```

Fluoclip, 2023-12-27:
> This worked for me
> 
> ```
> 
>   output_power: 8.5dB
> 
> ```

Where do you put this code? 
"output_power: 8.5dB"

ssieb, 2023-12-27:
In the `wifi:` section.