esphome/issues #4054: remote_transmitter component fails linking with pi pico w

stheid, 2023-01-21:
### The problem

Using the Remote Transmitter component with the Pi Pico W fails with a Linker error

### Which version of ESPHome has the issue?

 2022.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

No issue with HASS

### What platform are you using?

RP2040

### Board

rpipicow

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
'''
esphome:
  name: nastyctl

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_pwd

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

remote_transmitter:
  pin: GPIO13
  carrier_duty_percent: 50%

button:
  - platform: template
    name: "PWR Button"
    on_press:
      - remote_transmitter.transmit_nec:
          address: 0x4E87
          command: 0xE817
'''
```

### Anything in the logs that might be useful for us?

```txt
'''
Compiling .pioenvs/nastyctl/src/main.cpp.o
Linking .pioenvs/nastyctl/firmware.elf
/config/.esphome/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pioenvs/nastyctl/src/main.cpp.o: in function `setup':
main.cpp:(.text.setup+0x878): undefined reference to `_ZTVN7esphome18remote_transmitter26RemoteTransmitterComponentE'
/config/.esphome/platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: main.cpp:(.text.setup+0x87c): undefined reference to `_ZTVN7esphome18remote_transmitter26RemoteTransmitterComponentE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/nastyctl/firmware.elf] Error 1
========================= [FAILED] Took 32.97 seconds =========================
'''
```

### Additional information

Linker Error

BloodStainedCrow, 2023-02-09:
Seems like the same problem as #4033

stheid, 2023-02-09:
In deed, you are right. i did not find that issue when i was searching.

Closing this as a duplicate of #4033

closed 2023-02-09 by stheid