esphome/issues #4055: Can't use MQTT on Raspberry Pico W

utdrmac, 2023-01-22:
### The problem

```
$ esphome run garage.yaml --device /dev/tty.usbmodem144101
INFO Reading configuration garage.yaml...
Failed config

mqtt: [source garage.yaml:12]

  This feature is only available on ['esp32', 'esp8266'].
  broker: 10.10.10.10
  port: 1883
```

### Which version of ESPHome has the issue?

2022.11.5

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
---
esphome:
  name: garage-pico

rp2040:
  board: rpipicow

  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

mqtt:
  broker: 10.10.10.203
  port: 1883

output:
  - platform: gpio
    pin:
      number: 32  # 25 for Pico (non-W)
      mode: output
    id: LED

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
```
```

mirobiro, 2023-02-17:
I've asked around about this. Seemingly it's simply unsupported and won't be until somebody with the requisite skills decides to implement this. I can't fathom why it was considered worth doing for ESP8266 and ESP32 but not the Pico  W, seems bizarre.

skilau, 2023-08-28:
I have added support for it, and have issued a pull request for it here:
https://github.com/esphome/esphome/pull/5305

bdraco, 2026-03-12:
Closing — MQTT on RP2040 isn't supported yet. This is a feature request rather than a bug, so the best place for it is https://github.com/esphome/feature-requests. There was a PR (https://github.com/esphome/esphome/pull/5305) that attempted to add it but it was closed without merging.

closed 2026-03-12 by bdraco