esphome/issues #4057: Error: The program size is greater than maximum allowed. But worked on version 2022.10

quizzical86, 2023-01-22:
### The problem

I updated ESPHOME to 2022.12 and wanted to upgrade my devices. However one of them can't compile and generates the error:
Error: The program size (1874625 bytes) is greater than maximum allowed (1835008 bytes)

On ESPHome version 2022.10.2 the YAML compiled just fine.
I wonder whether the issue is due to the changes in the flash partition table in version 2022.12? I understand those are for the bluetooth proxy and I am using that.
Is there a way I could solve this issue?

### Which version of ESPHome has the issue?

2022.12.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.4

### What platform are you using?

ESP32

### Board

MH-ET LIVE D1 mini ESP32

### Anything in the logs that might be useful for us?

```txt
Linking /data/kamerlo/.pioenvs/kamerlo/firmware.elf
Error: The program size (1874625 bytes) is greater than maximum allowed (1835008 bytes)
RAM:   [==        ]  18.6% (used 61000 bytes from 327680 bytes)
Flash: [==========]  102.2% (used 1874625 bytes from 1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 399.23 seconds
```

quizzical86, 2023-01-31:
Could this change be the cause of my problem?
https://github.com/esphome/esphome/pull/4171

jgracio, 2023-02-05:
Here same error without any code change

quizzical86, 2023-02-06:
The only solution I found was to cut parts of my code, e.g. the logger functionality so no more checking the logs for issues :-(

marek-vrana, 2023-02-11:
same here, using nspanel with ble proxy configuration. Program is too big to fit even with the minimal features and no logging. Before it worked without any issues.

lukin79, 2023-04-27:
Same issue w/o code change. Quite annoying.

elik745i, 2023-07-23:
same thing: Error: The program size (1939513 bytes) is greater than maximum allowed (1835008 bytes)

toomone, 2023-07-24:
Also same with Home Assistant 2023.7.3 and ESPHome v2023.2.2 :
```
Linking .pioenvs/irrigation-controller/firmware.elf
RAM:   [====      ]  43.2% (used 35388 bytes from 81920 bytes)
Flash: [==========]  106.6% (used 532625 bytes from 499696 bytes)
Error: The program size (532625 bytes) is greater than maximum allowed (499696 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 149.29 seconds =========================
```

toomone, 2023-08-10:
I got it working by doing the firmware update not over the air. Plug on the computer and with web Esphome works nicely. Then the OTA update works again.

dabepp, 2024-01-20:
> Plug on the computer and with web Esphome works nicely.

Same issue with my esp32 :-(

`Linking .pioenvs/esphome-web-ee267c/firmware.elf
RAM:   [==        ]  18.3% (used 60076 bytes from 327680 bytes)
Flash: [==========]  112.8% (used 2070677 bytes from 1835008 bytes)
Error: The program size (2070677 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
========================= [FAILED] Took 117.91 seconds =========================`

Gonna post my YAML later if anybody is intrested in.

Tscherno, 2024-03-25:
Same issue here with three different devices. None of them can be updated OTA. I also can't easily reach them anymore for wired updating. That's really bad!

lsdlsd88, 2024-05-12:
still an issue... any ideas?