esphome/issues #4058: Ble_tracker Scan set param failed: 5

Salvora, 2023-01-22:
### The problem

I have multiple ESP32 devices and I use almost the same settings for each of them. However, the nodemcu ESP32 device spams me with multiple scan errors of the ble_tracker. The rest of the ESP32 devkitv1 devices does not give me the same error.

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.6

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

ble_tracker

### Example YAML snippet

```yaml
substitutions:
  devicename: "Mutfak ESP32S"
  roomname: "Mutfak"

esphome:
  name: mutfak-esp32s-nodemcu
  includes:
    - lgtv_uart.h

esp32:
  board: nodemcu-32s
  framework:
    type: arduino
    version: recommended

wifi:
  ssid: !secret esp_wifi_ssid
  password: !secret esp_wifi_pass
  id: mutfak_esp32s_wi_fi
  reboot_timeout: 30min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$devicename NodeMCU"
    password: !secret esp_ota_password
    ap_timeout: 5min

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esp_api_password
  encryption:
    key: !secret mutfak-esp32s-nodemcu_enc_key
  id: api_config

[... 407 lines cut ...]
[23:16:53][E][esp32_ble_tracker:195]: Scan set param failed: 5
[23:16:53][D][esp32_ble_tracker:327]: Starting scan...
[23:16:53][D][esp32_ble_tracker:186]: Stopping scan after failure...
[23:16:53][E][esp32_ble_tracker:195]: Scan set param failed: 5
[23:16:53][D][esp32_ble_tracker:327]: Starting scan...
[23:16:53][D][esp32_ble_tracker:186]: Stopping scan after failure...
[23:16:53][E][esp32_ble_tracker:195]: Scan set param failed: 5
[23:16:53][D][esp32_ble_tracker:327]: Starting scan...
[23:16:53][D][esp32_ble_tracker:186]: Stopping scan after failure...
[23:16:53][E][esp32_ble_tracker:195]: Scan set param failed: 5
[23:16:53][D][esp32_ble_tracker:327]: Starting scan...
```

### Additional information

I use the exact same configuration for other esp32 boards (devkitv1) but they don't have the same problem.
```
esp32_ble_tracker:
  scan_parameters:
    interval: 60s
    active: True
  id: esp_ble_tracker
```
At the same time, wifi connection is very unreliable.
This issue might be related to https://github.com/esphome/issues/issues/1503

Salvora, 2023-01-24:
I was wrong, all of my esp32 devices are spamming the same thing and having wifi connection problems.

jacek-dargiel, 2023-05-31:
@Salvora Have you ever figured this out?

kamnetanker, 2023-06-30:
Hello, you have issue in your configuration. In your scan parameters interval shall be ms. For example correctly working config looks like this:
```
esp32_ble_tracker:
  scan_parameters:
    interval: 60000ms
    active: false
```

adorobis, 2023-07-06:
I have the same issue. Trying to set the interval as 60000ms or 60s results in the same messages flooding the log.