esphome/issues #4060: UPdate to 2022.12.5 fails in HA with error unknown manifest

SlowmoDK, 2023-01-22:
### The problem

Can't install ghcr.io/esphome/esphome-hassio:2022.12.5: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.12.5&fromImage=ghcr.io%2Fesphome%2Fesphome-hassio&platform=linux%2Famd64: Internal Server Error ("manifest unknown")

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP8266

SlowmoDK, 2023-01-22:
seems to a image issue, poberly fixed by tomorrow 

i'll try update again later

krazos, 2023-01-22:
Same issue

jesserockz, 2023-01-22:
Yeah there have been some recent changes which seems to have broken the workflows. I have made major changes to the way they are published in esphome/esphome#4320

If you force HA to check for updates, it should work again as I patches the addon yaml to point to the `-{arch}` based images again for now.

krazos, 2023-01-22:
Update worked for me after forcing update check. Many thanks!

SlowmoDK, 2023-01-22:
Works after update refresh 

Thank you for quick reply and fix !

Murianwind, 2023-01-22:
I tried to reinstall after deleting, but the installation does not work.

Murianwind, 2023-01-22:
Wow... it's installed now. Thank you for correcting.

bcutter, 2023-01-22:
Supervisor log (running HA OS):
```
23-01-22 23:52:45 INFO (SyncWorker_1) [supervisor.docker.addon] Updating image ghcr.io/esphome/esphome-hassio-aarch64:2022.12.3 to ghcr.io/esphome/esphome-hassio:2022.12.5
23-01-22 23:52:45 INFO (SyncWorker_1) [supervisor.docker.interface] Downloading docker image ghcr.io/esphome/esphome-hassio with tag 2022.12.5.
23-01-22 23:52:46 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install ghcr.io/esphome/esphome-hassio:2022.12.5: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.12.5&fromImage=ghcr.io%2Fesphome%2Fesphome-hassio&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
```

bcutter, 2023-01-23:
Update: Update worked few minutes ago as it came up again (something on the back might have changed, maybe a new release with same version has been pushed?).

SlowmoDK, 2023-01-24:
FIXed

closed 2023-01-24 by SlowmoDK