esphome/issues #4061: bluetooth proxy comile fatal errors

glbailey, 2023-01-22:
### The problem

Created bluetooth proxy from https://esphome.github.io/bluetooth-proxies/ on a separate (from Home assistant) Linux Mint machine just fine. Device reports it was created with version 2022.9.4. Adopted with Home Assistant on a Linux Mint machine in a Virtual Box VM. When trying to install to the bluetooth proxy device, a compile error is reported and compile is aborted. 
- ESPHome addon v 2022.12.3
- Home Assistant (OVA) running in Virtual Box on Linux Mint on Desktop PC.
    Home Assistant 2023.1.2
    Supervisor 2022.12.1
    Operating System 9.4
    Frontend 20230104.0 - latest 
- Virtual Box v 6.1.38_Ubuntu
- Linux Mint 20.3 Cinnamon
    Kernel 5.4.0-137

VM has 2 processors (out of 8) assigned and 4GB memory assigned. Peak processor use observed (happens during other ESPHome compiles) is 69% and peak mem usage is 35%. Disk space assigned 32GB and 58% usage reported by HA.
HA has been running for over 10 months (not a new install)

The 18th line of the log attached shows the fatal error. This is fully repeatable.

Log file excerpt of fatal error (full error log attached below):

-   Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
-   XXX lineno: 998, opcode: 172
-   ERROR: Exception:
-   Traceback (most recent call last):
-   File "/usr/share/python-wheels/resolvelib-0.5.4-py2.py3-none-any.whl/resolvelib/resolvers.py", line 171, in _merge_into_criterion
-   crit = self.state.criteria[name]
-   KeyError: 'pyparsing'

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.2

### What platform are you using?

ESP32-IDF

### Board

ESP32 DEVKIT V1

### Component causing the issue

compiler

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-a2f6e0
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32:
  board: esp32dev
  framework:
[... 105 lines cut ...]
  File "/usr/lib/python3.9/zipfile.py", line 998, in _read1
    data = self._decompressor.decompress(data, n)
SystemError: unknown opcode
*** Error 2
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/esp32-bluetooth-proxy-a2f6e0/.pioenvs/esp32-bluetooth-proxy-a2f6e0/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:454 (message):
  /usr/bin/python3: Error while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:3 (project)
```

### Additional information

My suspicion was that it had something to do with python as the documentation (for manual installs) states "Your distribution probably already has Python installed. Confirm that it is at least version 3.9"
This is a problem since Linux Mint (and Ubuntu) reportedly considers 3.8 to be the latest and updates to that are apparently discouraged but I do not know if this is correct or not. At any rate, both my HA system and my other linux mint laptop report python3 at 3.8.10. I don't know how to reconcile that with the manual install page info.

glbailey, 2023-01-23:
UPDATE: after updating to ESPHOME 2022.12.5 and Home assistant 2023.1.7 the issue has now been eliminated. Compiles complete correctly.

closed 2023-01-23 by glbailey

Stooovie, 2023-03-24:
Not here. ESPHome 2023.3.1 on HA 2023.3.6, unable to update BTproxy to anything beyond 2022.12.6

ssieb, 2023-03-24:
@Stooovie, please stop commenting in random unrelated issues.