esphome/issues #4062: esphome/esphome docker images manifest unknown error

jfurtner, 2023-01-23:
### The problem

Attempting to update the docker image for esphome/esphome gives an error _manifest not found and manifest unknown_

2022.12.5:
`# docker pull esphome/esphome:2022.12.5
Error response from daemon: manifest for esphome/esphome:2022.12.5 not found: manifest unknown: manifest unknown
`

2022.12.4:
`# docker pull esphome/esphome:2022.12.4
Error response from daemon: manifest for esphome/esphome:2022.12.4 not found: manifest unknown: manifest unknown
`

### Which version of ESPHome has the issue?

2022.12.5

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

GuyKh, 2023-01-23:
Me too.
Both 2022.12.4 and 2022.12.5 **do not appear in docker hub tags**

Ulrar, 2023-01-23:
Not sure if this is related or not, but there's been issues with the latest version of buildx lately in other projects.
My understanding is `--format=docker` needs to be added to the build arguments for now, to keep generating the containers in the usual format instead of OCI which isn't well supported everywhere.

d00wirre, 2023-01-23:
Same issue on both docker.io and ghcr.io, 2022.12.3 is the latest on both.

d00wirre, 2023-01-24:
Works fine again, 2022.12.6 available

closed 2023-01-27 by jfurtner